pyvisa/pyvisa #460: pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

mdhayath64, 2019-10-01:
import pyvisa
rm=pyvisa.ResourceManager()
sg=rm.open_resource('TCPIP0::192.168.1.254::inst0::INSTR')
sg.query('*IDN?')

when i use these code iam gettin dat error

MatthieuDartiailh, 2019-10-02:
Could you please post the output of `pyvisa-info` (entered in a command line) ? and the kind of instrument you are trying to access ? Also since this is a TCPIP resource can you check that you ping the instrument (`ping 192.168.1.254` in a command line and get something that does not time out).

MatthieuDartiailh, 2019-11-03:
ping @mdhayath64

m47jiang, 2019-11-22:
@MatthieuDartiailh @mdhayath64 I'm having a similar issue here's my code:
I'm trying to communicate with a Keysight 34450A DMM, here's my code
`import pyvisa
import time

rm = pyvisa.ResourceManager('@py')
rm.list_resources()
inst = rm.open_resource('USB0::10893::45848::MY58140020::0::INSTR')
inst.read_termination = '\n'
inst.write_termination = '\n'
print(inst.write('*IDN?'))
print(inst.read_bytes(1))
print(inst.write('MEAS:VOLT:DC?'))
time.sleep(2)
print(inst.read_bytes(1))`

Here's my results 
`(20, <StatusCode.success: 0>)
b'Keysight Technologies,34450A,MY58140020,01.02-01.00\n'
(28, <StatusCode.success: 0>)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
`
pyvisa-info gives the following: 
`Machine Details:
   Platform ID:    Linux-4.15.0-70-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.5.2
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Oct  8 2019 13:06:37 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      TCPIP INSTR: Available 
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      TCPIP SOCKET: Available 
`

m47jiang, 2019-11-22:
Welp I found out my error, when you open the instrument resource, it defaults to using chunksizes of 20480 bytes, for some reason, the instrument doesn't return a \r\n or \n when the command gets read back... so the solution is to find out the exact size of the return value and input that as the chunk_size
`inst.chunk_size = 128`
and since it doesn't return a \r\n or \n, the program tries to read forever and generates a timeout... my instrument will actually generate a \n for some of the commands sent over, and others not!!

m47jiang, 2019-11-22:
I take it back, this is unreliable and doesn't work most of the time

MatthieuDartiailh, 2019-11-22:
I am surprised that you get the full message by requesting a single byte, this may be a bug in pyvisa-py. Another point is that since your query request a measurement be sure the instrument is properly trigged as otherwise it may not answer, but since you can get things to work (although unreliably) it may not be it.

Keysight has a VISA implementation for linux that you may be able to use, since the USB backend of pyvisa-py is not always extremely well behaved at the moment (it works well for some users but not for others).