pyvisa/pyvisa #461: Documentation / FAQ suggestion: start with a terminal app

rdpoor, 2019-10-04:
(first off: pyvisa is a great tool and superbly easy to use.  thank you!)

Suggestion for the FAQ: If you can't get "*IDN?" to work in a serial terminal, you probably won't be able to get it to work in pyvisa.  So test your connection using a serial terminal first.

Details:

After futzing around with pyvisa and failing to get a response from my HP / Agilent / Keyspan instruments, I had the idea to open a serial terminal on the appropriate USB port (using Putty or Tera Term on Windows, CoolTerm or Terminal / screen on macOS) and see if I could get a response there.

In my case, the default I/O settings for the HP instruments used the GPIB interface rather than the serial port, and used 9600 7-e-1 serial protocol (rather than 9600 8-n-1).  After configuring the instrument's I/O settings through the front panel, pyvisa could communicate with no trouble.

MatthieuDartiailh, 2019-10-04:
We could expand that kind of debugging advice it is true, but it also depends on the kind of resource you are trying to access. However I would rather expand the Communicating with your instrument section. If you have time to put up a PR I would be happy to review it.

MatthieuDartiailh, 2020-09-07:
Addressed in c12b3ef8

closed 2020-09-07 by MatthieuDartiailh