pyvisa/pyvisa #462: query_binary_values() manged under macOS

rdpoor, 2019-10-06:
I have code that does a screen grab of a Rigol scope:

    https://github.com/rdpoor/rigol-grab

which works perfectly under Windows 10 but mangles the .png data under macOS.  I'm not 100% certain that this is a pyvisa issue, but here goes:

## The code in question

        buf = self.rigol().query_binary_values(':DISP:DATA? ON,0,PNG', datatype='B')
        with open(filename, 'wb') as f:
            self.verbose_print('Capturing screen to', filename)
            f.write(bytearray(buf))

## Under Windows:

### The environment under Windows

```
r@NUC-10 MINGW64 ~/Projects/rigol-grab (master)
$ pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\r\.virtualenvs\rigol-grab-pxnv5k_r\scripts\python.exe
   Version:        3.7.4
   Compiler:       MSC v.1916 32 bit (Intel)
   Bits:           32bit
   Build:          Jul  8 2019 19:29:22 (#tags/v3.7.4:e09359112e)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 19922944
         Spec. Version: 5244928
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.0.2). Backend: N/A
      USB RAW: Available via PyUSB (1.0.2). Backend: N/A
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

```
### The results under Winows

[image]

### The `log_to_screen()` trace under Windows

```

$ python rigol_grab.py -v -a -f foo.png
2019-10-06 06:27:51,223 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2019-10-06 06:27:51,223 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2019-10-06 06:27:51,227 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll', None, None]
[... 479 lines cut ...]
2019-10-06 06:24:45,245 - pyvisa - DEBUG - viRead(1, <ctypes.c_char_Array_20480 object at 0x115fb6950>, 20480, 'c_uint(29)') -> 0
Capturing screen to foo.png
2019-10-06 06:24:45,315 - pyvisa - DEBUG - USB0::0x1AB1::0x04CE::DS1ZA184753245::INSTR - closing
2019-10-06 06:24:45,316 - pyvisa - DEBUG - viDisableEvent(1, 1073709055, 65535) -> 1073676291
2019-10-06 06:24:45,316 - pyvisa - DEBUG - viDiscardEvents(1, 1073709055, 65535) -> 1073676292
2019-10-06 06:24:45,316 - pyvisa - DEBUG - viClose(1,) -> 0
2019-10-06 06:24:45,316 - pyvisa - DEBUG - USB0::0x1AB1::0x04CE::DS1ZA184753245::INSTR - is closed
2019-10-06 06:24:45,316 - pyvisa - DEBUG - Closing ResourceManager (session: 4097)
2019-10-06 06:24:45,317 - pyvisa - DEBUG - viClose(4097,) -> 0
2019-10-06 06:24:45,332 - pyvisa - DEBUG - USB0::0x1AB1::0x04CE::DS1ZA184753245::INSTR - closing
```

## An observation

It's clear that the calls to viRead are quite different, e.g.:

Windows:

    viRead(209681936, <ctypes.c_char_Array_20480 object at 0x0CD23DF0>, 20480, 'c_ulong(54)') -> 1073676293

macOS:

    viRead(1, <ctypes.c_char_Array_20480 object at 0x115fb6950>, 20480, 'c_uint(537)') -> 1073676293

... but I'm not sure if or how that's significant.

MatthieuDartiailh, 2019-10-07:
That is extremely weird, but I don't see how PyVISA could do that (nothing is os dependent). It may be an issue with the underlying VISA implementation. Could you do the same test using a different interface (TCPIP for example) to see if you observe the same behavior ? I know that Rigol instruments behave oddly on USB and I am wondering if the Mac implementation of the VISA library you are using may be missing that.

rdpoor, 2019-10-07:
> I don't see how PyVISA could do that

Agreed (which is why I prefaced this report with "I'm not 100% certain that this is a pyvisa issue...".  

I notice that I'm running PyVISA 1.9.1 while some people using my code are running PyVISA 1.10.1.  So my slightly naive questions: 

- What's the most reliable way to upgrade to PyVISA 1.10.1?
- Is there a safe way to switch between the two, i.e. so I can do comparison testing?

I'll report back what I find.

MatthieuDartiailh, 2019-10-07:
For pyvisa I would suggest using pip and virtualenv (or conda env if you are using the Anaconda distribution), in which case you can easily switch between versions. For NI there are no good way to switch between two versions, you have to uninstall fully.

rdpoor, 2019-10-07:
> I would suggest using pip and virtualenv

... or pipenv, which has become my standard technique!  :)

> For NI there are no good way to switch between two versions

I noticed.  I did upgrade the NI library and it didn't make a difference.  My next thought is to print out the low-level byte stream interaction to and from the device and compare it to the (working) windows version.  Maybe that will help me pinpoint the problem. 

As this does not appear to be a pyvisa problem per-se, feel free to close this issue if you feel that's appropriate.

MatthieuDartiailh, 2019-10-07:
I would be happy to know if you can figure this one out. In particular I am most interested in knowing if you see the same behavior on TCPIP or not.

rdpoor, 2019-10-07:
Well, another naive question: is TCPIP available via USB?  Or do I move my scope next to my router and plug in via ethernet?

MatthieuDartiailh, 2019-10-07:
You need to connect your instrument using an ethernet cable. But if your computer has multiple ethernet port (or if you accept to lose internet access for a time) you can connect your instrument in local and bypass your router. Set a fix IP on the instrument and the computer on the same subnet and you are done.

closed 2020-09-07 by MatthieuDartiailh