pyvisa/pyvisa #464: Unexpected Timeout using "query_binary_values"

sclel016, 2019-10-12:
Hi,

I'm seeing some unexpected behavior while reading a scope waveform using pyvisa over tcpip. When I increase my the memory depth of my capture beyond ~14000 bytes of data I get
`pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.`

To troubleshoot I've been sniffing the LAN with wireshark. With a 14KB waveform, the trace has finished transmission after ~40ms. The transmission occurs in approximately the same amount of time for much larger waveforms.

So far I've tried increasing the chunk_size and timeout to no avail. Hope someone can give me a hand with this, I really like the framework!

Stew

MatthieuDartiailh, 2019-10-12:
Can you please post the output of `pyvisa-info` and the model of instrument you are communicating with ?

I have no obvious suggestion for you right now. One thing you could try is reading the answer in multiple call to read_bytes if you know how many bytes you expect to see if the error always occurs after the same volume of transferred data.

sclel016, 2019-10-12:
Machine Details:
   Platform ID:    Linux-5.2.20-1-MANJARO-x86_64-with-arch-Manjaro-Linux
   Processor:      

Python:
   Implementation: CPython
   Executable:     /opt/anaconda3/envs/instr_env/bin/python
   Version:        3.7.4
   Compiler:       GCC 7.3.0
   Bits:           64bit
   Build:          Aug 13 2019 20:35:49 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib' 

Communication with Siglent SDS 1202X-E

I'll try splitting up my read calls, fortunately the intrument allows for partial sample memory reads. Communication seems to be pretty rock solid over matlab w/ NI-VISA, I'm trying to get away from the windows stack if possible. Any other connection configurations you can think of?

MatthieuDartiailh, 2019-10-12:
You can also try to use the master branch of pyvisa-py (some of the fixes on it may address your issue). You can also have a look at https://www.keysight.com/main/software.jspx?ckey=2175637&nid=-33002.977662&cc=eng&lc=eng but not sure it will run on Manjaro.

sclel016, 2019-10-14:
So quick follow up. I installed latest master and ran into the same issue w/ interface timeout. I kept digging and began inspecting the way pyvisa-py was handling read events. Turns out that the reason I was getting a timeout was due to my misunderstanding of what `chunk_size` means. The whole waveform was getting processed as a single chunk (16Mpts * 8bits/pt). I upped the `chunk_size` accordingly and I'm having no issues now. It may be worth a quick description in the docs by someone who understands the codebase.

closed 2019-10-14 by sclel016