pyvisa/pyvisa #466: usb_control_in not Implemented

Zanobos, 2019-10-18:
Hi all,
In order to connect to a BK Precision 8600 (from a linux device, so no NI-visa, but I am using pyvisa-py as BE), I have to perform a change of control. 

I have been able to do that in python-ivi (usbtmc) with something like this:

`self.device.ctrl_transfer(bmRequestType=0xA1, bRequest=0xA0, wValue=0x0001, wIndex=0x0000, data_or_wLength=1) ` and after this instruction the instrument works correctly.

I am trying to perform the same operation using this library, for example using this instruction:

`res.control_in(request_type_bitmap_field=0xA1, request_id=0xA0, request_value=0x0001, index=0x0000, length=1)` but it fails:

(complete example):

```
>>> from visa import ResourceManager
>>> rm = ResourceManager()
>>> rm.list_resources()
(u'USB0::11975::34816::EDITED::0::INSTR',)
>>> res = rm.open_resource('USB0::11975::34816::EDITED::0::INSTR')
>>> res.control_in(request_type_bitmap_field=0xA1, request_id=0xA0, request_value=0x0001, index=0x0000, length=1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/andrea/Workspace/python-usb-test/venv/local/lib/python2.7/site-packages/pyvisa/resources/usb.py", line 46, in control_in
    request_id, request_value, index, length)
  File "/home/andrea/Workspace/python-usb-test/venv/local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1345, in usb_control_in
    raise NotImplementedError
NotImplementedError
```

Looking at the code, many USB operations are not implemented. Is this by design? 
Andrea

MatthieuDartiailh, 2019-10-18:
That function is rarely used and has not been implemented yet. If you have time to make a PR based on python-usbtmc I will be happy to review it.

Zanobos, 2019-10-18:
Here a PR for the relative function:
https://github.com/pyvisa/pyvisa-py/pull/212

As explained in the comment, it is working for the control_transfer part, however I would need an help for the read data stage part.

Best,
Andrea

MatthieuDartiailh, 2019-10-18:
Thanks I will try to have a look as soon as I can. May take some time since I have to dive into the specifications.