pyvisa/pyvisa #467: Using set_visa_attribute

Chilldose, 2019-10-23:
Hi,

I may have a bit of a problem/misunderstanding on how to use the set_visa_attribute method.
As far as I have understood the syntax to set a VISA attribute via pyvisa is:

import pyvisa
rm = pyvisa.ResourceManager()
instr = rm.open_resource('TCPIP0::1.2.3.4::56789::SOCKET')
instr.set_visa_attribute(pyvisa.attributes.AttrVI_ATTR_SUPPRESS_END_EN, pyvisa.constants.VI_TRUE)

So in words: **instr** is a TCP/IP Socket connection and I want the "VI_ATTR_SUPPRESS_END_EN" VISA attribute set to True via "pyvisa.constants.VI_TRUE".

The problem now is that I get a ctypes.ArgumentError when I do so. :
     Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\*****\lib\site-packages\pyvisa\resources\resource.py", line 282, in set_visa_attribute
    return self.visalib.set_attribute(self.session, name, state)
  File "C:\*****\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1649, in set_attribute
    return library.viSetAttribute(session, attribute, attribute_state)
ctypes.ArgumentError: argument 2: <class 'TypeError'>: wrong type

What am I doing wrong here?

Thanks in advance!

MatthieuDartiailh, 2019-10-23:
This feature is extremely poorly documented (it is not really documented actually). The attributes defined in pyvisa.attributes are descriptors (ie properties) and are dynamically attached to the resource class.
As a consequence you can do the following:
```
import pyvisa
rm = pyvisa.ResourceManager()
instr = rm.open_resource('TCPIP0::1.2.3.4::56789::SOCKET')
instr.VI_ATTR_SUPPRESS_END_EN = True
```

some common attribute have a Python alias (such as VI_ATTR_SEND_END_EN, aliased as send_en).

Another option (which I do not recommend) would be:
```
import pyvisa
rm = pyvisa.ResourceManager()
instr = rm.open_resource('TCPIP0::1.2.3.4::56789::SOCKET')
instr.set_visa_attribute(pyvisa.constants.VI_ATTR_SUPPRESS_END_EN, pyvisa.constants.VI_TRUE)
```

MatthieuDartiailh, 2019-10-23:
Please see if #468 makes things clear enough.

Chilldose, 2019-10-24:
Thank you! 

It works!

commit eae079d3 (2019-10-24): Merge #468

468: Document how to access the VISA attributes r=MatthieuDartiailh a=MatthieuDartiailh

Closes #467
PR #468 (2019-10-24): Document how to access the VISA attributes
Closes #467
closed 2019-10-24 by Chilldose