pyvisa/pyvisa #469: How does pyvisa find GPIB devices?

j-gooding, 2019-10-24:
I am using an NI GPIB ENET/1000 device with the linux 488.2 driver software. Pyvisa is able to see the GPIB0::1::INSTR device and communicate with it. Everything works great. 

I would like to dockerize the python code, which leads me to my question, how does pyvisa discover the GPIB0::1::INSTR device? What /dev/* files, etc. do I need to mirror to the docker instance in order for pyvisa to find it inside the docker instance? Thank you.

MatthieuDartiailh, 2019-10-26:
The instrument detection is done through the assertion of one line of the GPIB bus. When dockerizing I would be mostly concerned by the access to the hardware.

j-gooding, 2019-10-26:
Exactly. I’m sure it can be done, I just don’t know what to give it access to.

MatthieuDartiailh, 2019-10-28:
You could try to reach to linux-gpib people. They would know more about that kind of low level question.

neilpanchal, 2019-11-07:
@j-gooding what about using a virtual python environment? virtualenv, conda env, or pipenv?

Dockerization for a service that interacts with hardware could be difficult to manage and make it work.

MatthieuDartiailh, 2019-11-20:
I am closing this pyvisa simply relies on the underlying GPIB library. Feel free to re-open if the need arises.

closed 2019-11-20 by MatthieuDartiailh

MatrixManAtYrService, 2021-01-05:
I ran across this page: https://www.cl.cam.ac.uk/~osc22/tutorials/gpib_usb_linux.html which seems to indicate that the devices you need to pass on to the docker container are:

- /dev/gpib1
- /dev/gpib2
- ...
- /dev/gpib15

But I'm not currently set up to test that idea.