pyvisa/pyvisa #470: open_resource doesn't open or timeout, hangs indefinitely

Grant-Giesbrecht, 2019-11-08:
I'm writing a pyvisa app which controls lab instruments connected to my Mac via ethernet. Strangely, I'll run the program a few times and everything works great, then, seemingly randomly, it will fail to connect to the instruments. Even trying to open an instrument with pyvisa in a python shell won't work. The only way to get pyvisa to connect again is to restart the computer. The way it fails is I'll call open_resource, but the function never returns and never times out.

Does anyone know what could make open_resource() hang indefinitely? I thought it had a timeout built in? Do you have any thoughts on how a program could seemingly "break" pyvisa even after closing, as mine seems to do?

Here's what I get when I try to connect to the instrument while the problem is present:
```
>>> import pyvisa as pv
>>> pv.log_to_screen()
>>> rm = pv.ResourceManager()
2019-11-08 10:21:09,016 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2019-11-08 10:21:09,017 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2019-11-08 10:21:09,019 - pyvisa - DEBUG - Automatically found library files: ['/Library/Frameworks/visa.framework/visa', None, None, None, None]
2019-11-08 10:21:09,026 - pyvisa - DEBUG - No user defined library files
2019-11-08 10:21:09,026 - pyvisa - DEBUG - The NI implementation available
2019-11-08 10:21:09,028 - pyvisa - DEBUG - Automatically found library files: ['/Library/Frameworks/visa.framework/visa', None, None, None, None]
2019-11-08 10:21:09,029 - pyvisa - DEBUG - No user defined library files
2019-11-08 10:21:09,045 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2019-11-08 10:21:09,045 - pyvisa - DEBUG - Created library wrapper for /Library/Frameworks/visa.framework/visa
2019-11-08 10:21:10,124 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x10f0eb8c0>',) -> 0
2019-11-08 10:21:10,124 - pyvisa - DEBUG - Created ResourceManager with session 4097
>>> rm.open_resource("TCPIP0::192.168.1.19::INSTR")
2019-11-08 10:21:20,193 - pyvisa - DEBUG - viParseRsrcEx(4097, 'TCPIP0::192.168.1.19::INSTR', 'c_ushort(6)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x10f15c710>, <ctypes.c_char_Array_256 object at 0x10f15c830>, <ctypes.c_char_Array_256 object at 0x10f15c5f0>) -> 0
2019-11-08 10:21:20,193 - pyvisa - DEBUG - TCPIP0::192.168.1.19::INSTR - opening ...
```
It gets stuck here indefinitely.

MatthieuDartiailh, 2019-11-08:
Could you post the output of `pyvisa-info` ?

Grant-Giesbrecht, 2019-11-08:
Sure, here it is:

Machine Details:
   Platform ID:    Darwin-19.0.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7
   Version:        3.7.4
   Compiler:       Clang 6.0 (clang-600.0.57)
   Bits:           64bit
   Build:          Jul  8 2019 14:54:52 (#v3.7.4:e09359112e)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 32, 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

MatthieuDartiailh, 2019-11-09:
This is indeed unusual. The only things that comes to my mind is that somehow at one point the state of the NI VISA gets corrupted (crash at a wrong time). I have seen this happen on Windows but usually with different consequences. 
In your case did you get any unexpected interruption before you see the appearance of that issue ?

Grant-Giesbrecht, 2019-11-09:
The only interruptions that I can think of would be that the instruments I'm testing have been physically disconnected from my computer while pyvisa was connected. The problem didn't manifest immediately after this, however. 

Assuming NI VISA is getting corrupted, do you know a way to reset NI VISA without restarting the computer?

MatthieuDartiailh, 2019-11-09:
No sadly I don't. Out of curiosity can you still ping the IP of the instrument ? Did you try switching to pyvisa-py after ni breaks ? It may help pinpoint who is at fault.

Grant-Giesbrecht, 2019-11-10:
I'm still able to ping the instrument when pyvisa can't connect. Good idea about pyvisa-py, I'll give that a shot tomorrow when I'm back at the lab.

Grant-Giesbrecht, 2019-11-13:
pyvisa-py works!  I couldn't connect to the instruments with pyvisa, but pyvisa-py had no problem. I suppose it's NI-VISA that's malfunctioning, then. 

I'm not sure if this is related, but perhaps it has something to do with many NI apps being 32 bit and my OS (macOS Catalina) dropping support for 32 bit apps. All of the NI support applications like NI-VISA Configuration, Driver Wizard, NI-VISA Server can't run anymore for that reason. Though I'd think if this were the problem I was running into pyvisa never would have connected, but pyvisa was able to connect sometimes while I was using NI and Catalina.

MatthieuDartiailh, 2019-11-13:
pyvisa-py is not perfect but I am happy to know it works for you. MacOS is not an easy platform to do instrument automation and I would not be surprised that Catalina broke a couple of other things.

MatthieuDartiailh, 2019-11-20:
I am closing this since there does not seem to be much that can be fixed on pyvisa end. Feel free to re-open if the need arises.

closed 2019-11-20 by MatthieuDartiailh

unixneo, 2020-01-19:
pyvisa works fine on Catalina.

See this macOS Catalina related discussion for pyvisa and RIGOL:

https://www.unix.com/programming/283575-python-screen-capture-rigol-1054z-macos-catalina-using-ni-visa-2.htm