pyvisa/pyvisa #471: pyvisa memory error

saarangp, 2019-11-18:
I'm getting this error when I pull data from my scope, the data acquisition is occurring several times per hour and is crashing after 2 days. 

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "agingScript_v3.py", line 84, in acquire_data
    value = float(dmm.query('READ?')) #pull measurement and convert to float
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 613, in query
    return self.read()
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 336, in read
    resp = raw_read(recv_chunk)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 240, in read
    data = array_to_bytes(self.usb_recv_ep.read(size, self.timeout))
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 981, in read
    buff = util.create_buffer(size_or_buffer)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/util.py", line 162, in create_buffer
    return array.array('B', _dummy_s * length)
MemoryError

MatthieuDartiailh, 2019-11-20:
Looks like there is a memory leakage somewhere in your program. It is hard to say from your report if it comes from pyvisa or from something else. I would suggest to first try to determine where the may be coming from. (here is a reference that may help: https://medium.com/zendesk-engineering/hunting-for-memory-leaks-in-python-applications-6824d0518774)

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh