pyvisa/pyvisa #472: ModuleNotFoundError

thinker5112, 2019-11-19:
When I have successfully installed the pyvisa, it is showing that ModuleNotFoundError : No module named pyvisa.
[image]

MatthieuDartiailh, 2019-11-20:
I am not familiar with pycharm but did you check you installed pyvisa in the right environment ?

thinker5112, 2019-11-20:
Yes I have checked it. And it was installed in the right environment.

Regards,
S. D. Pavan Kumar
Member Technical Engineer,
HCL Technologies Ltd.

::DISCLAIMER::
The contents of this e-mail and any attachment(s) are confidential and intended for the named recipient(s) only. E-mail transmission is not guaranteed to be secure or error-free as information could be intercepted, corrupted, lost, destroyed, arrive late or incomplete, or may contain viruses in transmission. The e mail and its contents (with or without referred errors) shall therefore not attach any liability on the originator or HCL or its affiliates. Views or opinions, if any, presented in this email are solely those of the author and may not necessarily reflect the views or opinions of HCL or its affiliates. Any form of reproduction, dissemination, copying, disclosure, modification, distribution and / or publication of this message without the prior written consent of authorized representative of HCL is strictly prohibited. If you have received this email in error please delete it and notify the sender immediately. Before opening any email and/or attachments, please check them for viruses and other defects.

MatthieuDartiailh, 2019-11-20:
How did you install pyvisa and how did you check ?

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh