pyvisa/pyvisa #473: ResourceManager().last_status error

whyven, 2019-11-20:
Hello - 
I am having trouble resolving an error that occurs when I try to print the `rm.last_status`.  I am trying to implement timeout error handling.  In my code I am using `rm.visalib.last_status` to see if I can catch the timeout error.  However, when a timeout error occurs, it always reports 0.  So, I added some debugging code into my function to print `rm.last_status` as shown below... 
```
try:                
            self.scope.write( 'RUNS' )
            self.scope.query( '*OPC?' )
 
# --- Catch Timeout from lack of triggers
 except visa.VisaIOError as q:
            msg = 'Trigger failed: '
            msg += str(q)
            self.log.warning(msg)
            # ============================================================ DEBUG
            print( "VISA Error Recived = {}".format( q ) )
            print( "VISA Error in if statement = {}".format( VISA_TIMEOUT ) )
            print( "Session handle = {}".format( self._rm.session ))
            print( "last status = {}".format( self._rm.last_status ) )
             # ============================================================  END: DEBUG
            if ( self._rm.visalib.last_status == VISA_TIMEOUT ):
                self.log.warning( 'No trigger present.' )
                self.update_par( self.scopeErr, 'No trigger' )
                self.scope.clear()
                self._send_command( cmd='SYST:ERR:ALL?', query=True )
                self.update_par( self.scopeErr, self._resp )
                return None
            else:
                self._rm.close() 
                raise

```
Now I am getting the following error: 
```
Error: The session 1010711 does not seem to be valid as it does not have any last status
```
Here is my code for instantiating the ResourceManager in my class.
```
def connect_to_scope( self, *args, **kwargs  ):
        ''' Description: Configure the connection to the oscilloscope using VISA socket
                         connection. 
            
        '''
        try: 
            self.scope = self._rm.open_resource( 'TCPIP::'+ self.hostS.value.value +'::SOCKET' )
            self.scope.write_termination = '\n'
            self.scope.set_visa_attribute( visa.constants.VI_ATTR_TERMCHAR, 10 )
            self.scope.set_visa_attribute( visa.constants.VI_ATTR_TERMCHAR_EN, True)
            self.scope.timeout = 10000
        except Exception as e:
            self.log.error('open: '+str(e))
            self.update_par( self.adoStatus, str(e) )
            self.connOpened = 0
            return 1
        
        self.connOpened = 1
        self.log.info('Opened connection to '+str(self.hostS.value.value))
        self.log.info('Visa resource session: {}'.format( self._rm.session ))
        self.log.info('Visa resources: {}'.format( self._rm.list_resources() ))
        
        return self.initialize_scope()
```
I should also note that `rm.list_resources()` returns an empty dict... I've added query='?*::SOCKET' and still nothing. I know I am doing something wrong, I am just not sure what.  So any help would be greatly appreciated. 

Please let me know if there is any additional information you need in order to help and thank you in advance.

Here is the pyvisa-info: 
```
Machine Details:
   Platform ID:    Linux-3.10.0-957.10.1.el7.x86_64-x86_64-with-redhat-7.6-Maipo
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /cfs/x/usrLocals/WS/anaconda2/bin/python
   Version:        2.7.15
   Compiler:       GCC 7.3.0
   Bits:           64bit
   Build:          Dec 14 2018 19:04:19 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.0

Backends:
   ni:
      Version: 1.10.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 
```

MatthieuDartiailh, 2019-11-20:
Pyvisa-py currently does not handle properly the `last_status` attribute (it is never updated). I created https://github.com/pyvisa/pyvisa-py/issues/215 to track this. However you can achieve your goal without inspecting the `last_status` simply by inspecting the exception you catch which has a `return_code` attribute. 

Regarding the fact that the list of instrument is empty, once again this is pyvisa-py limitation which is tracked at https://github.com/pyvisa/pyvisa-py/issues/165.

whyven, 2019-11-20:
Thanks for the fast response!  
BTW, the attribute is `error_code` **not** `return_code`.  
Using that has solved my issue.

closed 2019-11-20 by whyven

MatthieuDartiailh, 2019-11-20:
My bad sorry