pyvisa/pyvisa #474: ViOpen Note found

camerin, 2019-11-21:
When executing the following program i have experience the error below. The error exists in both the NI and the Agilent Visa library. 
`import visa
import ctypes
import os
from ctypes.util import find_library
##lib_name = find_library('Visa')
##print(lib_name)
##filename="C:\\Program Files (x86)\\IVI Foundation\\VISA\\WinNT\\agvisa\\agbin\\visa32.dll"#
filename="C:\\Program Files (x86)\\IVI Foundation\VISA\\WinNT\\ktvisa\\ktbin\\visa32.dll"
print(os.path.isfile(filename))
##print(filename)
##hllDll = ctypes.WinDLL (filename)

rm = visa.ResourceManager(filename)`

`Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 135, in _init
    lib = Library(self.library_path)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\ctypes\__init__.py", line 369, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'C:\Program Files (x86)\IVI Foundation\VISA\WinNT\NIvisa\NiVisaCom.dll'. Try using the full path with constructor syntax.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\user\Desktop\git\AdnEMIScanScript\src\adn_emi_scan_script\SpiceRack_RF_API.py", line 16, in <module>
    rm = visa.ResourceManager("C:\\Program Files (x86)\\IVI Foundation\\VISA\\WinNT\\NIvisa\\NiVisaCom.dll")
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 1553, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 1525, in open_visa_library
    return cls(argument)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 109, in __new__
    obj._init()
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 137, in _init
    raise errors.LibraryError.from_exception(exc, self.library_path)
pyvisa.errors.LibraryError: Error while accessing C:\Program Files (x86)\IVI Foundation\VISA\WinNT\NIvisa\NiVisaCom.dll: Could not find module 'C:\Program Files (x86)\IVI Foundation\VISA\WinNT\NIvisa\NiVisaCom.dll'. Try using the full path with constructor syntax.`
Running python -m visa info
`Machine Details:
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\user\AppData\Local\Programs\Python\Python38-32\python.exe
   Version:        3.8.0
   Compiler:       MSC v.1916 32 bit (Intel)
   Bits:           32bit
   Build:          Oct 14 2019 19:21:23 (#tags/v3.8.0:fa919fd)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Could not get more info:
            function 'viOpen' not found
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'`

currently i am working on getting pyvisa-py working  with USB devices as a work around

MatthieuDartiailh, 2019-11-21:
How did you install NI VISA ? This reminds of a situation in which NI VISA did not install correctly. Also in your traceback your are trying to access NiVisaCom.dll which I assume exposes a COM interface and not the expected C interface. This does not match the code posted above which seems closer to what I would expect. Could you post the traceback matching the code ? 

Also can you get NI-MAX to work or not ? If not it may point to a bad installation.

camerin, 2019-11-21:
I installed using the NIvisa installer. I have also tried pointing directly to the DLL for NI, not just using @ni that resulted in the same error. I have also tired the Agilent dlls with the same error. 

NIMAX and Keysight IO both can talk to the instruments just fine.

Here is the trace back. I am in parallel trying to get this to working on getting LIB usb

import visa
filename="C:\\Program Files (x86)\\IVI Foundation\VISA\\WinNT\\ktvisa\\ktbin\\visa32.dll"
print(os.path.isfile(filename))
rm = visa.ResourceManager(filename)

Traceback (most recent call last):
  File "C:\Users\user\Desktop\git\AdnEMIScanScript\src\adn_emi_scan_script\SpiceRack_RF_API.py", line 26, in <module>
    rm = visa.ResourceManager(filename)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 1553, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 1525, in open_visa_library
    return cls(argument)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 109, in __new__
    obj._init()
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 144, in _init
    functions.set_signatures(self.lib, errcheck=self._return_handler)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 168, in set_signatures
    apply("viOpen", [ViSession, ViRsrc, ViAccessMode, ViUInt32, ViPSession], required=True)
  File "C:\Users\camerinh\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 95, in _internal
    set_signature(library, function_name, argtypes, restype, errcheck_)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 241, in set_signature
    func = getattr(library, function_name)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\ctypes\__init__.py", line 382, in __getattr__
    func = self.__getitem__(name)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\ctypes\__init__.py", line 387, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'viOpen' not found

MatthieuDartiailh, 2019-11-21:
Out of curiosity is your OS 32 or 64 bits ? If 64 do you observe the same issue using a 64 bits python ? I asking because there are some unresolved issues on 32 bits (never managed to find a machine to get to the bottom of them).

camerin, 2019-11-21:
I am using a 64 bit OS. i have tried both 64Bit python using the 64Bit DLL and 32 bit python using the 32 bit DLL. for both Agilent and NI. similar trace in all 4 cases. ends in "AttributeError: function 'viOpen' not found"

MatthieuDartiailh, 2019-12-03:
I must say I am running out of ideas here. If I come up with anything else I will let you know.