pyvisa/pyvisa #476: write_raw returns a tuple

pigrew, 2019-11-25:
The documentation comment of `MessageBasedInterface.write_raw` says it returns `int`, but it actually returns `Tuple[int,pyvisa.constants.StatusCode]`.

Could the documentation be updated to show it returning a tuple (since that's what `visalib.read(...)` returns)?

Thanks!

MatthieuDartiailh, 2019-11-26:
Returning a tuple does not line with the other methods of MessageBasedResource (including other write method). To me the current situation is a bug and the code not the documentation needs to be updated.

It will be backward incompatible but checking the number of characters written is not generally needed so hopefully it will only affect few people.

MatthieuDartiailh, 2020-07-23:
This is now fixed on master.

closed 2020-07-23 by MatthieuDartiailh