pyvisa/pyvisa #477: list_resources returns devices that have been disconnected

rahndnr, 2019-11-28:
I am using pyvisa 1.9.1 with Python 3.6.3 on a Windows 10 Enterprise x64 machine.
When I create a pyvisa.ResourceManager and call list_resources, the resulting listinstrs includes USBdevices that have been disconnected.  I get the same listinstrs even if all USB devices are disconnected.
Rebooting my machine does not solve the problem.
The error message reads as follows:

Traceback (most recent call last):
    self.dev = resmgr.open_resource("USB0::0<xxxxx::xxxx::xxx::0>::INSTR")
  File "C:\Python36\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Python36\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Python36\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Python36\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "C:\Python36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_NCIC (-1073807264): The interface associated with this session is not currently the controller in charge.

MatthieuDartiailh, 2019-11-28:
Can you please post the output of `pyvisa-info` ? I am wondering what backend you are using.

rahndnr, 2019-11-29:
>>pip pyvisa-info
ERROR: unknown command "pyvisa-info"
PS C:\aZhaoming\GoniCounter\v43> pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\python36\python.exe
   Version:        3.6.3
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct  3 2017 18:11:49 (#v3.6.3:2c5fed8)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377166368
         Spec. Version: 5244928
      #2: C:\windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377166368
         Spec. Version: 5244928

MatthieuDartiailh, 2019-11-29:
PyVISA simply reports whatever the VISA library reports. I remember that usually Keysight Connection Expert does not discard disconnected instrument automatically. One can remove them manually but I don't know if it can be configured to be automatic.

rahndnr, 2019-11-29:
How does one remove them manually?

MatthieuDartiailh, 2019-11-29:
In Keysight Connection Expert, the instruments are listed on the left hand side. When you click on one, details are displayed on the right and there is either a button or a menu to remove it (I don;'t remember exactly).

zackles, 2020-01-29:
I confirm having the same problem with visa from Keysight.
@rahndnr did you manage de remove them manually?

rahndnr, 2020-01-29:
Yes, manual removal works fine.

closed 2020-01-29 by MatthieuDartiailh