pyvisa/pyvisa #480: Connecting to Keithley2700 - empty list of resources

Dshisher, 2019-12-03:
Hi, I am pretty new to programming so I am not completely sure that my problem is from my use of pyvisa. 
I am trying to connect to Keithley2700 through Python for a long time and with a GPIB-USB-HS.

The simple code which i try to run is:

import visa
rm = visa.ResourceManager('@py')
print(rm.list_resources())

and i always receive an empty list.

The packages that i have installed to make it work are: 
pyvisa
pyvisa-py
and:
pyusb
libusb1
Which i am not sure that are really needed for the GPIB.

I have installed the driver of NI - Visa from there website and i can connect my device through "NI -Visa interactive control".

When I type: "python -m visa info":

Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 42 Stepping 7, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Programs\Python\Python37-32\python.exe
   Version:        3.7.3
   Compiler:       MSC v.1916 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 25 2019 21:26:53 (#v3.7.3:ef4ec6ed12)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

I will be glad to get the solution and to know what i did wrong...
If there is additional information needed I will upload.

MatthieuDartiailh, 2019-12-03:
GPIB drivers are required to run GPIB-USB adapters. In your case you should install the VISA implementation of the manufacturer of your adapter (I would guess National Instrument in your case). And you won't need to use `pyvisa-py` as a backend, simply use the default backend (no argument to the `ResourceManager`).

Dshisher, 2019-12-03:
Thanks for your answer.

I already have the NI driver: ni488k
I uninstalled the pyvisa-py and deleted the argument as you said and now i have an error:

Traceback (most recent call last):
  File "deviceVisa.py", line 12, in <module>
    rm = visa.ResourceManager()
  File "C:\Programs\Python\Python37-32\lib\site-packages\pyvisa\highlevel.py", l
ine 1553, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Programs\Python\Python37-32\lib\site-packages\pyvisa\highlevel.py", l
ine 1520, in open_visa_library
    wrapper = _get_default_wrapper()
  File "C:\Programs\Python\Python37-32\lib\site-packages\pyvisa\highlevel.py", l
ine 1494, in _get_default_wrapper
    raise ValueError('Could not locate a VISA implementation. Install either the
 NI binary or pyvisa-py.')
ValueError: Could not locate a VISA implementation. Install either the NI binary
 or pyvisa-py.

MatthieuDartiailh, 2019-12-03:
You not only need the driver but also the VISA library from NI that can be downloaded from http://www.ni.com/en-us/support/downloads/drivers/download.ni-visa.html#329456

Dshisher, 2019-12-03:
I already installed it, how do I verify that it is installed right?
do i need to add it somehow to path?

MatthieuDartiailh, 2019-12-03:
Did you restart at the end of the install ? Pyvisa should pick it up automatically but does not seem to do so.

Dshisher, 2019-12-03:
I have uninstall the NiVisa package - rebooted - installed it again - rebooted and still same results...

MatthieuDartiailh, 2019-12-03:
You can try to locate the visa.dll library on your system and pass the path to the resource manager. You could also try using a 64 bits version of Python. I have seen people experience issue on 32 bits.

Dshisher, 2019-12-04:
I have installed the python 64bit, put visa64.dll in path and now it seems to work.
Thank you.

closed 2020-07-23 by MatthieuDartiailh

itt-icc, 2022-11-10:
I’ve fixed it by install pyusb
_pip install pyusb_