pyvisa/pyvisa #481: Rigol DS1104z Issues

meawoppl, 2019-12-09:
I have been using a script with my older rigol to dump screen shots, after upgrading to the model in the title, I get this:
```
meawoppl@monolith:~/repos/rigol-grab$ python3 rigol_grab.py -v
('USB0::6833::1230::DS1ZC213101592::0::INSTR',)
Scanning: USB0::6833::1230::DS1ZC213101592::0::INSTR
Opening USB0::6833::1230::DS1ZC213101592::0::INSTR
/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA_py-0.4.0.dev0-py3.6.egg/pyvisa-py/protocols/usbtmc.py:119: UserWarning: Unexpected MsgID format. Consider updating the device's firmware. See https://github.com/pyvisa/pyvisa-py/issues/20
  warnings.warn('Unexpected MsgID format. Consider updating the device\'s firmware. See https://github.com/pyvisa/pyvisa-py/issues/20')
Traceback (most recent call last):
  File "rigol_grab.py", line 106, in <module>
    grabber.grab(filename=opts.filename, auto_view=opts.auto_view)
  File "rigol_grab.py", line 24, in grab
    buf = self.rigol().query_binary_values(':DISP:DATA? ON,0,PNG', datatype='B')
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA-1.11.0.dev0-py3.6.egg/pyvisa/resources/messagebased.py", line 715, in query_binary_values
    data_points, chunk_size)
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA-1.11.0.dev0-py3.6.egg/pyvisa/resources/messagebased.py", line 520, in read_binary_values
    chunk_size=chunk_size))
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA-1.11.0.dev0-py3.6.egg/pyvisa/resources/messagebased.py", line 356, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA_py-0.4.0.dev0-py3.6.egg/pyvisa-py/highlevel.py", line 398, in read
    ret = self.sessions[session].read(count)
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA_py-0.4.0.dev0-py3.6.egg/pyvisa-py/usb.py", line 128, in read
    USBTimeoutException)
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA_py-0.4.0.dev0-py3.6.egg/pyvisa-py/sessions.py", line 530, in _read
    current = reader()
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA_py-0.4.0.dev0-py3.6.egg/pyvisa-py/usb.py", line 108, in _usb_reader
    return self.interface.read(count)
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA_py-0.4.0.dev0-py3.6.egg/pyvisa-py/protocols/usbtmc.py", line 422, in read
    response = BulkInMessage.from_bytes(resp)
  File "/home/meawoppl/miniconda3/lib/python3.6/site-packages/PyVISA_py-0.4.0.dev0-py3.6.egg/pyvisa-py/protocols/usbtmc.py", line 117, in from_bytes
    msgid, btag, btaginverse = struct.unpack_from('BBBx', data)
struct.error: unpack_from requires a buffer of at least 4 bytes
```

The byte-dump that go into the "quirky" call look like the following:
```
b'Author\x00RIGOL\xe7\x88\x8c\xc9\x00\x00\x00\x11tEXtSource\x00DSZ series?Z\x1d\x97\x00\x00\x00)tEXtDescription'
b'\x00DSZ_Printing equipment output\xc1\xa55\xc1\x00\x00\x00\x1ctEXtCopyright\x00RIGOL TECHNO'
b'LOGIES\xe3\x14a\xaf\x00\x00\x88\xefIDATx\x01\xed\xbd+\x9c$\xc9\x91\xed\x9d\xa0A\x03\x81\x06\x03\n\x0c(0\xa0\x81@\x03\x81\x02\x02\x05\x06\x0cX\xd0`A\x81\x01\r\x04\x04\x07\x0eX\xd0`\x81'
b'>>~\xfcx^i\xc5\xde1\x12\x0e\xb5C\xf61\xf7b\r\xeci/"sq\xe3=\xde\xc6\x8c|\xd1l\xbc&N\xfa\x8b\x04,>I\xb1\xc8\xbc\n\xfe\x8e%\xe0R^\x85\x91\xab+x\xaeb\xe4\x1ad\x07\x81'
b'X\xbb\x99\x9d&\x92\x13\xa8]}\xfe\x92il\xfb/xo?=\xf3.\xe0\xf1\xad\xc9>\x0b\x01\xacBd\xad16.\xf7\x9f+GpA\xbf\xfc\x13\xef\xde/g{\xca\x0c\xcb\x8dv\xe2U\xd4\xda\x15\x11\x82\xdf\x10'
b'\x86\xd1@\x96\xb7k\x8dL\xf2c\n~\x8e\x93r\xb7\x0f\xc6P=\xf3]\xda\xff\xf8\xf7GH+\xb0\xc5+\xec]\xceQ\x93"\x81\xc8\xd5\xd7\xd8\xf6/\x02\x8b\xcd\xdd\xdd\xe5\x1eT.u\xa9\xe2\xfd\'\x7f\x0b*:\xd9'
b"\xf8^`\xd9\xf3!\xffS\x9f|b\x94t0*\xb0 \x9ex\xe0\x1bW\x10R\xfe\x93>|\x13\x0b\x1e\xf8\xfdG\x84\xb5'X\xfe_&~\xf7\xdb\x97V\x8b\x03\xb0:{`\xc6ok\x99\x13\x7f\x82\x10\xff\xba\x92\xc7"
```

I am happy to fix/pr this, but I don't know well enough the history around the rigol quirk.  

Much thanks.

meawoppl, 2019-12-09:
It looks like occasionally it returns empty byteblocks during the read, if i have it ignore those, it permutes further forward, and eventually hits a read timeout that i don't understand.

MatthieuDartiailh, 2019-12-11:
Could you post the logs when you ignore empty blocks ? I must say I never got a good occasion to dive into the that portion of the code (and get a setup on which I could experiment in depth) so anything you can dig up is interesting.

unixneo, 2020-01-19:
I have found that using the LAN (ethernet) port results in less corrupted screen capture images when using rigol_grab.py

See for example, this macOS Catalina discussion from Jan 2020:

https://www.unix.com/programming/283575-python-screen-capture-rigol-1054z-macos-catalina-using-ni-visa.html

tai, 2022-10-20:
I'm also hit by this (DS1054Z returning empty response). Since Rigol is known for broken USBTMC/VISA implementation (seems LXI over LAN is somewhat more stable), I'm not sure how I can fix it.

However, I found out that following library built directly on top of python-usbtmc library can successfully fetch screenshot image (doc on the site is wrong, but library works): https://github.com/jtambasco/RigolOscilloscope

The most interesting part of the code (in rigol1000z.py) is following:

`raw_img = self._read_raw(3850780)[11:-4]`

So it avoids chunked read and just read whole buffer and then strip off some bytes on both ends.

I tried to follow this style by specifying large chunk_size in pyvisa query_binary_values() call, but so far, it's not working. Hope someone can figure the issue out as most tools seem to use pyvisa/pyvisa-py instead of going direct with python-usbtmc.

tai, 2022-10-20:
I've figured out a minimal code to get DS1000Z working (with usbtmc.py):
https://gist.github.com/tai/408053a0955bf1712a5809240aa5544f
https://github.com/python-ivi/python-usbtmc/pull/62

This might help to get it working with pyvisa as well.

yaqwsx, 2023-03-21:
I also wanted to you PyVISA with Rigol DS1054Z, and I ran into this issue.

After closer inspection, I discovered that sometimes the scope gets into a state where it sends messages chunked even when they aren't supposed to be. Therefore, it helps to lower the chunk size to `64 - header_size` and also, wait for a little before requesting the next chunk.  I hope the others will find this information useful.

I tested these modifications with pyvisa-py. I was able to get a reliable connection. Nevertheless, it hurts the performance a little. Bellow, I attach the patch 

```
--- pyvisa_py/protocols/usbtmc.py	2023-03-21 07:10:56.486997585 +0100
+++ pyvisa_py/protocols/usbtmc.py	2023-03-21 07:14:51.747924805 +0100
@@ -299,7 +299,7 @@
 
 class USBTMC(USBRaw):
     # Maximum number of bytes per transfer (for sending and receiving).
-    RECV_CHUNK = 1024**2
+    RECV_CHUNK = 64 - 12
 
     find_devices = staticmethod(find_tmc_devices)
 
@@ -489,5 +489,5 @@
             # Detect EOM only when device sends all expected bytes.
             if len(response.data) >= response.transfer_size:
                 eom = response.transfer_attributes & 1
-
+            time.sleep(0.0001)
         return bytes(received)
```

@tai: I wasn't able to replicate your success with directly writing to `/dev/usbtmcN`.