pyvisa/pyvisa #482: pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

jenhaoyeh, 2019-12-10:
I am trying to use pyvisa to connect a Keysight Network Analyser (E5071C). I can successfully do that in one Dell laptop with CentOS 7.5. But I cannot do that in a Intel NUC with CentOS 7.7, where I get this error:

```
>>> import visa
>>> rm = visa.ResourceManager()
>>> rm.list_resources()
('ASRL1::INSTR', 'ASRL2::INSTR', 'ASRL3::INSTR', 'ASRL4::INSTR', 'TCPIP0::169.254.158.139::A-E5071C-15158::INSTR')
>>> rm.open_resource('TCPIP0::169.254.158.139::A-E5071C-15158::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jenhao/.local/share/virtualenvs/vna-multiplexer-zkou9k9q/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/jenhao/.local/share/virtualenvs/vna-multiplexer-zkou9k9q/lib/python3.6/site-packages/pyvisa/resources/resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/home/jenhao/.local/share/virtualenvs/vna-multiplexer-zkou9k9q/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/jenhao/.local/share/virtualenvs/vna-multiplexer-zkou9k9q/lib/python3.6/site-packages/pyvisa/ctwrapper/functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "/home/jenhao/.local/share/virtualenvs/vna-multiplexer-zkou9k9q/lib/python3.6/site-packages/pyvisa/ctwrapper/highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.
```

Here are the "python -m visa info" for both cases. In the successful one, it is:
```
python -m visa info
Machine Details:
   Platform ID:    Linux-3.10.0-862.3.3.el7.x86_64-x86_64-with-centos-7.5.1804-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/lab/.local/share/virtualenvs/vna-multiplexer-BYfvy9Wc/bin/python
   Version:        3.6.8
   Compiler:       GCC 4.8.5 20150623 (Red Hat 4.8.5-39)
   Bits:           64bit
   Build:          Aug  7 2019 17:28:10 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: /usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
```

In the failed one, there are two backends because I tried installing pyvisa-py as well (unfortunately both not working.)
```
python -m visa info
Machine Details:
   Platform ID:    Linux-3.10.0-1062.4.3.el7.x86_64-x86_64-with-centos-7.7.1908-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/jenhao/.local/share/virtualenvs/vna-multiplexer-zkou9k9q/bin/python
   Version:        3.6.8
   Compiler:       GCC 4.8.5 20150623 (Red Hat 4.8.5-39)
   Bits:           64bit
   Build:          Aug  7 2019 17:28:10 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.19.2.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 19923456
         Spec. Version: 5244928
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

Any suggestion is appreciated. Thanks.

MatthieuDartiailh, 2019-12-11:
This sounds crazy. Out of curiosity if you use pyvisa-py can you open the connection ? Can you ping the resource from the second machine ?

I must say I don't have a good feeling what may cause this issue. The specification implemented is more recent on the machine it is not working but I find that surprising it would cause such an issue in particular if you can list the resource. On Linux you can also give a shot to Keysight VISA.

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh

abhishek123963, 2020-12-21:
I am trying to use pyvisa using python version 2.7   in ubantu version 16.04 to communicate with keithley 2604B sourcemeter and keithley 2002 digital multimeter .now getting this error...please help me ..

Traceback (most recent call last):
  File "smsa.py", line 18, in <module>
    resources = rm.list_resources('@py')
  File "/usr/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 238, in list_resources
    raise errors.VisaIOError(errors.StatusCode.error_resource_not_found.value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

here is information about -m visa info
Machine Details:
   Platform ID:    Linux-4.4.0-197-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.12
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Oct  5 2020 13:56:01 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.0.1)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0b2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         libgpib.so.0: cannot open shared object file: No such file or directory
      TCPIP SOCKET: Available

MatthieuDartiailh, 2020-12-24:
Python 2.7 is not supported anymore please update to Python 3 and latest Pyvisa and Pyvisa- py versions.

Sourc3w, 2021-07-01:
> Python 2.7 is not supported anymore please update to Python 3 and latest Pyvisa and Pyvisa- py versions.

Hi Matthieu, Im a big fan of You since Im working with PyVisa, I have the same problem, that this guy, I having a lot of problems to connect PyVisa with scope, I have used Pyvisa with Keithley, TTi but with Teledyne Waverunner 6100 is imposible for me. The only method I have get connection is with that:
`import win32com.client #import the pywin32 library
scope=win32com.client.Dispatch("LeCroy.ActiveDSOCtrl.1") #creates instance of the ActiveDSO control
scope.MakeConnection("IP:192.168.100.115") #Connects to the oscilloscope.  Substitute your IP address 
scope.WriteString("VBS app.Measure.ShowMeasure = true",1) #Automation command to show measurement table
#scope.WriteString("""VBS 'app.Measure.P1.ParamEngine="Mean" ' """,1) #Automation command to change P1 to Mean
scope.WriteString("VBS? 'return=app.Measure.P1.Out.Result.Value' ",1) #Queries the P1 parameter
value = scope.ReadString(80)#reads a maximum of 80 bytes
print (value) #Print value to Interactive Window
scope.Disconnect() #Disconnects from the oscilloscope`

But pref Pyvisa.
Best regard man, I really really am a big fan of You, You are very expert.

zuchiko88, 2022-03-04:
> Closing for lack of feedback feel free to reopen if necessary.

Is it possible to reopen #482 by jenhaoyeh?
"pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system."

I am experiencing a similar issue with the same exact instrument E5071C Network Analyzer by Agilent Technologies.

Here is the result of pyvisa-info:

UserWarning: GPIB library not found
. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details: 
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\zuchiko88\AppData\Local\Programs\Python\Python310\python.exe
   Version:        3.10.2
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan 17 2022 14:12:15 (#tags/v3.10.2:a58ebcc)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.2
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      TCPIP INSTR: Available
      TCPIP SOCKET: Available

I am basically just trying to first communicate with the instrument and retrieve the device information via TCPIP:

import pyvisa

rm = pyvisa.ResourceManager()

inst_ip = "TCPIP0::192.168.100.101::INSTR"
inst = rm.open_resource(inst_ip)
inst.timeout = 5000 # ms

inst_info = inst.query('*IDN?').split(',')
print("Test Instrument Make:", inst_info[0])
print("Test Instrument Model:", inst_info[1])
print("Test Instrument SN:", inst_info[2])

inst.close()

MatthieuDartiailh, 2022-03-06:
Couple of questions: 
- can you ping the IP of the instrument ?
- since you are on Windows with NI-VISA installed did you try looking for your instrument in NI-MAX ?

zuchiko88, 2022-03-08:
> Couple of questions:
> 
> * can you ping the IP of the instrument ?
> * since you are on Windows with NI-VISA installed did you try looking for your instrument in NI-MAX ?

Nevermind, I just got it to work.
Yes, I was able to ping the IP.
The code was fine as well; it was the Network Analyzer that was not configured properly to communicate with the host.
I basically just navigated through the Network Analyzer settings: System > Misc Setup > Network Setup > turn on: SICL-LAN Server

Thank you for the quick response though! I might be posting another issue related under the same instrument if I don't figure it out soon.

THEliberator03, 2023-01-12:
> I basically just navigated through the Network Analyzer settings: System > Misc Setup > Network Setup > turn on: SICL-LAN Server

I'm having the same issue with the E5071C Network Analyzer but turning on the SICL-LAN option does not seem to solve the problem, is there any other configuration you think it might help?
As of now i have this network configuration in my device:
- Telnet Server ON
- SICL-LAN Server ON
- SICL-LAN Address 17
- Web Server OFF

MatthieuDartiailh, 2023-01-12:
@THEliberator03 same questions:

- can you post the output of pyvisa-info ?
- can you ping the IP of the instrument ?
- since you are on Windows with NI-VISA installed did you try looking for your instrument in NI-MAX ?

THEliberator03, 2023-01-12:
- I haven't been able to use the pyvisa-info command (I'm fairly new to using this module and working on Ubuntu with pip), but this is the info that pip show gives me:
  - Name: PyVISA-py
   Version: 0.6.0
  - Name: PyVISA
Version: 1.13.0
- I have no problem doing a ping to the device from the local network.
- As i said in the first answer I'm working on Ubuntu 22.04, but even if NI-VISA is also available on Linux distros I'm not using it.

For what I've read in this issue thread I think this is a problem related to this specific device and the way it works to be controlled by LAN, not something wrong with Pyvisa.

iblackford, 2023-08-30:
Hi Matthieu, 

After much searching on the internet I came across this thread, and I think it's probably my best hope at trying to get my setup working. 

I have a lecroy Waverunner 104MXi that I'm trying to get working with pyvisa. I'm running the following versions:

PyVisa: 1.13.0
PyVisa-py: 0.70
Python; 3.8

I can't seem to run pyvisa-info, maybe there's some path arguments or something I have to set in order to make this work? i'm running Win10. 

Here's my code:
    rm2 = pyvisa.ResourceManager('@py')
    resources = rm2.list_resources('?*')
    print(resources)
    my_instrument = rm2.open_resource('TCPIP::169.254.67.225::INSTR')

My problem is that I can see the device in the resource list, but I get a slew of errors when trying to open the resource:

('TCPIP::192.168.229.140::INSTR', 'TCPIP::169.254.67.225::INSTR')
Traceback (most recent call last):
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\tcpip.py", line 456, in after_parsing
    self.interface = Vxi11CoreClient(host_address, port, self.open_timeout)
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\tcpip.py", line 360, in __init__
    rpc.TCPClient.__init__(self, host, prog, vers, open_timeout)
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\protocols\rpc.py", line 826, in __init__
    RawTCPClient.__init__(self, host, prog, vers, port, open_timeout)
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\protocols\rpc.py", line 452, in __init__
    self.connect(1e-3 * open_timeout)
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\protocols\rpc.py", line 486, in connect
    raise RPCError("can't connect to server")
pyvisa_py.protocols.rpc.RPCError: can't connect to server

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\pythonProject\main.py", line 22, in <module>
    my_instrument = rm2.open_resource('TCPIP::169.254.67.225::INSTR')
  File "C:\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 3284, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\pythonProject\venv\lib\site-packages\pyvisa\resources\resource.py", line 282, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "C:\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 3209, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\tcpip.py", line 85, in __new__
    return newcls(resource_manager_session, resource_name, parsed, open_timeout)
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\sessions.py", line 321, in __init__
    self.after_parsing()
  File "C:\pythonProject\venv\lib\site-packages\pyvisa_py\tcpip.py", line 458, in after_parsing
    raise errors.VisaIOError(constants.VI_ERROR_RSRC_NFOUND)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

Thanks in advance for any help that you or others may be able to provide, i'm at a bit of a loss here on where to start. :(

arr-ee, 2023-08-30:
@iblackford can you connect to the scope using official software? What about pyvisa + NI VISA?

arr-ee, 2023-08-30:
I _think_ the issue might be that many LeCroy scopes use VICP instead of VXI-11. pyvisa-py does support it by requires zeroconf and pyvicp installed + using VICP instead of TCPIP prefix (this should come up in list devices with those libs installed).

additionally/alternatively you might want to check out https://github.com/glscopeclient/scopehal-apps if you haven’t already. It’s a very different type of project, but afaik it’s quite functional, and author has at least one LeCroy scope so it’s more likely to be working.

iblackford, 2023-08-30:
Wow, changing the prefix to VICP instead of TCPIP fixed it! I don't see my device show up in resources with the VICP prefix, but if I manually set it to VICP, it works! 

My code:
          import pyvisa
          
          if __name__ == '__main__':
              rm2 = pyvisa.ResourceManager('@py')
              resources = rm2.list_resources('?*')
              print(resources)
              my_instrument = rm2.open_resource('VICP::169.254.67.225::INSTR')
              print(my_instrument.query('*IDN?'))

The results:
              ('TCPIP::192.168.229.140::INSTR', 'TCPIP::169.254.67.225::INSTR')
              *IDN LECROY,WR104MXI,LCRY0612N21448,8.1.0

Thanks, I can't wait to start playing with other commands to the scope.

vikranth202, 2024-01-04:
Hi I am trying to interact with yokogawa6370D optical spectrum analyser using pyvisa (1.10.1 version) on python3 

my code :
import pyvisa as visa
AQ6370D= visa.ResourceManager('@py').open_resource('TCPIP::192.168.241.245::INSTR')
output : VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

now instead of using TCPIP i used VICP but now I get 
output :  TimeoutError: timed out

I am able to ping the device in command prompt

vikranth202, 2024-01-04:
Hi I am trying to interact with yokogawa6370D optical spectrum analyser(OSA) using pyvisa (1.10.1 version) on python3
The OSA is over SOCKET protocol
my code :
import pyvisa as visa
AQ6370D= visa.ResourceManager('@py').open_resource('TCPIP::192.168.241.245::10001::SOCKET')
print (AQ6370D.query("*IDN?"))

I get  the following : 
error: [Errno 10054] An existing connection was forcibly closed by the remote host

I am able to ping the device in command prompt. I tried introducing a sleep command for 3seconds before the print command but didnot help.

LongnoseRob, 2024-01-04:
Looking at the [programming manual](https://cdn.tmi.yokogawa.com/1/6057/files/IMAQ6370C-17EN.pdf) page 28ff, it looks like that some authenfication might be required.
Also the device seems to be able to support VXI-11, so I wouild recommend to confirm the settings on the device beside the IP adress.

Also it would be better to open a new issue with this problem in stead of "recycling" this old issue