pyvisa/pyvisa #484: Agilent 8164A Data issue

DrSaucyRossy, 2019-12-18:
Hello, 

Thanks in advance, I would just like to start off by saying if this post is somehow against the code of conduct or outside the purview of this thread I apologize and I will delete the post. I am using an Agilent (now Keysight) 8164A to measure some devices. I am running my code using spyder with anaconda 4.80 and python 3.7.4. The following is my code for running a wavelength sweep:

import visa
import time
import numpy as np
import pandas as pd
from scipy.signal import savgol_filter
from datetime import datetime
import matplotlib as plt

def plot(x,y):
    plt.plot(x,y)
    plt.x_label('Wavelength (nm)')
    plt.y_label('Power (W)')
    return plt.show

scan_speed = 5  # nm/s only 0.5 5 40 allowed
avg_time=1E-4 # photodiode average time
wavelength_start = 1620
wavelength_step = 0.1
wavelength_finish = 1640 + wavelength_step
wavelength = np.arange(wavelength_start,wavelength_finish,wavelength_step) # nm

rm = visa.ResourceManager()
Agilent_8164A = rm.open_resource('GPIB0::20::INSTR')
Agilent_8164A.write('*cls') # instrument setting reset
points = len (wavelength)
Agilent_8164A.write('trig:conf LOOP') # an output trigger is autometically works as input trigger
Agilent_8164A.write('TRIG1:OUTP DIS') # PD output trigger is disabled
Agilent_8164A.write('TRIG1:INP SME') # PD will finish a function when input trigger is abled
Agilent_8164A.write('TRIG0:OUTP STF')# TLS will send a output trigger when sweep starts (input trigger generated)
Agilent_8164A.write('TRIG0:INP IGN') # (TLS input trigger is ignored)

# sensor setting
Agilent_8164A.write('init1:cont 1')# continuous detection mode
Agilent_8164A.write('sens1:pow:atim 1MS')# set the averagetime to 1ms for sensor 2
Agilent_8164A.write('sens1:pow:rang:auto 0')#set auto ranging off
Agilent_8164A.write('sens1:pow:rang -10DBM')#sense range (maximum power that can be detected)
Agilent_8164A.write('sens1:pow:unit 1')# set the unit of power: 0[dBm],1[W]
Agilent_8164A.write('sens1:pow:wav 1550nm') # set senser wavelength centered at 1550 nm

# tunable laser setting
Agilent_8164A.write('outp0:path high')# choose which path of tuable laser. output1 [low power high sens] output2 [high power]
Agilent_8164A.write('sour0:pow:unit 0')# set source power unit
Agilent_8164A.write('sour0:pow 0')#sset laser power {unit will be according to the power unit set before}
Agilent_8164A.write('sour0:AM:stat OFF')
#continuous sweep setting
Agilent_8164A.write('wav:swe:mode CONT')
Agilent_8164A.write('wav:swe:spe ' + str(scan_speed) + 'nm/s') # only 0.5 5 40 allowed
Agilent_8164A.write('wav:swe:star ' + str(wavelength_start) + 'nm')
Agilent_8164A.write('wav:swe:step ' + str(wavelength_step) + 'nm')
Agilent_8164A.write('wav:swe:stop ' + str(wavelength_finish) + 'nm')
Agilent_8164A.write('wav:swe:cycl 1') #Set the number of cycles
Agilent_8164A.write('sens1:func:par:logg ' + str(points) + ',' + str(avg_time)) # Sets the number of data points and the averaging time for the logging data acquisition function
Agilent_8164A.write('sens1:func:stat stab,star') #Enables Stability data acquistion and starts data acquistion
Agilent_8164A.write('wav:swe STAR') # Start wavelength sweep
Agilent_8164A.write('sour0:wav:swe:llog 1') #Logging the wavelength during the sweep
time.sleep(10); #Allows for Agilent scan to run otherwise
#values = Agilent_8164A.query_binary_values('sour0:read:data? llog', datatype='d', is_big_endian=False, datapoints = points)# get wavelength data from logging operation
power = Agilent_8164A.query_binary_values('sens1:func:res?'); #get power data
P = savgol_filter(power,11,3) # Smooths data
Agilent_8164A.write('sens1:pow:rang:auto 1') #set auto ranging on
Agilent_8164A.write('TRIG1:INP IGN') # PD will finish a function when input trigger is enabled
Agilent_8164A.write('sour0:wav 1620nm') # set the input wavelength back to 1550 nm for the next measurment
data = np.vstack((wavelength, power))
data = np.transpose(data)
plot(data[:,0],data[:,1])
x =  'Sweep '
timing = datetime.now()
y = timing.strftime("%d-%m-%Y %H--%M--%S")
z = '.csv'
pd.DataFrame(data).to_csv(x+y+z , header = ['Wavelength (nm)','Power (W)'], index = None)

My problem is two-fold. The 1st and bigger problem is when I try to get the wavelength data without using data_points I get the following error:

 'FutureWarning: Reading binary data without a known length is error prone, and should be avoided. This capability will be removed in future versions.
If the instrument does not report the length of the block as part of the transfer, it may be because the size is fixed or can be accessed from the instrumentin using a specific command. You should find the expected number of points and pass it using the `data_points` keyword.'

When I  include data_points I get a timeout error and I am not sure as to why this is the case. I originally had this code written in Matlab where it worked just fine for this problem. I have worked around it by just creating an array with what the wavelength and using that for my x-data but would prefer to get the actual data from the Agilent.

The second problem is one that existed with the Matlab code which is if my sweep includes more than 201 points the sweep fails, I have tried working around this by turning this into a function which if my sweep is greater than 201 points I cut it up into smaller sizes running the sweep multiple times and appending the data but this again causes the sweep to crash. This may just be a problem with the Agilent itself and so there is no solution here but I thought I should ask just in case.

DrSaucyRossy, 2019-12-18:
I sorted out my second problem by altering my code by altering time.sleep(10) to:
time.sleep(2*wavelength_stop-wavelength_start)/scan_speed)

The problem I was running into the sweep was still running while I was trying to collect data.

MatthieuDartiailh, 2019-12-18:
First the warning should not interrupt your program so my first question is what result do you get when the warning show up. Second could you post what you obtain if you attempt to only read the first few bytes of the instrument response using `read_bytes`. I am surprised that an ieee header does not contain the length.

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

DrSaucyRossy, 2021-01-29:
I recently reran the code to try and figure out what the problem was. My mistake was including the data_points keyword argument. I removed this and it ran fine.

closed 2021-01-29 by DrSaucyRossy