pyvisa/pyvisa #485: Connection closed by Host on movein call

gmnewma, 2020-01-06:
I am using pyvisa 1.9.1 with Python 3.6.3 on a Windows 10 Enterprise x64 machine.

On my development machine I am able to use the move-in function to transfer Spectral Dynamics digitizer memory data from the device to a python variable.  The data I receive from this function is correct.    When I move this code to another machine a console error is displayed and logged to the agilent logger software:

error 10053 - connection closed by host.

The only information found on what can cause this is the operating systems Keep Alive value is causing the TCP IP connection to close.  I don't believe this is the cause of the problem because on the problematic machine I can obtain the identity of the hardware using the appropriate VISA register commands.

Please let me know what steps might help resolve this issue.

Thank you,
Gordon

MatthieuDartiailh, 2020-01-08:
To be honest I never used register based resources so I am not sure I will be able to help a lot. Out of curiosity, are you using the same version of the VISA library on each computer ? I am assuming you are running Keysight VISA on both, correct ? (I say Keysight since the part of Agilent dealing with VISA is Keysight now)
Do you see the same behavior no matter the size of the memory block you are attempting to transfer ? Since you can read the identity, I am wondering if you may be running in some kind of timeout like error.
Can you test transferring the data using the Keysight Connection Expert (I don't know what can be done with it for register based instruments) ? If you experience the same issue using Keysight software you can try contacting their support, since it will mean the issue is outside of pyvisa.

gmnewma, 2020-01-09:
Hi Matthieu,

Are you using the same version of the VISA library on each computer?  
- We originally had 1.9.1 on both machines then we tried 1.10.1 on the problematic machine with the same results.

Do you see the same behavior no matter the size of the memory block you are attempting to transfer?
- Yes, we have the same behavior regardless of the length to be retrieved.

Can you test transferring the data using the Keysight Connection Expert?
-I appears to only support message-based commands.

I provided bad information in the original post.  The development machine that works is a Windows 7.  The machines that do not work are Windows 10 platforms.

We have a C program that uses the VISA libraries that works on both the Win 7 and Win 10 computers.  This is the reason we believe that pyvisa is the source of our problem.

"VI_ERROR_CONN_LIST - connection for the given session has been lost"

Source file generating exception: ctwrapper\functions.py line: 1013 in move_in_16

Using VISA I/O monitor we tracked down the exception stack to the following: 
VISA::viMoveIn16 -> SICL::iblockmove()

The attached files c-visa-iblockmovex.png (a successful move using C program) and pyvisa-script-iblockmovex.png (an unsuccessful move using python) illustrate that the problem appears to be 32 bit vs. 64 bit addressing.  

We have tried the following in the constructor call to the resource manager:
     rm = ResourceManager( 'c:/windows/visa32.dll')
     rm = ResourceManager( 'c:/windows/visa64.dll')

I am assuming you are running Keysight VISA on both, correct?
- On the Win 7 machine we are using Agilent Version 16.3.  On the Win 10 computer we originally had a version of the keysight library.  When unable to use the move_in function we installed the Agilent Version 16.3 on the Win 10 hardware hoping that this would fix the problem.  The same problem occurs using pyvisa with either the Keysight or Agilent software, but the C program worked with both., 

Thank you,
Gordon.

gmnewma, 2020-01-09:
could not attach the files but the C log shows all attributes as 32-bit.  The pyvisa log shows the following attributes as 64-bit: count, dest_offset, src_handle, src_offset

gmnewma, 2020-01-09:
When I installed a 32-bit version of python then move_in worked as expected.  I'm hoping that this could be considered a problem to be fixed, if not then I will have to change other libraries - numpy 64-bit will not event install with pip in the 32-bit environment.

Thank you,

Gordon

MatthieuDartiailh, 2020-01-09:
Thanks for the detailed testing. Could you try sending me the logs by email or posting them to a gist that you would share ?
Is your C program compiled and linked against the 32bit or 64bit version of VISA ? If 32 bit, could you confirm it still works when compiled against the 64 bit version ? It is simply that since a 64 bit Python cannot use a 32 bit library I would like to make sure that the 64 bit version of visa does work as expected.
I am not yet sure where the problem exactly is but I sure hope we will be able to fix it.

gmnewma, 2020-01-15:
When setting the extended=True I received a different error: VI_ERROR_NSUP_OPER (unsupported operation).  The logged information had the error occurring in the Agilent VISA VISA:viMoveIn16Ex.  The parameters found in the log are as follows:
buf: out: uint16
length: in: uint64 (6144)
offset: in: uint64 (48)
space: in: uint16 (1)
status: ret: int32
vi: in: uint32 (1)

The input parameter values are correct.  Not sure why space and the vi session are not 64-bit.

Our plan right now is to reach to Agilent/Keysight and see if we can get help from them.  I'm going to start creating a 32-bit python version of our application.   Please let me know if you have any more thoughts on the MoveIn problem.

Thanks

MatthieuDartiailh, 2020-01-15:
Using extended was a cheap way to to fix the definition of ViBusAddress without touching the code. The proper fix is to go into ctwrapper/types and make ViBusAddress the same as ViBusAddress64.

MatthieuDartiailh, 2020-01-15:
Could you try the above ?

gmnewma, 2020-01-15:
Change to:
ViBusAddress, ViPBusAddress = _type_pair(ViUInt64)
is this correct?

gmnewma, 2020-01-15:
Change to:
 ViBusAddress, ViPBusAddress = _type_pair(ViUInt64)
This had no obvious effect.  I can still do the identity command but still get the connection lost error on the MoveIn call.

MatthieuDartiailh, 2020-01-15:
That was the one I had in mind. Do you see any difference in the low-level reports ?

MatthieuDartiailh, 2020-01-15:
Actually ViBufSize requires the same fix:
```python
ViBusAddress, ViPBusAddress = _type_pair(ViUInt64)
ViBusSize     = ViUInt64
```
Can you check ? And sorry for the lack of details earlier I was on mobile.

gmnewma, 2020-01-15:
I checked just now and have the same connection lost error.  

The SICL.imoveblockex log entry shows a value for src_handle of 2^64 - 1.  

Thank you working on the issue with me.

MatthieuDartiailh, 2020-01-16:
Ok at this point Keysight is your best bet. I am convinced that the above mentioned types are indeed wrong on 64 bits system but if it is not enough to fix the issue, I am at a loss since that makes us conformant with the VISA spec. I have some contacts at Keysight so let me know if you do not manage to get through, I could give it a try too.

MatthieuDartiailh, 2020-01-28:
any update @gmnewma ?

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh