pyvisa/pyvisa #487: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

mra7072, 2020-01-16:
def setup():
        rm=pyvisa.ResourceManager()
        resources = rm.list_resources()
        print(resources)
        print(rm)
        if resources == None:
            print("No connected devices found")
    
        DMM = rm.open_resource(resources[0],
            baud_rate=9600, data_bits=8, flow_control=4,
            parity = Parity.none,stop_bits=StopBits.two)

        #clear buffers
       
        #set read/write terminators to dictate end of command"
        DMM.read_termination = '\n'
        DMM.write_termination = '\n'
        DMM.write("*rst; status:preset; *cls")
        #Enable remote mode to interface with multimeter
        #DMM.write("*IDN?")

        print("Connection Established with " + DMM.read())
        return DMM

MatthieuDartiailh, 2020-01-19:
You can try using the pyvisa-py backend on your windows 10 machine to check if the issue is specific to the Pi. According to some references (https://hallard.me/enable-serial-port-on-raspberry-pi/) using the serial port on the Pi can require a bit of work too.

mra7072, 2020-01-22:
Yes, so I tried that and it allowed me to deduce the issue. I needed to add some delays otherwise it was timing out on the pi.

mra7072, 2020-01-22:
Thank you again for your help!

MatthieuDartiailh, 2020-01-22:
Great you could figure out a solution !

closed 2020-01-22 by MatthieuDartiailh