pyvisa/pyvisa #488: pyVISA: "PyUSB does not seem to be properly installed"

luzdal, 2020-01-16:
Hello!

Trying to connect to a Keithley 2280S-60-3 device via USB. I encounter the seemingly common problem of initially getting a "()" shown in the screenshot below. I've found other threads that talk about this problem but end up working it out by using other programs instead, but I really need to get this working through pyvisa which in theory should be possible.

[image]

So I tried **.get_instrument**:
[image]
Which gives the following error:
<span style="color: red">VisaIOError</span>: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

Skimmed through the pyvisa [documentation](https://readthedocs.org/projects/pyvisa/downloads/pdf/latest/#page=33&zoom=100,96,96) which mentions something about "configuring the backend" (sect 1.1.2). Not sure what that means exactly but this is what I think they suggest is to try **.ResourceManager('@py')**:

[image]
Which gives the new error:
ValueError: PyUSB does not seem to be properly installed.
Please refer to PyUSB documentation and 
install a suitable backend like 
libusb 0.1, libusb 1.0, libusbx, 
libusb-win32 or OpenUSB.
No backend available

This is my pyvisa-info:
[image]

I also tried **pip install pyusb** PyUSB "[documentation](https://github.com/pyusb/pyusb)" which returns:
Requirement already satisfied: pyusb in /Users/air/anaconda3/lib/python3.6/site-packages (1.0.2)

SOS!

luzdal, 2020-01-16:
Realized that having NI VISA simultaneously communicating to the Keithley USB device will prevent pyvisa from reading it. Closed NI VISA and now pyvisa is showing the device with **rm.list_resources()**

closed 2020-01-16 by luzdal

MatthieuDartiailh, 2020-01-16:
Happy you made it work !

ScotchWhiskey, 2022-09-21:
Any answer to the above question. I think I have the same problem!

MatthieuDartiailh, 2022-09-22:
What happens is that is NI-VISA claims the USB interface, pyusb does not see it anymore. You probably need to stop all background process from NI-VISA.

Honestly if you have NI-VISA installed and running just use the ivi backend. It is much more stable and using the python backend will not bring you anything.

joctavio287, 2023-02-06:
Hello, I'm having the same problem but I don't know how to stop NI-VISA

MatthieuDartiailh, 2023-02-06:
As I mentioned in my earlier answer if you have NI-VISA installed and running simply use it rather than trying to use pyvisa-py. The conflict handling for USB resources does not make it worthwhile.

Julesghes, 2024-05-16:
Hey 
I installed NI-VISA but I still have this problem. 
it is to control Tektronic afg3102
do you any idea ?

MatthieuDartiailh, 2024-05-16:
Please open a new issue following the template so that I have enough information to answer.