pyvisa/pyvisa #489: Support of UART to RS232 connection of Raspberry Pi 4

mra7072, 2020-01-22:
I'm connecting a rpi4 to a agilient 34401A multimeter. Currently, I'm able to connect from USB connection from pi to RS232 connection to the multimeter just fine. I would like to migrate to using UART on the pi to RS232. Would this be possible? 

Would the resource manager be able to detect the connection?

MatthieuDartiailh, 2020-01-22:
Usually adapters are meant to be transparent so no matter the starting port it should look like just a normal serial port from Python so I would expect it to just work.

mra7072, 2020-01-22:
Sweet!! thanks a lot!! hope it works, ill update you. I ordered the connector... PYVISA is awesome, helping with my senior design project in unversity!! thanks a lot for your support

MatthieuDartiailh, 2020-01-29:
Let me know how it works !

mra7072, 2020-01-29:
Got it to work, so for UART i had to specifiy ASLR/dev/ttyS0 as the resource name explicity in the open_resource() method. When I list resources, I'm unable to see it as an actual resource, would you know why? It works regardless. Also, I noticed now that I'm using UART to RS232 compared to USB to RS232, I can no longer query a command without it timing out. I must write the command, add some delay, then read. So this kind of slows down things unfortunately for me. Is there any way to get around this?

MatthieuDartiailh, 2020-02-02:
I don't know enough about UART to help I fear. Regarding the listing issue you can try asking on the pyserial bugtracker.

closed 2020-07-23 by MatthieuDartiailh