pyvisa/pyvisa #490: pyvisa very slow with TCPIP

nbenm, 2020-01-27:
I use pyvisa to access Rigol DS1054Z scope data.

If I use the following code, it is **extremely slow:**

```
rm = visa.ResourceManager('@py')
scope = rm.open_resource('TCPIP0::192.168.1.24::inst0::INSTR')
```
If I use the following code, it's much better but **still slow**:

```
rm = visa.ResourceManager('@py')
scope = rm.open_resource('TCPIP0::192.168.1.24::5555::SOCKET')
```
But if I use usb with the following code, it is very fast:

rm = visa.ResourceManager('@py')
scope = rm.open_resource('USB0::0x1AB1::0x04CE::DS1ZA182410617::INSTR')
Commands that are sent are very simple. They are in a loop to access a great quantity of data:

```
:WAV:STAR start_value
:WAV:STOP end_value
:WAV:DATA?
```
I get ascii data.

I must add that sending these commands with netcat (nc) is the fastest way to get data.

**My question:**

Is there a way to speed up TCPIP access?

MatthieuDartiailh, 2020-01-29:
Could you post the output of `pyvisa-info` for reference ? 

I am not sure why the TCPIP backend would be so slow. Could you try profiling the code execution ?

nbenm, 2020-01-29:
@MatthieuDartiailh  thanks for your response.
Here's the pyvisa-info result:

```
Machine Details:
   Platform ID:    Darwin-19.2.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /usr/local/opt/python/bin/python3.7
   Version:        3.7.6
   Compiler:       Clang 11.0.0 (clang-1100.0.33.16)
   Bits:           64bit
   Build:          Jan  3 2020 13:56:27 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 32, 64
         Vendor: National Instruments
         Impl. Version: 16777216
         Spec. Version: 5244672
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

Since the low execution speed concerns only the reading instructions, I put the profiling instructions only on the instructions:
```
cp.enable()
rawdata = scope.query(":WAV:DATA?")[11:-1]
scope.write(":WAI")
cp.disable()
```

`scope.query` is executed 78 times (38+1+38+1).
**Total script execution elapsed time is: 44.74 sec**
Total `scope.query` execution time is: 71.74 sec

Here's the report sorted by cumtime:
```
         66204 function calls in 41.758 seconds

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
       78    0.001    0.000   41.740    0.535 messagebased.py:596(query)
       78    0.003    0.000   41.739    0.535 messagebased.py:409(read)
       78    0.012    0.000   41.732    0.535 messagebased.py:379(_read_raw)
      768    0.005    0.000   41.700    0.054 highlevel.py:332(read)
      768    0.267    0.000   41.695    0.054 tcpip.py:430(read)
    11625   41.192    0.004   41.192    0.004 {built-in method select.select}
    11469    0.178    0.000    0.178    0.000 {method 'recv' of '_socket.socket' objects}
    12237    0.037    0.000    0.037    0.000 {method 'extend' of 'bytearray' objects}
     2304    0.015    0.000    0.021    0.000 sessions.py:336(get_attribute)
      156    0.001    0.000    0.019    0.000 messagebased.py:204(write)
      156    0.000    0.000    0.018    0.000 messagebased.py:194(write_raw)
      156    0.000    0.000    0.017    0.000 highlevel.py:354(write)
      156    0.002    0.000    0.017    0.000 tcpip.py:511(write)
      156    0.013    0.000    0.013    0.000 {method 'send' of '_socket.socket' objects}
    12549    0.007    0.000    0.007    0.000 {built-in method builtins.len}
     2304    0.005    0.000    0.005    0.000 attributes.py:115(in_resource)
       78    0.003    0.000    0.003    0.000 {method 'decode' of 'bytearray' objects}
      768    0.002    0.000    0.003    0.000 __init__.py:1356(debug)
     2304    0.002    0.000    0.002    0.000 {built-in method builtins.isinstance}
      768    0.002    0.000    0.002    0.000 common.py:61(<lambda>)
     1692    0.002    0.000    0.002    0.000 {built-in method builtins.min}
       78    0.001    0.000    0.002    0.000 resource.py:198(ignore_warning)
       78    0.001    0.000    0.001    0.000 contextlib.py:116(__exit__)
      768    0.001    0.000    0.001    0.000 {built-in method time.time}
      768    0.001    0.000    0.001    0.000 __init__.py:1614(isEnabledFor)
       78    0.000    0.000    0.001    0.000 contextlib.py:237(helper)
      156    0.000    0.000    0.001    0.000 {built-in method builtins.next}
     1002    0.001    0.000    0.001    0.000 resource.py:98(session)
       78    0.001    0.000    0.001    0.000 contextlib.py:81(__init__)
      156    0.000    0.000    0.001    0.000 highlevel.py:164(ignore_warning)
     1536    0.001    0.000    0.001    0.000 {built-in method builtins.max}
       78    0.000    0.000    0.001    0.000 contextlib.py:107(__enter__)
      234    0.000    0.000    0.000    0.000 {method 'endswith' of 'str' objects}
       78    0.000    0.000    0.000    0.000 {method 'index' of 'bytearray' objects}
       78    0.000    0.000    0.000    0.000 {method 'difference_update' of 'set' objects}
      156    0.000    0.000    0.000    0.000 {method 'encode' of 'str' objects}
       78    0.000    0.000    0.000    0.000 {method 'update' of 'set' objects}
       78    0.000    0.000    0.000    0.000 {built-in method builtins.getattr}
       78    0.000    0.000    0.000    0.000 {method 'disable' of '_lsprof.Profiler' objects}
```

As you can see, most of time is spent during read.

I don't know if this can help reducing the read time.

I have tried to set/unset the `VI_ATTR_TCPIP_NODELAY` attribute, but same result.
I'm not sure it will solve the problem.

I read **ASCII** data. Each read returns 203135 characters.

We can see here that 78 calls, result in 768 `tcpip.py` reads.
We can also see that this implies 11625 `select.select`.
So is there a way to reduce these numbers, by increasing a buffer size for example ?

I'm sure it's not a hardware issue. I don't use wifi anymore, I use a good quality crossover cable for testing.

MatthieuDartiailh, 2020-01-29:
Could you try installing the development version of pyvisa-py to confirm that the issue still exist ?

nbenm, 2020-01-29:
How can I do this?
I've installed pyvisa dev version using: `pip3 install -U https://github.com/pyvisa/pyvisa/zipball/master`
But I don't know how install pyvisa-py dev version

MatthieuDartiailh, 2020-01-29:
Same but replacing the second `pyvisa` by `pyvisa-py` in the url.

nbenm, 2020-01-29:
I don't understand really what's happening.
I had installed two Python packages: `pyVISA` and `pyVISA-py`
I was the `using import visa`
and `rm = visa.ResourceManager('@py')`
I had then 44 sec execution time for the full script. 41 sec just for `:WAV:DATA?` query instruction.

After installing pyvisa-py dev version, it's the same.

But after uninstalling pyvisa-py and replacing `using import visa` with `using import pyvisa as visa`
and replacing `rm = visa.ResourceManager('@py')` with `rm = visa.ResourceManager()`
total execution time falls to 28.5 sec including 26.5 sec for `:WAV:DATA?` query instruction.

Do you have any explanation?
Maybe there was a conflict between the two packages.
If I understand well, I'm not using anymore pyVISA-py, but only pyVISA (dev version)
```
pip3 list|grep -i visa
PyVISA          1.11.0.dev0
```

To be completely complete in my tests, I have uninstalled pyVISA dev version, and reinstalled Pypi version. I have same results as with dev version.
```
pip3 list|grep -i visa
PyVISA          1.10.1
```

Can we say it is related to the use of pyVISA-py?

**So it's much better now.**

But when I will use in the futur use Rigol's "Long Memory", I will have much more data to retrieve. Nearly 10 or 20 times more.
I would take up to 10 minutes...

MatthieuDartiailh, 2020-01-29:
The main difference here is that you are now using the VISA library from National instrument written in C that is faster for low-level operations. Still I am surprised by the transfer time. Could you profile again ? 

Also I don't get why is query called so many times in your test. Is it in a loop ?

nbenm, 2020-01-30:
Yes, it's in a loop.

Here's a small script I have written for test purpose.

[stats_pour_github.txt](https://github.com/pyvisa/pyvisa/files/4133851/stats_pour_github.txt)

I read only one channel from the scope.
Memory depth is 600K.
Data are ASCII, so there are 38 loops + 1 with less data.
Profiling is only for **reading channel data**.
using it this way, here are the stats:
```
         11739 function calls in 12.948 seconds

   Ordered by: cumulative time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
       39    0.000    0.000   12.948    0.332 messagebased.py:596(query)
       39    0.001    0.000   12.937    0.332 messagebased.py:409(read)
       39    0.005    0.000   12.933    0.332 messagebased.py:379(_read_raw)
      384   12.882    0.034   12.920    0.034 functions.py:1571(read)
      423    0.012    0.000    0.034    0.000 highlevel.py:155(_return_handler)
      423    0.012    0.000    0.013    0.000 highlevel.py:35(_args_to_str)
       39    0.000    0.000    0.011    0.000 messagebased.py:204(write)
       39    0.000    0.000    0.011    0.000 messagebased.py:194(write_raw)
       39    0.008    0.000    0.011    0.000 functions.py:1852(write)
      384    0.005    0.000    0.005    0.000 {method 'extend' of 'bytearray' objects}
      423    0.003    0.000    0.005    0.000 __init__.py:1715(debug)
      384    0.003    0.000    0.004    0.000 __init__.py:47(create_string_buffer)
      423    0.002    0.000    0.003    0.000 enum.py:284(__call__)
      423    0.001    0.000    0.002    0.000 __init__.py:1756(log)
      423    0.001    0.000    0.002    0.000 types.py:116(from_param)
       39    0.002    0.000    0.002    0.000 {method 'decode' of 'bytearray' objects}
      423    0.001    0.000    0.002    0.000 __init__.py:1765(isEnabledFor)
      423    0.001    0.000    0.001    0.000 enum.py:526(__new__)
     1614    0.001    0.000    0.001    0.000 {built-in method builtins.isinstance}
      423    0.001    0.000    0.001    0.000 {method 'from_param' of '_ctypes.PyCSimpleType' objects}
      807    0.001    0.000    0.001    0.000 __init__.py:1614(isEnabledFor)
      423    0.000    0.000    0.001    0.000 types.py:58(from_param)
      384    0.000    0.000    0.001    0.000 __init__.py:1356(debug)
     1692    0.001    0.000    0.001    0.000 {method 'append' of 'list' objects}
       39    0.000    0.000    0.001    0.000 resource.py:198(ignore_warning)
       39    0.000    0.000    0.000    0.000 contextlib.py:116(__exit__)
       78    0.000    0.000    0.000    0.000 {built-in method builtins.next}
       39    0.000    0.000    0.000    0.000 contextlib.py:237(helper)
      423    0.000    0.000    0.000    0.000 {built-in method _ctypes.byref}
      462    0.000    0.000    0.000    0.000 resource.py:98(session)
       78    0.000    0.000    0.000    0.000 highlevel.py:166(ignore_warning)
       39    0.000    0.000    0.000    0.000 contextlib.py:107(__enter__)
       39    0.000    0.000    0.000    0.000 contextlib.py:81(__init__)
       78    0.000    0.000    0.000    0.000 {method 'endswith' of 'str' objects}
       39    0.000    0.000    0.000    0.000 {method 'difference_update' of 'set' objects}
       39    0.000    0.000    0.000    0.000 {method 'update' of 'set' objects}
       39    0.000    0.000    0.000    0.000 {built-in method builtins.getattr}
       78    0.000    0.000    0.000    0.000 {built-in method builtins.len}
       39    0.000    0.000    0.000    0.000 {method 'encode' of 'str' objects}
       39    0.000    0.000    0.000    0.000 {method 'disable' of '_lsprof.Profiler' objects}
```

If I replace `rm = visa.ResourceManager()` with `rm = visa.ResourceManager('@py')` for using pyvisa-py, I have the following stats:
```
         32716 function calls in 19.051 seconds

   Ordered by: cumulative time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
       39    0.000    0.000   19.051    0.488 messagebased.py:596(query)
       39    0.001    0.000   19.049    0.488 messagebased.py:409(read)
       39    0.006    0.000   19.046    0.488 messagebased.py:379(_read_raw)
      384    0.002    0.000   19.031    0.050 highlevel.py:386(read)
[... 16 lines cut ...]
       39    0.000    0.000    0.001    0.000 contextlib.py:116(__exit__)
       39    0.000    0.000    0.001    0.000 tcpip.py:514(write)
      384    0.001    0.000    0.001    0.000 __init__.py:1614(isEnabledFor)
       78    0.000    0.000    0.000    0.000 {built-in method builtins.next}
       39    0.000    0.000    0.000    0.000 contextlib.py:237(helper)
      384    0.000    0.000    0.000    0.000 {built-in method time.time}
      768    0.000    0.000    0.000    0.000 {built-in method builtins.max}
      462    0.000    0.000    0.000    0.000 resource.py:98(session)
       78    0.000    0.000    0.000    0.000 highlevel.py:166(ignore_warning)
       39    0.000    0.000    0.000    0.000 contextlib.py:81(__init__)
       39    0.000    0.000    0.000    0.000 contextlib.py:107(__enter__)
       39    0.000    0.000    0.000    0.000 {method 'index' of 'bytearray' objects}
       39    0.000    0.000    0.000    0.000 {method 'send' of '_socket.socket' objects}
       78    0.000    0.000    0.000    0.000 {method 'endswith' of 'str' objects}
       39    0.000    0.000    0.000    0.000 {method 'difference_update' of 'set' objects}
       39    0.000    0.000    0.000    0.000 {built-in method builtins.getattr}
[... cut ...] thod 'update' of 'set' objects}
       39    0.000    0.000    0.000    0.000 {method 'encode' of 'str' objects}
       39    0.000    0.000    0.000    0.000 {method 'disable' of '_lsprof.Profiler' objects}
```
In the real world, I have much more data to process.
Clearly, py-visa is ~1.5 times faster than pyvisa-py

I have also tried a python script using telnetlib3 python library. It's the same as pyvisa-py.
Finally, I have tried a shell script using `nc` (netcat). Results were same as pyvisa.

MatthieuDartiailh, 2020-01-31:
It appears that each read is split in about 10 low level calls. To improve performance you could increase the `chunk_size` attributes of the resources to read larger blocks at a time.

However if performances are crucial I would suggest moving away from ASCII data transfer and using binary transfer instead.

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh