pyvisa/pyvisa #491: How do I send low level GPIB commands using PyVISA?

kitmonisit, 2020-01-31:
**Full disclosure:** I have posted this question on [Stack Overflow](https://stackoverflow.com/q/59984992/366309) but does not seem to be getting any traction. There is a [similar question there](https://stackoverflow.com/q/59288728/366309) (not mine) which has only heard crickets for more than a month after it was posted. So, I would like to apologize if this is not the correct forum to ask.

So I have initialized a GPIB instrument like so:

    import pyvisa
    rm = pyvisa.ResourceManager()
    kk = rm.open_resource('GPIB0::1::INSTR')
    
    # kk is some Keithley 2400 source meter unit

According to the [PyVISA docs][1], `kk` has a `send_command` method to write GPIB commands.  So I try the following lines, without success:

    kk.send_command('SDC')
    kk.send_command(b'SDC')
    kk.send_command(4) # SDC corresponds to the decimal value 4 (but I don't understand this)

The reason why I want to do this is because the [Keithley 2400 manual (PDF)][2] recommends sending the low level `DCL` or `SDC` on the bus to quickly cause the instrument to abort whatever it is doing. They are claimed to be faster than the SCPI `:abort` command.

I have done a lot at the SCPI level in the past but I am very new to GPIB. I am most likely missing a lot of things such as: asserting REN, ATN, or any other GPIB lines that I am just starting to learn about.

The manual shows on page D-9 and D-10 some codes related to GPIB commands, but I don't know what to make of these:
[![Keithley 2400 SMU User's Manual page D-9][3]][3]
[![Keithley 2400 SMU User's Manual page D-10][4]][4]

**Question:** What is the correct way to send low-level GPIB commands using the `send_command` method of a `GPIBInstrument` instance in PyVISA? I would greatly appreciate a minimal example that shows the essential boilerplate code.

  [1]: https://pyvisa.readthedocs.io/en/latest/api/resources.html#pyvisa.resources.GPIBInstrument.send_command
  [2]: https://download.tek.com/manual/2400S-900-01_K-Sep2011_User.pdf
  [3]: https://i.stack.imgur.com/2HUC9.png
  [4]: https://i.stack.imgur.com/zP3X2.png

MatthieuDartiailh, 2020-01-31:
GPIB commands need to be sent as bytes and the interface currently provided by Pyvisa is not very user friendly. 

In your case you need to do:
```python
kk.send_command(b"\x04")  # SDC: hex value is 04
```

kitmonisit, 2020-02-03:
So I got back to the lab and tried the following:

    kk.send_command(b'\x04')
    kk.send_command('\x04')

All in all the codes I tried are `0x04 SDC` `0x14 DCL` `0x3F UNL` and `0x5F UNT` but I'd get the following error:

    VisaIOError: VI_ERROR_NSUP_OPER (-1073807257): The given session or object reference does not support this operation.

I looked around in the related issues and I thought I might need to do `open_resource('GPIB0::INTFC')` in order to operate the bus (and not a specific instrument) but it's not listed in `list_resources()`:

    ('ASRL3::INSTR', 'GPIB0::3::INSTR', 'GPIB0::11::INSTR', 'GPIB0::24::INSTR')

The 3 instruments above with address 3, 11, and 24 I can identify physically. I have an NI GPIB-USB-HS connected to a USB port.  That must be `ASRL3::INSTR`?  So I try it out:

    bus = open_resource('ASRL3::INSTR')
    bus.send_ifc() # Error. This is a SerialInstrument
    bus.send_command('\x04') # Error. This is a SerialInstrument
    bus.write_raw('\x04') # (1, <StatusCode.success: 0>)
    bus.write('\x04') # (3, <StatusCode.success: 0>) Additional two bytes must be the CR LF termination?

Though I'm not really sure how to validate that `0x04 SDC` has really taken effect.

MatthieuDartiailh, 2020-07-23:
Sorry I completely skipped your answer. By default list_resource filter resources and return only INSTR resources. You should nonetheless be able to access GPIB::INTFC.

kitmonisit, 2021-01-12:
Sorry I haven't gotten back to this in a long time. So this is what I am doing now:

``` python
import pyvisa
rm = pyvisa.ResourceManager()
intf = rm.open_resource("GPIB0::INTFC")

# 0x14 is DCL (Device CLear)
cmd = bytes([0x14, ])
intf.send_command(data=cmd)
```

which gives me the following error

> VI_ERROR_RSRC_LOCKED (-1073807345): Specified type of lock cannot be obtained, or specified operation cannot be performed, because the resource is locked.

MatthieuDartiailh, 2021-01-12:
This is the proper way to go but I have no idea what caused the interface to be locked in the first place. 

Do you access at the same time the interface using a third party software (such as NI-Max). Using it while using PyVISA may cause issues. It would not be completely unreasonable for NI-Max to lock the interface.