pyvisa/pyvisa #492: Unclear requirements for pyvisa-shell to work with SCPI

AndKe, 2020-02-01:
I have an electronic load (ET5410) which is a SCPI device. it's id is set to 0.

```
Machine Details:
   Platform ID:    Linux-5.3.0-26-generic-x86_64-with-Ubuntu-19.10-eoan
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.5
   Compiler:       GCC 9.2.1 20191008
   Bits:           64bit
   Build:          Nov 20 2019 09:21:52 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

I connect to it like: 
```
(visa) list
( 0) ASRL/dev/ttyS4::INSTR
( 1) ASRL/dev/ttyS0::INSTR
( 2) ASRL/dev/ttyUSB0::INSTR
(visa) open 2
ASRL/dev/ttyUSB0::INSTR has been opened.
You can talk to the device using "write", "read" or "query".
The default end of message is added to each message.
```
(Never specified a baud-rate - which is 115200)  
And now I am lost as I can't find any working query - and I do not know whatever it is due to the missing linux-gpib, or something else.

Please help me understand how to ensure connection, and some sommands so I can figure out other possible commands.
Thank you.

MatthieuDartiailh, 2020-02-02:
Your resource is a serial resource, as a consequence you can disregard the missing linux-gpib. To set the baud rate you can use the `attr`  command (use `attr list` to know all available attributes). 

Next you need to determine the proper write and read termination character (this should be in your instrument manual). 

The manual should also list valid commands (SCPI only defines a format not a generic set of instruction).

AndKe, 2020-02-02:
Thank you. That explains the prominently placed option to termination character.

The manufacturer does not respond (maybe due to Chinese new year) - is there (somewhere) any database/library with known equipment commands?

MatthieuDartiailh, 2020-02-02:
Not that I know of no. You never got the instrument manual ?

AndKe, 2020-02-03:
The manual mentions the SCPI interface, nothing more.
(I thought that there it maybe had some standard interrogation command that dumped all commands and expected data types.)
In any case: this is not a pyvisa issue, so I'll close the issue, and try to get "east tester" to respond.
Thank you

closed 2020-02-03 by AndKe