pyvisa/pyvisa #493: Termination characters are not used in RNDLAB power supply RND LAB 320-KA3005P via RS232 Serial Port

RAVIVANAMA, 2020-02-06:
Hello support team, 

I am using a pyvisa library to communicate with different Power supply devices via RS232 serial port.

I am facing issue to communicate with **RND LAB 320-KA3005P** Power Supply.
Termination characters are not handled in the mentioned power supply model.

I tried to write and read the data  as shown below :

rm = pyvisa.ResourceManager("@py")
device_connection = rm.open_resource("ASRLCOM6::INSTR")
# By default write_termination added as '\r\n'
# But RND LAB Power Supply required as None as below 
device_connection.write_termination = None
# read_termination characters are used '\n', '\r', '0' throws  error as
# exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed
# because termination characters not used in the RND LAB Power Supply, when its responds back.
device_connection.read_termination = "\n"
# Tried with below line with space  "  " setting 
#  where there is space in the string got terminated and received  data  till there.
device_connection.read_termination = "  "

#device_connection.write('*IDN?')
# print(device_connection.read())
# read/write is same as query() 
print([device_connection.query("*IDN?")])

Version Info:
Python 3.5
PyVisa version -1.10.1
Pyserial version – 3.4

Do you have any suggestion/solution for the issue?

MatthieuDartiailh, 2020-02-06:
You can try setting the `read_termination` to `None`.

RAVIVANAMA, 2020-02-06:
Already Tried by setting to None . But no use.
Issue as below :
2020-02-06 15:57:46,911 - pyvisa - DEBUG - Serial.write b'*IDN?'
2020-02-06 15:57:46,912 - pyvisa - DEBUG - ASRLCOM6::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2020-02-06 15:57:48,961 - pyvisa - DEBUG - ASRLCOM6::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "D:/MyWork/project/workspace/ConTest/refactoring/SerialPort_test.py", line 20, in <module>
    print(device_connection.read_raw())
  File "C:\Python36\lib\site-packages\pyvisa\resources\messagebased.py", line 377, in read_raw
    return bytes(self._read_raw(size))
  File "C:\Python36\lib\site-packages\pyvisa\resources\messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Python36\lib\site-packages\pyvisa-py\highlevel.py", line 350, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
2020-02-06 15:57:48,971 - pyvisa - DEBUG - ASRLCOM6::INSTR - closing
2020-02-06 15:57:49,016 - pyvisa - DEBUG - ASRLCOM6::INSTR - is closed
2020-02-06 15:57:49,023 - pyvisa - DEBUG - Closing ResourceManager (session: 2269829)

MatthieuDartiailh, 2020-02-06:
Did you check the instrument does answer to your query, by using for example `read_bytes(1)` ? Also could you post the output of `pyvisa-info` so that I know what version of pyvisa-py you are running and on what OS you are ? Also note that some instrument can be slow to answer and require a pause between write and read.

MatthieuDartiailh, 2020-02-06:
You can have a look at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#getting-the-instrument-configuration-right for usual steps to debug your situation.

MatthieuDartiailh, 2020-02-06:
Also it may be useful to check that the serial parameters (parity etc) are correct. Currently you are using pyvisa defaults value that may not match your instrument expectations

RAVIVANAMA, 2020-02-06:
Here the pyVisa Version info:
Machine Details:
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python36\python.exe
   Version:        3.6.1
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 21 2017 17:54:52 (#v3.6.1:69c0db5)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

RAVIVANAMA, 2020-02-06:
> Could you post the output of `pyvisa-info` so that I know what version of pyvisa-py you are running and on what OS you are ? Also note that some instrument can be slow to answer and require a pause between write and read.

Here the pyVisa Version info:
Machine Details:
Platform ID: Windows-10-10.0.17763-SP0
Processor: Intel64 Family 6 Model 142 Stepping 10, GenuineIntel

Python:
Implementation: CPython
Executable: C:\Python36\python.exe
Version: 3.6.1
Compiler: MSC v.1900 32 bit (Intel)
Bits: 32bit
Build: Mar 21 2017 17:54:52 (#v3.6.1:69c0db5)
Unicode: UCS4

PyVISA Version: 1.10.1

Backends:
ni:
Version: 1.10.1 (bundled with PyVISA)
Binary library: Not found
py:
Version: 0.3.1
ASRL INSTR: Available via PySerial (3.4)
USB INSTR:
Please install PyUSB to use this resource type.
No module named 'usb'
USB RAW:
Please install PyUSB to use this resource type.
No module named 'usb'
TCPIP INSTR: Available
TCPIP SOCKET: Available
GPIB INSTR:
Please install linux-gpib to use this resource type.
No module named 'gpib'

RAVIVANAMA, 2020-02-06:
> Also it may be useful to check that the serial parameters (parity etc) are correct. Currently you are using pyvisa defaults value that may not match your instrument expectations

checked with port settings everything is fine.
with read_bytes(1) method working fine.  i dont want to provide the size.
but i wan to use write and read methods only .

RAVIVANAMA, 2020-02-06:
> You can have a look at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#getting-the-instrument-configuration-right for usual steps to debug your situation.

MatthieuDartiailh, 2020-02-06:
Why did you close ?

RAVIVANAMA, 2020-02-06:
I didn’t closed. I thought you closed it. Please re-open.
With read_bytes(1)working.
I want to use read() only. Please let me know it is possible.

RAVIVANAMA, 2020-02-07:
After using read_bytes(1) read all bytes and last character is None as show below the status
The last character is as expection error:
2020-02-07 09:29:29,223 - pyvisa - DEBUG - ASRLCOM6::INSTR - reading 1 bytes (last status None)
Error at Reading
2020-02-07 09:29:31,224 - pyvisa - DEBUG - ASRLCOM6::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
**Buffer content: bytearray(b'')**

MatthieuDartiailh, 2020-02-08:
Serial instrument have different ways to signal an end of message. You can select which one is used through the `end_input` attribute of the resource, which expect one of the value of the `SerialTermination` enumeration found in `pyvisa.constants`.  Given your instrument documentation is not really clear on the topic you may have to experiment a bit. In all cases, you should set `read_termchar` to `None` to disable the checks of the termination character. Hopefully we will manage to fix your issue and improve the documentation based on what we learn.

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback, feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh