pyvisa/pyvisa #494: writing hangs while waiting to trigger

amirc2013, 2020-02-13:
Hello,
I have CXAc n9000a and im using pyvisa.
If somehow my cxa missed the trigger - I get stuck for ever... 
If I put timeout - Ill get an error - and then when I will try to write any other command - I'll get stuck again (or time out again).

Why ? How can I fix it ?
Is the problem is with you or the instrument ?

MatthieuDartiailh, 2020-02-13:
Could you post the output of `pyvisa-info` and a minimal script illustrating you are trying to do ? I never worked with that instrument so it may not be easy for me to figure out what is wrong.

amirc2013, 2020-02-13:
I dont know what `pyvisa-info` means but this is the error output :

> Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\SSS\Python_SSS\lib\site-packages\pyvisa\resources\messagebased.py", line 223, in write
    count = self.write_raw(message.encode(enco))
  File "C:\SSS\Python_SSS\lib\site-packages\pyvisa\resources\messagebased.py", line 201, in write_raw
    return self.visalib.write(self.session, message)
  File "C:\SSS\Python_SSS\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1866, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\SSS\Python_SSS\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

About my code - this is not really matter since it is about its *current* state.
I will post a different code that bring the same error 👍 
`
rm = visa.ResourceManager()
print(rm.list_resources())
cxa = rm.open_resource("TCPIP0::signal::inst0::INSTR")
print(rm.list_resources())
cxa.timeout = 9000

cxa.write("*RST")

`

MatthieuDartiailh, 2020-02-13:
Time out on write are rather uncommon. Did you try restarting the instrument ? `pyvisa-info` is a command line tool that report the state of your installation, simply type it in a command prompt (I assume python is on the PATH).

amirc2013, 2020-02-13:
Oh. The pyvisa installed correctly ! :)
I know that since I made 1000 traces with it.

The problem is that the instrument "stucked".
I don't get any replay from pyvisa unless I stop the measurement (by pressing esc in the scope).
It happens when the CXA misses the trigger. Is there any command that unlock the CXA and free it from been remotely ?

MatthieuDartiailh, 2020-02-13:
As I said I never used that instrument but the manual may have the answer you are looking for. I could try to look in it too but later.

amirc2013, 2020-02-13:
I would appreciate that cause I didn't find an answer for that

MatthieuDartiailh, 2020-02-13:
You can try to use `cxa.clear()`. It will not make the instrument go to local but it may manage to reset the connection to a working state.

amirc2013, 2020-02-13:
I think it works !!
I will update again if it will return. Lets wait until tomorrow before closing - ok ?

MatthieuDartiailh, 2020-02-13:
Sure. Happy to know I could help.

amirc2013, 2020-02-14:
Seems like it did the job.
Thank you so much !

closed 2020-02-14 by amirc2013