pyvisa/pyvisa #495: GPIB instruments query timeout even though they are listed in list_resources

HuangJunye, 2020-02-18:
Hi! I recently installed pyvisa, NI-VISA, etc. on a new computer to connect to my instruments. I have trouble to communicate with GPIB instruments. The instruments are visible in list_resources, but timeout error arises whenever I send a query. Do you have any idea how to solve this issue?

I read related issue #337, but my situation is slightly different. Thank you in advance for the help!

## Symptoms
- GPIB instruments are visible in list_resources()
- `VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.` arises whenever I send a query.
- Communication with serial instruments are working fine.
- I use Agilent Technologies 82357B GPIB-USB interface
- GPIB instruments can communicate using NI-MAX and Keysight Connection Expert.

## Output of `pyvisa-info`
```
Machine Details:
   Platform ID:    Windows-10-10.0.17134-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\programdata\anaconda3\python.exe
   Version:        3.7.3
   Compiler:       MSC v.1915 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 27 2019 17:13:21 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 19922944
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 19922944
         Spec. Version: 5244928
```

MatthieuDartiailh, 2020-02-18:
You say #397 does not contain any relevant information for you but what have you tried so far ? 

You can also have a look at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html for detailed discussion of steps that can be used to debug communication with an instrument.

HuangJunye, 2020-02-19:
@MatthieuDartiailh Thanks for sharing the link. It's really helpful. The problem was indeed caused by not setting the write and read termination character. But I didn't need to do that in my old computer for some reasons, so I wasn't aware of this setting at all. 

There is one small problem. `inst.query('*IDN?')` still throw timeout error, though when I do `inst.write('*IDN?')` and then `inst.read()` is OK. Query only works with delay, but even a extremely short delay works which doesn't really make sense. `inst.query('*IDN?', delay=0.0000000000001)` Why do you think that is the case?

Anyway, thank you very much for pointing to the link. The problem is solved now.

closed 2020-02-19 by HuangJunye

MatthieuDartiailh, 2020-02-19:
`time.sleep` has a resolution of about a ms on modern OSes so I guess this is why even a really small delay can help.