pyvisa/pyvisa #496: [COM] Communication issue with DCA-X using Pyvisa

killertoilet, 2020-02-20:
Trying to simply connect to an instrument like I have hundreds of times in the past, but for some reason it claims resource not found. I can ping the instrument from my PC but cannot connect using Pyvisa. 

rm = visa.ResourceManager()
myInstrument = rm.open_resource("TCPIP0::xx.xx.xx.xx::inst0::INSTR")

VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

The one thing that might have changed is that I installed Keysight EMpro which comes with a python package. Could that have somehow messed up environment variables here?

MatthieuDartiailh, 2020-02-21:
That sounds weird. Did you try to restart the instrument (it may not be answering properly) ? Can you still connect using Keysight connection expert ?

killertoilet, 2020-02-21:
I can connect to other instruments but not this one. Strange, guess not a pyvisa issue then.

closed 2020-02-21 by killertoilet