pyvisa/pyvisa #497: 34470A pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339)

painbird7, 2020-02-26:
Getting "Timeout expired before operation completed" error when the script tries to pull in worth of 130,000 readings from 34470A DMM unit. Everything runs well if the number of samples is relatively small.

**Traceback**
Traceback (most recent call last):
  File "/opt/performancescripts/runFio.py", line 175, in <module>
    power.StopVoltageMeasurements(args['dmmAddress'])
  File "/usr/local/lib/python3.6/site-packages/power/power.py", line 104, in StopVoltageMeasurements
    output = session.read()
  File "/usr/local/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "/usr/local/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib64/python3.6/site-packages/pyvisa-py/highlevel.py", line 350, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

**To Reproduce**
--------------
Steps to reproduce the behavior:
Pull samples up to 130,000 readings in the buffer and try to read it in.

**My code**
def StopVoltageMeasurements(dmmAddress):
	meter = "TCPIP0::" + dmmAddress +"::inst0::INSTR"
	rm = visa.ResourceManager()
	session = rm.open_resource(meter)
	session.write("ABOR")
	session.write("FETC?")
	output = session.read()
	with open('/home/powerMeasurements.txt','a+') as f:
			f.write(output)
	session.close()
	rm.close()

**Output of `pyvisa-info`**
[administrator@smarttest-03 performancescripts]$ pyvisa-info
Machine Details:
   Platform ID:    Linux-3.10.0-1062.el7.x86_64-x86_64-with-centos-7.7.1908-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /bin/python3
   Version:        3.6.8
   Compiler:       GCC 4.8.5 20150623 (Red Hat 4.8.5-39)
   Bits:           64bit
   Build:          Aug  7 2019 17:28:10 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

MatthieuDartiailh, 2020-02-26:
Given the large number of sample did you try to increase the timeout ? It is possible that the total transmission takes longer than the default timeout.

painbird7, 2020-02-26:
Hi Matthieu, I glanced through and couldn't find the line of code that controls the timeout limit. Please help if time permits.

MatthieuDartiailh, 2020-02-26:
The timeout is controlled by the timeout property of the resource which is expressed in ms. So I would do the following (I also changed the code to use a context manager for the resource):

```python
def StopVoltageMeasurements(dmmAddress):
    meter = "TCPIP0::" + dmmAddress +"::inst0::INSTR"
    rm = visa.ResourceManager()
    with rm.open_resource(meter) as session:
        session.write("ABOR")
        session.write("FETC?")
        # Give 10 s to the system to complete the transfer
        session.timeout = 10000
        output = session.read()
    with open('/home/powerMeasurements.txt','a+') as f:
        f.write(output)
    rm.close()

painbird7, 2020-02-26:
Thank you very much Matthieu. I will give it a try and let you know. 10000 didn't make a difference, trying 100000 now.

painbird7, 2020-02-26:
Hi Matthieu, 100000 didn't cut it but I'll go ahead close the ticket as this is not a pyvisa bug. I will try out different timeout limits. Thank you again!

closed 2020-02-26 by painbird7