pyvisa/pyvisa #499: commit 0147607 can't find pyvisa-py

greyltc, 2020-03-07:
Trying with the latest from the master branch, 0147607:
```
$ ipython
Python 3.8.2 (default, Feb 26 2020, 22:21:03) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.13.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pyvisa as visa                                                                                                                                                                                                                 

In [2]: rm = visa.ResourceManager()                                                                                                                                                                                                           
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-2-fc4334da8ef3> in <module>
----> 1 rm = visa.ResourceManager()

/usr/lib/python3.8/site-packages/pyvisa/highlevel.py in __new__(cls, visa_library)
   1576     def __new__(cls, visa_library=''):
   1577         if not isinstance(visa_library, VisaLibraryBase):
-> 1578             visa_library = open_visa_library(visa_library)
   1579 
   1580         if visa_library.resource_manager is not None:

/usr/lib/python3.8/site-packages/pyvisa/highlevel.py in open_visa_library(specification)
   1543             wrapper = 'ivi'
   1544         else:
-> 1545             wrapper = _get_default_wrapper()
   1546 
   1547     cls = get_wrapper_class(wrapper)

/usr/lib/python3.8/site-packages/pyvisa/highlevel.py in _get_default_wrapper()
   1517     except ValueError:
   1518         logger.debug('Did not find pyvisa-py package')
-> 1519     raise ValueError('Could not locate a VISA implementation. Install either the IVI binary or pyvisa-py.')
   1520 
   1521 

ValueError: Could not locate a VISA implementation. Install either the IVI binary or pyvisa-py.
```
Everything is fine with the latest release, 1.10.1

MatthieuDartiailh, 2020-03-07:
Can you post the output of pyvisa-py please ?

greyltc, 2020-03-08:
> the output of pyvisa-py

I'm Sorry, I can't understand what you're looking for. Should `pyvisa-py` be a command I have? It's not on my path.

MatthieuDartiailh, 2020-03-08:
Sorry my bad I meant `pyvisa-info`

greyltc, 2020-03-08:
With the latest commit (0147607) from master:
```
pyvisa-info
Machine Details:
   Platform ID:    Linux-5.5.8-arch1-1-x86_64-with-glibc2.2.5
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.8.2
   Compiler:       GCC 9.2.1 20200130
   Bits:           64bit
   Build:          Feb 26 2020 22:21:03 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.0.dev0

Backends:
   ivi:
      Version: 1.11.0.dev0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Could not instantiate backend
      -> Wrapper not found: No package named pyvisa-py
```
With the 1.10.1 release:
```
pyvisa-info
Machine Details:
   Platform ID:    Linux-5.5.8-arch1-1-x86_64-with-glibc2.2.5
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.8.2
   Compiler:       GCC 9.2.1 20200130
   Bits:           64bit
   Build:          Feb 26 2020 22:21:03 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.4.0.dev0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.0 r[1865]')
      GPIB INTFC: Available via Linux GPIB (b'4.3.0 r[1865]')
```

greyltc, 2020-03-08:
I should mention that I'm running the latest commit from the master branch of pyvisa-py.

MatthieuDartiailh, 2020-03-08:
Thanks for the report. I will try to reproduce. Do you see pyvisa-py in `pip list` when pyvisa fails ?

greyltc, 2020-03-08:
Yeah. `pip list` shows me:
```
...
PyTweening                    1.0.3                 
pytz                          2019.3                
pyusb                         1.0.2                 
PyVISA                        1.11.0.dev0           
PyVISA-py                     0.4.0.dev0            
PyWavelets                    1.1.1                 
pyxdg                         0.26                  
PyYAML                        5.3        
...
```
And in case it matters, I don't use pip to manage my python packages.

MatthieuDartiailh, 2020-07-23:
This should now work with the master branch of each package. Feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh