pyvisa/pyvisa #501: [COM] Communication issue with GPIB instruments using PyVisa 1.10.1

ominaelee, 2020-03-16:
Instrument details
------------------
* Model: NI GPIB-USB-HS
* Communication: GPIB

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\Desense\PycharmProjects\simple_auto\venv\Scripts\python.exe
   Version:        3.7.3
   Compiler:       MSC v.1916 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 25 2019 21:26:53 (#v3.7.3:ef4ec6ed12)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 19923456
         Spec. Version: 5244928

Issue description:
GPIB related actions and commands are very slow comparing to NI-VISA Interactive Panel
Ex. Open Resource
PyVisa code:
rm = visa.ResourceManager()
my_instrument = rm.open_resource('GPIB0::8::INSTR')
=> it takes up to 2 sec while NI-VISA Interactive Panel takes 

Attached NI IOTrace snapshots FYR
[image]
[image]

MatthieuDartiailh, 2020-03-16:
Not sure if it is related but are you running your script through PyCharm ? If you are do you see the same by calling your script from the command line.

ominaelee, 2020-03-16:
Yes, I am using PyCharm. 
Thanks for giving insightful comments. It actually runs fine if calling script from the command line. May I have your recommendations if using Pyvisa in PyCharm? Appreciate your help

MatthieuDartiailh, 2020-03-16:
PyCharm debugger may not play nicely with ctypes or something personally I don't use it so I cannot really help. Try to look if you can change how the script is run. Good luck

ominaelee, 2020-03-17:
I tried several scenarios and found out it is due to both PyCharm and Python class.

If I run the script below in CMD line, it large response times are observed if query voltage from PS. (PS voltage query running not in a class is around 50 msec)

import sys
sys.path.extend(['C:\\Vince\\Automation\\test_automation', 'C:/Vince/Automation/test_automation'])
import pyvisa as visa # TODO <2020-0317> WA

class E3631A: #WA
def __init__(self, addr):
self.name = 'E3631A'
self.rm = visa.ResourceManager()
self.inst = self.rm.open_resource(addr)

def preset(self):
self.inst.write('*CLS;*RST;*WAI')
error_code, error_msg = self.inst.query('SYST:ERR?').split(',')
if int(error_code) != 0:
raise Exception(error_msg)

def output_state(self, enable: bool):
state = '1' if enable else '0'
b = self.inst.query(f":OUTPut:STATe {state};:OUTPut:STATe?")
print(f"=> PS Output State is {b.strip()}")
error_code, error_msg = self.inst.query('SYST:ERR?').split(',')
if int(error_code) != 0:
raise Exception(error_msg)

def set_channels(self, channel_6V, channel_25V):
self.inst.write(f"APPL P6V,{channel_6V['voltage']},{channel_6V['current']}"
f";:APPL P25V,{channel_25V['voltage']},{channel_25V['current']}")
error_code, error_msg = self.inst.query('SYST:ERR?').split(',')
if int(error_code) != 0:
raise Exception(error_msg)

def measure_voltage(self) -> float:
v_meas = float(self.inst.query(f":MEAS:VOLT?"))
error_code, error_msg = self.inst.query('SYST:ERR?').split(',')
if int(error_code) != 0:
raise Exception(error_msg)
return v_meas

def measure_current(self) -> float:
i_meas = float(self.inst.query(f":MEAS:CURR?"))
error_code, error_msg = self.inst.query('SYST:ERR?').split(',')
if int(error_code) != 0:
raise Exception(error_msg)
return i_meas

def __del__(self):
del self.inst

def main():
ps = E3631A('GPIB0::8::INSTR')
ps.output_state(True)
print(ps.measure_voltage())
print(ps.measure_voltage())
print(ps.measure_voltage())
print(ps.measure_voltage())
del ps
exit(0)

if __name__ == '__main__':
main()

MatthieuDartiailh, 2020-03-17:
That is weird could you try profiling the code to identify where the slow down is ?

ominaelee, 2020-03-17:
I have traced this down to visa library wrapper function (write and read), but could not get any clues. Looking at IO trace log in [comment 1 ](https://github.com/pyvisa/pyvisa/issues/501#issue-581961871), I could not tell the differences in between.

MatthieuDartiailh, 2020-03-17:
Could you do it again on the class case ? I don't see how encapsulating things in a class could degrade performance.

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh