pyvisa/pyvisa #502: Unable to read Buffer data for python 3.8/python 3.7

raquib-shefu, 2020-03-18:
Instrument details
------------------
* Model: Keithley 2400
* Communication: RS232
my Code :: 
import visa
import numpy as np
import time as t

rm = visa.ResourceManager()
print(rm.list_resources())
smu = rm.open_resource("ASRL8::INSTR")
smu.write('*RST')
smu.write(':SOUR:FUNC:MODE VOLT')
smu.write(':SOUR:LIST:VOLT 5')
smu.write(':ARM:COUNT 1')
smu.write('TRIG:COUN 2500')  # %maximum 2500
smu.write(':TRIG:DEL 0.0')
smu.write(':SOUR:DEL 0.0')
smu.write(':SOUR:VOLT:RANGE 1')
smu.write(':SENSE:CURR:PROT 100e-3')
smu.write(':SENSE:FUNC:CONC OFF')
smu.write(':SENSE:FUNC "CURR:DC"')
smu.write(':SENSE:CURR:RANGE 1e-3')
smu.write(':SENSE:CURR:NPLC 0.01')
smu.write(':SENSE:AVERAGE:STAT OFF')
smu.write(':SYSTEM:AZERO:STAT OFF')
smu.write(':TRAC:CLE')
smu.write(':TRAC:POINTS 2500')  # %max 2500
smu.write(':TRAC:FEED SENSE1');
smu.write(':SOUR:VOLT:MODE LIST');
smu.write(':TRAC:FEED:CONT NEXT');
smu.write(':OUTPUT ON');
smu.write(':INIT')
t.sleep(5)
smu.write(':FORM:ELEM CURR,TIME')  # select for current
smu.write("trace:data?")

smu.write(':OUTP OFF')

MatthieuDartiailh, 2020-03-19:
Could you clarify what you mean by not available when referring to `read` ? Also did you read this section of the documentation about debugging communication: https://pyvisa.readthedocs.io/en/latest/introduction/communication.html

raquib-shefu, 2020-03-20:
I have read the document and I am supposed to use **values = inst.query_ascii_values('CURV?')**  for reading ascii vaues.When I write smu.que... nothing is coming up. I have given a screenshot of my problem. So the direct question will be for sending command I am using **smu.write ()** What Shoud I use for reading the value back to my pc. According to the documentation it should be **smu.query_ascii_values()**. But it is missing in the module. 
[image]

MatthieuDartiailh, 2020-03-20:
Don't trust your  completion engine here. PyVISA will return the appropriate resource based on the address which the auto completion engine cannot follow. Using type annotation to indicate you get a MessagebasedResource (or the exact resource but I don't think you need to be that specific) may alleviate your problem but I am not sure.

raquib-shefu, 2020-03-20:
Thats great! Its atleast started working but I am getting the timeout error with the following **commands**
I=smu.query_ascii_values("trace:data?")
print(I)
**OUTPUT**
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
I have read the earlier case where you have mentioned about testing the IDN? check. It is working for me. I am not understanding how to send the command and to tell stoop reading when it reaches end of the buffer.

MatthieuDartiailh, 2020-03-20:
Depending on the amount of data to retrieve it may take more than the default timeout value. Try to increase the timeout or check if you can read the data when you transfer less data.

raquib-shefu, 2020-03-22:
Thank you very much for your help earlier. Manipulating the **.timeout** worked perfectly and am able to read. But it looks like the my RS232 communication can not read 2500 values (max limit of Keithley 2400 SMU)  where I did read all the data back with Matlab instrument control toolbox. 
So this is what I want: 
► I want to read all the measurement values from the instrument with my RS232 communicaiton.
 **Problem** in my understanding is :  
►I can read around 200 points  turn with read_bytes(n) / read() / read_raw() etc available functions.  ► When I increase the data point nothing happens and my program gets hang.
► I guess this is happening because of the limitation of RS232  communication. 
**My solution** so far: 
►As I can declare the read_byte(n) size I can make this function run in a loop. (similar solution worked for Matlab)  but the problem occurs when the last reading is happening.
► The read_bytes(n) does not understand if i get '\r' and stop.
So is there any way to control this reading ?

MatthieuDartiailh, 2020-03-22:
`read_bytes` can take the `break_on_termchar` keyword to return early which is what you want.

raquib-shefu, 2020-03-23:
I have already tried print(smu.read_bytes(100,break_on_termchar=True)) where the actual byte number is 27 and last byte is '\r'. It does not work. Am I doing something wrong ?

MatthieuDartiailh, 2020-03-23:
I am not sure what can be causing the problem. First if you know in advance the number of bytes, just read what you need that is the easiest/safest method. 

If you want to dig a bit more, look at the value of the `end_input` attribute and if it is not set to `constants.SerialTermination.termination_char` try setting to this value and see if it changes the result.

raquib-shefu, 2020-03-23:
I did not understand this sorry, Could you please elaborate? I searched for end_input and I got only this.  
class AttrVI_ATTR_ASRL_END_IN(EnumAttribute):
    """VI_ATTR_ASRL_END_IN indicates the method used to terminate read
    operations.
    """
    resources = [(constants.InterfaceType.asrl, 'INSTR')]

    py_name = 'end_input'

    visa_name = 'VI_ATTR_ASRL_END_IN'

    visa_type = 'ViUInt16'

    default = constants.SerialTermination.termination_char

    read, write, local = True, True, True

    enum_type = constants.SerialTermination

MatthieuDartiailh, 2020-03-23:
The object that you got from `open_resource` is an instance of `SerialInstrument` (based on the address). This class has a property `end_input` (dynamically set which makes it very hard to discover, I am working on that).  So try something like:

```python
import pyvisa
from pyvisa.constants import SerialTermination
import numpy as np
import time as t

rm = pyvisa.ResourceManager()
print(rm.list_resources())
smu = rm.open_resource("ASRL8::INSTR")
print(smu.end_input)
smu.end_input = SerialTermination.termination_char
```

raquib-shefu, 2020-03-23:
With that I have tried this: 
```
smu.write('*IDN?')
data = smu.read_bytes(81, break_on_termchar=True)
print(data)

```
 For 80 bytes the output is : 
b'KEITHLEY INSTRUMENTS INC.,MODEL 2400,4378817,C34 Sep 21 2016 15:30:00/A02  /V/M\r'
For 81 byte is gets hang. That means it is not understanding to stop  after \r

MatthieuDartiailh, 2020-03-23:
By curiosity could you post the output of the `pyvisa-info` command ? I am wondering which backend you are using.

Also I just tested and found a bug in read_bytes implementation... Since I am in the middle of some big clean up it will be part of it. In the meantime you can copy-paste the implementation of `read_bytes` and replace this test
```python
if break_on_termchar and status == termchar_read: 
    break
```
by 
```python
if break_on_termchar and (status == termchar_read or status == constants.StatusCode.success):  
    break
```
I am sorry for that .

raquib-shefu, 2020-03-24:
Out put for pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 13, GenuineIntel

Python:
   Implementation: CPython
   Executable:     D:\My Drive\PHD in UAH\Phd thesis\My Research\Neural Network\Instruments\Kethley2400\SMU2400_desktoppython\desktop_venv\Scripts\python.exe
   Version:        3.8.1
   Compiler:       MSC v.1916 32 bit (Intel)
   Bits:           32bit
   Build:          Dec 18 2019 22:39:24 (#tags/v3.8.1:1b293b6)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\Windows\System32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 19923456
         Spec. Version: 5244928

raquib-shefu, 2020-03-24:
I have change the section you recommended and it still does not work.

MatthieuDartiailh, 2020-03-24:
I am surprised I tested that change reading the IDN just like you did. Without this modification I cannot read the answer if I request too many bytes whereas I can with the patch. Could you share your updates script ?

raquib-shefu, 2020-03-24:
```
import visa
# import numpy as np
import time as t
from pyvisa.constants import SerialTermination

reading = 20
rm = visa.ResourceManager()
print(rm.list_resources())
smu = rm.open_resource('ASRL21::INSTR')
del smu.timeout
smu.read_termination = '\r'
print(smu.end_input)
smu.end_input = SerialTermination.termination_char
# smu.write_termination = '\r'+'\n'
smu.write('*IDN?')
data = smu.read_bytes(80, smu.end_input)
```

MatthieuDartiailh, 2020-03-24:
Do you have a single python environment and are you sure you edited the right copy of messagebased.py ? Could you add `print(status)` before the if clause, so that we can check what is happening ?

raquib-shefu, 2020-03-24:
i am using **python 3.8**. yes I have edited the right copy as I have opened a separate project only for this control.  

I have used 80 byte count and with print(status) before if clause is following:  
('ASRL1::INSTR', 'ASRL21::INSTR')
SerialTermination.termination_char
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success_max_count_read
StatusCode.success
b'KEITHLEY INSTRUMENTS INC.,MODEL 2400,4378817,C34 Sep 21 2016 15:30:00/A02  /V/M\r'

Whem I do use 81 , the output printed upto StatusCode.success, and then gets hang. 
I am using only single environment. If you want I can upload entire project. Do you think that python 3.8 has anything to do with it ?

MatthieuDartiailh, 2020-03-24:
I get the issue. I misread your code. The signature of `read_bytes` is 
```python
def read_bytes(
        self, count: int, chunk_size: int = None, break_on_termchar: bool = False
    ) -> bytes:
```
so currently you are passing a chunk size but not specifying `break_on_termchar`. Try passing `break_on_termchar=True` by keyword that should fix your issue.

raquib-shefu, 2020-03-24:
Yes it is working now. Thank you very much for you time.

MatthieuDartiailh, 2020-03-24:
Could you check one thing ? Since you can manually get all the data, you may be able to use `read` or `read_ascii_values` by specifying a smaller chunk size (through the `chunk_size` attribute of the resource (`smu` in your case)). If everything works manually with 200 a chunk size of 200 should work. If it does I will add an extra note to the documentation about this.

raquib-shefu, 2020-03-24:
I have tried with read() and read_ascii_values(). Both is working. I have measured a huge data set with with almost 60000 bytes. They are working. I did not use any chunk size attribute though as i  did not understand your point of specifying smaller chunk size?

MatthieuDartiailh, 2020-03-25:
What did you change ? Earlier you reported that you could not read more than 200 points.

raquib-shefu, 2020-03-25:
These are the following changes I made according to your suggestion and it is working perfectly now. Thank you very much. 
►`from pyvisa.constants import SerialTermination`
►`smu.end_input = SerialTermination.termination_char`
►in the message_based_resource, I have added                 
   ```
 if break_on_termchar and (status == termchar_read or status == constants.StatusCode.success):
                        break
```
►The finally I have added this portion for reading 
```
s1 = ''
for i in range(1, 8, 1):
s = smu.read_bytes(10000,break_on_termchar=True)
s1 = s1 + (s.decode("utf-8"))[1:]
```
```
I think these are all the changes that make things work for my case.

raquib-shefu, 2020-03-25:
I think the change in the message_based_resource is the most critical one. After that when I applied 
break_on_termchar=True, read_bytes() and read() and read_ascii_values() are working for more 2500 data point where previously it was reading only 200 points.I hope I answered your query.

MatthieuDartiailh, 2020-03-25:
The changes in messagebased only affects `read_bytes` not the other ones which is why I was curious what allowed read to work properly.

raquib-shefu, 2020-03-25:
i did not add read_termination=\r’ previously. I guess that is the reason. Actually I was so focused on read_byte() and did so many changes, I lost tract. But if you want I can check again from the beginning why it was not working at first. Plz Let me know.

MatthieuDartiailh, 2020-03-25:
If you have a bit of time, I would appreciate to know what made `read` work. I assumed you had no termination character issue since you could read the `*IDN?`. Since you changed only two settings it should be fast to check.

raquib-shefu, 2020-03-25:
I found it. These are the two line I added for those two. Previously i was selecting different timeout, thats why if it was more than 200, i was getting timeout_error. Then deleted that.  
```
del smu.timeout
smu.read_termination = '\r'
```

MatthieuDartiailh, 2020-03-25:
Ok that makes sense. If the transfer is slow you need a long enough timeout. I will close the issue then. Thanks for checking.

closed 2020-03-25 by MatthieuDartiailh