pyvisa/pyvisa #503: [COM] Communication issue with Tektronix AFG1062 using Python 3.7 and Pyvisa

JosephGeorges, 2020-03-30:
I am attempting to upload a waveform to the Tektronix AFG1062. 

I want to upload the waveform shown below to the arbitrary waveform generator. I am new to pyvisa and am wondering if I am doing something wrong in my code. The generator gets reset to factory settings, and the output turns on all through code. But I don't get any waveform generated. 

[image]

**This is the code I have:**

import pandas as pd
import numpy as np
import pylab as plt
import pyvisa

#Initialize list of data points 
filename = 'ISO_16750-2_2010 4_6_4a_load_dump.xlsx'
df = pd.read_excel(filename)

data_points = []
data_points = list(df['Data Points'])

rm = pyvisa.ResourceManager()
afg = rm.open_resource('USB0::0x0699::0x0353::1813139::INSTR')
print(afg.query("*IDN?"))
afg.write('*rst')
afg.write('*cls')

#Waveform Parameters
num_points = len(data_points)
sample_rate = 9.216e5 

t = np.linspace(0,num_points/sample_rate, num_points, endpoint = True)

afg.write_ascii_values('WLISt:WAVeform:DATA Load_Dump,', data_points)
afg.query('*opc?')

afg.write('source1:waveform Load_Dump')
afg.write('awgcontrol:run:immediate')
afg.query('*opc?')
afg.write('output1 on')

MatthieuDartiailh, 2020-03-30:
I did not check your instrument manual but on a AWG 5214 I am used to do:
```python
awg.write("WLIST:WAVEFORM:DELETE '{}'".format(name))
awg.write("WLIST:WAVEFORM:NEW '{}' , {}, INTeger" .format(name, len(waveform))
header = "WLIS:WAV:DATA '{}',0,{},".format(name, len(waveform))
awg.write_binary_values(header, waveform, datatype='B')
```
Note that you have may have to create the waveform before uploading data and note also the quotes around the waveform name. If you cannot get it to work I will try to look at your instrument manual.

JosephGeorges, 2020-03-30:
Thanks for the response! I tried it and I got the following error for the  last line: 

struct.error: required argument is not an integer

I'm not sure if I am able to go straight from using linspace with numpy to uploading to the AFG. I have attached the AFG 1000 Series Programmers manual if you want to have a look at it. 

[AFG1000-Programmer-Manual-EN.pdf](https://github.com/pyvisa/pyvisa/files/4405093/AFG1000-Programmer-Manual-EN.pdf)

MatthieuDartiailh, 2020-03-30:
Thanks for the manual. 

As is often the case the manual is not perfectly explicit and in particular does not specify the format of the waveform. Since the resolution of the instrument is 14bits I assume that each value is encoded into two bytes and if things are similar to the 5214 the two highest one are unused.
What it means is that your pulse is going to be encoded in an integer of 14 bits, and the actual voltage will be controlled by the low/high of amplitude/offset values. If you need negative value it makes sense to encode 0 as 2**13 in which case you will keep the offset at 0, if you only need positive values you can encode 0 as 0 and set low to 0. If the following I am going to assume you want negative values too. Here is how you can convert your numpy array of floating point number (should be between -1 and 1) into a waveform and transfer it (hopefully):

```python
# waveform: your data 

# Create an array with all values set to 0 in the above convention
to_transfer = np.ones(len(waveform), dtype=np.uint16)*(2**13)

# Convert your data to fit within the format
to_transfer += np.require(np.rint(8191*waveform), np.uint16)

# Check everything is fine
if to_transfer.max() > 16383 or to_transfer.min() < 0:
    raise ValueError('Analogical values out of range.')

# Write the data to the instrument
awg.write_binary_values("DATA:DATA EMEMory,", waveform, datatype='x')
```

You can also try to read an existing waveform to check I got the format right.

Hope this helps

JosephGeorges, 2020-03-30:
So I tried and it raised the ValueError during the check.

'Analogical values out of range.'

Do you know why?

MatthieuDartiailh, 2020-03-30:
This most likely means that your input is not normalized between -1 and 1. Be sure to normalize your waveform before trying to send it.

JosephGeorges, 2020-03-31:
Okay so I used a for loop to normalize the waveform array between -1 and 1. Now I am getting an error with the very last line:

struct.error: pack expected 0 items for packing (got 5520)

The entire error message is:

> Traceback (most recent call last):
>   File "Z:\PyVisa\Load_Dump_4a.py", line 45, in <module>
>     afg.write_binary_values("DATA:DATA EMEMory,", waveform_norm, datatype='x')
>   File "C:\Users\georges\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 298, in write_binary_values
>     block = util.to_ieee_block(values, datatype, is_big_endian)
>   File "C:\Users\georges\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\util.py", line 515, in to_ieee_block
>     return to_binary_block(iterable, header, datatype, is_big_endian)
>   File "C:\Users\georges\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\util.py", line 496, in to_binary_block
>     return header + struct.pack(fullfmt, *iterable)
> struct.error: pack expected 0 items for packing (got 5520)

I know I have 5520 elements in my waveform array. Is this the issue? 

Also posted my updated code below for your reference:
```
import pandas as pd
import numpy as np
import pylab as plt
import pyvisa

#Initialize list of data points 
filename = 'ISO_16750-2_2010 4_6_4a_load_dump.xlsx'
df = pd.read_excel(filename)

waveform = []
waveform = list(df['Data Points'])

waveform_norm = []
for i in waveform:
    i = 2*(i-min(waveform))/(max(waveform) - min(waveform)) - 1
    waveform_norm.append(i)

waveform_norm = np.array(waveform_norm)

rm = pyvisa.ResourceManager()
afg = rm.open_resource('USB0::0x0699::0x0353::1813139::INSTR')
print(afg.query("*IDN?"))
afg.write('*rst')
afg.write('*cls')

#Waveform Parameters
num_points = len(waveform_norm)
sample_rate = 9.216e5 

t = np.linspace(0,num_points/sample_rate, num_points, endpoint = True)
##plt.plot(t,waveform_norm)
##plt.show()
           
#Create an array with all values set to 0 in the above convention
to_transfer = np.ones(len(waveform_norm), dtype=np.uint16)*(2**13)

#Convert your data to fit within the format
to_transfer += np.require(np.rint(8191*waveform_norm), np.uint16)

#Check everything is fine
if to_transfer.max() > 16383 or to_transfer.min() < 0:
    raise ValueError('Analogical values out of range.')

#Write the data to the instrument
afg.write_binary_values("DATA:DATA EMEMory,", waveform_norm, datatype='x')
```

JosephGeorges, 2020-03-31:
That is indeed a lot more efficient. Thanks for the help! After running the pyvisa-info command I see that I've installed pyvisa 1.1. That is an older version and I should be using the latest version is that correct?

The output of `pyvisa-info` command is below:

`Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\georges\appdata\local\programs\python\python37-32\python.exe
   Version:        3.7.7
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 10 2020 09:44:33 (#tags/v3.7.7:d7c567b08f)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 19923456
         Spec. Version: 5244928
`

MatthieuDartiailh, 2020-03-31:
Yeah that version should work. Do you still have the same error ?

JosephGeorges, 2020-03-31:
Yup! I changed argument to `to_transfer` rather than `waveform` in the last line as you mentioned. However same error I am getting.

MatthieuDartiailh, 2020-03-31:
That does not really make any sense... Can you go in `C:\Users\georges\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\util.py` and add a couple of `print` in `to_binary_block` to check the value of `fullfmt` and `array_length`. I don't see how that logic can fail.

JosephGeorges, 2020-03-31:
So I added the `print `statements in the `util.py` file and I am still getting the same error but this time the output of the print statements were: 
```
5520
<5520x
```

Since the error is a `struct.error: pack expected 0 items for packing (got 5520)` I assume it doesn't like my waveform list. Just not sure why.

JosephGeorges, 2020-03-31:
Okay so I tried changing the last parameter to `datatype = 'x'` to `datatype = 'f'` and the error is now gone. However nothing got uploaded to my function generator.

JosephGeorges, 2020-03-31:
This is a PDF I found explaining the binary file structure for AFG3000 series generators. I assume AFG1000 would be very similar if not the same. I don't know if this will help or not.

[AFG3000 tfw-file format.pdf](https://github.com/pyvisa/pyvisa/files/4409435/AFG3000.tfw-file.format.pdf)

MatthieuDartiailh, 2020-03-31:
My bad the format should have been `"h"` not `"x"`... I should have checked, sorry for the wasted time. "f" is going to create some gibberish. After the upload try to use the front panel to save the waveform and then check the content.

Reading the instrument of you manual it does not seem to use .tfw file. Usually those files (the AWG5214 has something similar) are used in more complex situation.

JosephGeorges, 2020-03-31:
No worries! You've already got me a lot further than I would have been! 

I tried changing the datatype to `'h'` and got no errors. Still can't find anything in my generators memory however. Don't know where the waveform is being saved. 

Regarding TFW files. I've tried to save basic tfw files to usb and then transfer them to the generator via usb and it has worked. I am wondering if I can go directly from python to the generator as TFW or how to change my current python waveform into TFW.

MatthieuDartiailh, 2020-03-31:
Did you try to query the waveform after transfer (using `query_binary_values`) to check it was properly uploaded ? You could also just check the size of the waveform to see if it is your waveform. Also after transfer you could use DATA:COPY EMEMory,USER0 to copy the waveform you uploaded into the USER0 waveform, it may make it easier to find it later to check it.

JosephGeorges, 2020-03-31:
I added the query function underneath the statement we already had: 
```
afg.write_binary_values("DATA:DATA EMEMory,", to_transfer, datatype='h')
afg.query_binary_values('CURV?', datatype='h', is_big_endian=True)
```

It returned the error: 

> Traceback (most recent call last):
>   File "Z:\PyVisa\Load_Dump_4a.py", line 43, in <module>
>     afg.query_binary_values('CURV?', datatype='h', is_big_endian=True)
>   File "C:\Users\georges\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 715, in query_binary_values
>     data_points, chunk_size)
>   File "C:\Users\georges\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 496, in read_binary_values
>     offset, data_length = util.parse_ieee_block_header(block)
>   File "C:\Users\georges\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\util.py", line 331, in parse_ieee_block_header
>     raise ValueError("Could not find hash sign (#) indicating the start of"
> ValueError: Could not find hash sign (#) indicating the start of the block.

Do you know where we need to add the "#"?

MatthieuDartiailh, 2020-03-31:
Reading the manual you sent I did not see any CURV command. I would suggest `DATA:DATA? EMEMory` which is the symmetric of the command used to upload.

JosephGeorges, 2020-03-31:
Okay I tried `DATA:DATA? EMEMory` instead of CURV. 

I am still getting the same error as before:

> ValueError: Could not find hash sign (#) indicating the start of the block.

MatthieuDartiailh, 2020-03-31:
Did you try transferring to another space so that you can inspect it on the instrument ?

For reading the waveform try to use `query` and post the instrument response. I am curious to see what the response looks like.

JosephGeorges, 2020-03-31:
Actually, I think I figured out the reason it wasn't working. 

I changed `"DATA:DATA EMEMory,"` to simply `"DATA EMEMory,"` and my scope which is connected to the AFG outputs the following results: 

[image]

So I think that's it if I'm not mistaking! 

Wow! The manual is really bad at explaining what needs to go in the arguments. 

Thank you for all your help!

MatthieuDartiailh, 2020-03-31:
Amazing ... Both commands are supposed to be identical ! That is why I hate those manuals, it is impossible to do anything without but even with them it is still a nightmare. Thanks for your fast testing and sorry again for the stupid format issue.

closed 2020-03-31 by MatthieuDartiailh

JosephGeorges, 2020-03-31:
No worries at all!

No way I would be this far without your help anyway so thank you! Really appreciate all the help!

JosephGeorges, 2020-04-01:
Hey! Sorry for the bother again, but could you tell me where you found the information regarding the binary format of the AFG?

 Just want to know so I know where to look for future projects!

MatthieuDartiailh, 2020-04-01:
Sadly that was mostly guesswork, knowing the AWG5214 and also looking at the spec sheet to know the resolution of the AWG.

JosephGeorges, 2020-04-13:
Hey Matthieu, 

So from where we left off last time. The python script loads any waveform I need into the edit memory of my AFG. 

I am now trying to take the content of the edit memory, copy it to user1 (or any of the other user memory locations), and then finally assign the contents of user1 to channel 1 and play the waveform. 

Some of the commands I have tried are: 

`afg.write('DATA:COPY USER1, EMEM')`
`afg.write('source1:function USER1')`

These two lines from the AFG1000 manual supposedly copies the content of the edit memory and assigns it to user1 (first line) and then assigns the content in user1 to source1 (channel1) (second line). 

When I run my code, I get no errors, however the waveform stays in the edit memory and doesn't move anywhere else. Here is the full code: 

```
import pandas as pd
import numpy as np
import pylab as plt
import pyvisa

#Initialize list of data points 
filename = 'ISO_16750-2_2010 4_6_4b_load_dump.xlsx'
df = pd.read_excel(filename)

waveform = []
waveform = np.array(df['Voltage'])

w_min, w_max = waveform.min(), waveform.max()
waveform = 2*(waveform - w_min)/(w_max - w_min) - 1

rm = pyvisa.ResourceManager()
afg = rm.open_resource('USB0::0x0699::0x0353::1813139::INSTR')
print(afg.query("*IDN?"))
afg.write('*rst')
afg.write('*cls')

#Waveform Parameters

sample_rate = 9.216e5 #Sampling Period is 9.216e-5, Thus Sample_Rate = 1/Sample_Period

t = np.linspace(1, 6000, 5520, endpoint = True)
plt.plot(t,waveform)
##plt.show()

#Create an array with all values set to 0 in the above convention
to_transfer = np.ones(len(waveform), dtype=np.uint16)*(2**13)

# Convert your data to fit within the format
to_transfer += np.require(np.rint(8191*waveform), np.uint16)

#Check for errors
if to_transfer.max() > 16383 or to_transfer.min() < 0:
    raise ValueError('Analogical values out of range.')

afg.write('source1:function ememory') #Sets the AFG to ARB mode
afg.write('output1 on') #Turns on output 1 of the AFG

#Write the data to the instrument
afg.write_binary_values("DATA EMEMory,", to_transfer, datatype='h', is_big_endian = True) #Copies the waveform into editable memory of AFG
afg.write('DATA:COPY USER1, EMEM')
afg.write('source1:function USER1') 
```

MatthieuDartiailh, 2020-04-13:
Did you try to reverse USER1 and EMEM ? This is not what the manual suggests but a to/from order would seem natural and since we have already found some issues with the manual I would not be totally surprised. 
I would also try to query "SYST:ERR?" just in case an error in not reported on the instrument screen.

JosephGeorges, 2020-04-13:
So I tried switching the places of USER1 and EMEM. Nothing changed. I also queried for "SYST:ERR?" as suggested and it didn't give me any output at all. 

The code just continues to get executed without errors as before. But still no content in User0, the content is only in the EditMemory.

JosephGeorges, 2020-04-13:
Update: 

After executing the original code and placing the waveform in the edit memory, I can then use the same command: `afg.write('DATA:COPY USER0,EMEMory')` in the Python SHELL and the content DOES get copied into USER0. 

When adding this line to the script, nothing happens. Do you know why this could be?

MatthieuDartiailh, 2020-04-13:
The instrument may need some time to fully process putting the waveform in the edit memory after the transfer. I have seen similar things on the AWG5014 where you need to wait after transferring large waveform before you can run them. You could try sleeping for a couple of seconds after uploading the waveform before attempting the copy.

JosephGeorges, 2020-04-14:
Okay, so I think I got that to work! 

However, do you know why the AFG period is always 1 ms whenever I upload any waveform. When I try changing the period from 1 ms to something higher, the waveform on the scope starts to flatten out.  

Sorry for all the questions btw

MatthieuDartiailh, 2020-04-14:
Looking at the manual quickly I did not see anything about setting the sampling rate which means that you may have a fixed time between points and so you end up with whatever period match your number of points. Big guessing there.

For your previous issue was the sleep the right solution ?

JosephGeorges, 2020-04-14:
My apologies, I forgot to get back to you on that. Yes I added a timeout of 10,000 and added OPC commands in between each line that requires a lot of time to process. 

Worked like a charm. Thank you!

DavidLutton, 2020-04-14:
I case this helps:
With a HP 33120A running arbitrary waveform data, the frequency/period becomes the consumption rate of all of the samples.

**Try setting to 0.5ms and seeing if you output frequency/period halves**

JosephGeorges, 2020-04-14:
Hey David, 

I'm not sure I know what you mean. Do you mean manually changing the setting to 0.5ms on the AFG its self?  When I try to change the sample_rate in my Python code, it looks like it has no affect on the AFG. The period still goes to 1 ms. 

This is the output of my scope when connected to CH1 of the AFG: 

[image]

When I begin to increase the period to 10ms on the AFG manually, the output on the scope becomes: 

[image]

DavidLutton, 2020-04-14:
Yes I meant changing the setting to 0.5ms on the AFG.

When the AFG is set to 1ms the on time of your signal is 2.5 divisions ~500us
When the AFG is set to 10ms the on time of your signal is >10 _(~25 ?) divisions ~5000us_

When the AFG is set to 0.5ms the on time of your signal **should be** 1.25 divisions ~250us

This period setting on the AFG controls the total time that all the samples are outputted, before it starts from the first sample again.

JosephGeorges, 2020-04-15:
Hi David,

Yes, I believe it is just a matter of playing with the settings on the scope such as the timebase, frequency/period and so on. 

Thank you!

supersibas11, 2021-01-25:
Hi guys,

It's my first week using PyVisa. I was trying to upload a waveform to an AWG5202 following this example:

https://github.com/tkzilla/visa_control_examples/blob/master/Python/AWG/awg_simple_waveform_sender.py

I guessed it did not change a lot between AWG70k and 5202, but it seems I'm wrong. The error I'm getting at the end of the execution is the following:

[image]

I understand that there seems to be an encoding or a datatype problem when I use write_binary_values, isn't it?

I found this thread and I tried the solutions Mathieu proposed (normalizing the data and converting the floating point to an array), but I did not succeeded, I also tried tp use all the headers types in the write_binary_values function as well as another encoding type, but same result. The waveform is effectively charged into the AWG, but everything is in 0.

I feel the solution is simple, but my lack of experience does not allow me to find it.

Here below, you can find my code, but as I already indicated it, is very similar to that one in the first link.

import pyvisa as visa  
import numpy as np  

print('NumPy Version:', np.__version__)
print('PyVISA Version:', visa.__version__)

# Set up VISA instrument object
rm = visa.ResourceManager()
awg = rm.open_resource('TCPIP0::169.254.138.146::inst0::INSTR')
awg.encoding = 'latin_1'
awg.write_termination = None
awg.read_termination = '\n'
awg.timeout = 25000

print("Connected to ", awg.query('*IDN?'))
awg.query('*opc?')

# Reset and clear the AWG status
awg.write('*rst')
awg.write('*cls')

# These can be changed as a function of the signal requirements
name = 'test_wfm'
sampleRate = 10e9
recordLength = 500000
freq = 100e6

# Create waveform
t = np.linspace(0, recordLength/sampleRate, recordLength, dtype=np.float32)
wfmData = np.sin(2 * np.pi * freq * t)

'''
w_min, w_max = wfmData.min(), wfmData.max()
waveform = 2*(wfmData - w_min)/(w_max - w_min) - 1

to_transfer = np.ones(len(waveform), dtype=np.uint16)*(2**13)
to_transfer += np.require(np.rint(8191*waveform), np.uint16)

if to_transfer.max() > 16383 or to_transfer.min() < 0:
    raise ValueError('Analogical values out of range.')
'''

# Send waveform Data
awg.write("wlist:waveform:delete '{}'".format(name))
awg.write('wlist:waveform:new "{}", {}'.format(name, recordLength))
stringArg = 'wlist:waveform:data "{}", 0, {}'.format(name, recordLength)
awg.write_binary_values(stringArg, wfmData)
awg.query('*opc?')

# Load waveform, being playback, and turn on output
awg.write('source1:waveform "{}"'.format(name))
awg.write('awgcontrol:run:immediate')
awg.query('*opc?')
awg.write('output1 on')

# Check for errors
error = awg.query('system:error:all?')
print('Status: {}'.format(error))

awg.close

Thank you so much for your help and time.

JosephGeorges, 2021-01-26:
Hi supersibas11, 

Have you tried it without the termination characters you have declared at the beginning of the code? 

In my script I didn't require the use of termination characters.

supersibas11, 2021-01-26:
Hi Joseph,

Thank you for your answer. Yes, I just tried without and I have the same issue. I had also tried with '\r\n' but it does not seem to be the issue.

supersibas11, 2021-01-27:
I solved the issue,

I had a " , " missing in my stringArg line before .write_binary_value. That was the reason why I had headers problems.