pyvisa/pyvisa #504: How do I read until the TCPIP buffer is empty?

NitramC, 2020-04-11:
The motion controller I'm working with (PI E-873.3QTU) uses '\n' as both a delimiter and a termination character. Here is an example response to a query (white space for clarity):
axis1=10.0\n axis2=20.0\n axis3=30.0\n

I want to ignore the termination characters and simply read everything present in the buffer, then parse the data afterward (because I don't know the data length a priori). I can do this with a Python socket object by calling socket.recv().

Is there a way to do this with a pyvisa TCPIP resource?

This thread is similar: https://github.com/pyvisa/pyvisa/issues/118
But I couldn't get read_binary_values() to work for me (using pyvisa version 1.10.1). 

Help is much appreciated.

NitramC, 2020-04-11:
Note: for a SerialInstrument resource, polling bytes_in_buffer is a good solution. That property doesn't seem to be present for a TCPIP resource.

MatthieuDartiailh, 2020-04-11:
You could try to set the termination character (ie `read_termination`) to None. In which case VISA won't rely on the presence of the termchar to stop reading.

MatthieuDartiailh, 2020-04-11:
If that does not work I have another potential fix but it would be lower level so let's try that first.

NitramC, 2020-04-11:
Setting `read_termination` to `None` doesn't seem to prevent `read()` from stopping after the first '\n', but it does leave '\n' in the response. So effectively, `read()` acts just like `read_raw()`.

MatthieuDartiailh, 2020-04-11:
Can you try something along the following lines:
```
instr.read_termination = None
print(instr.visalib.read(instr.session, 1024))
```
I would like to know what status VISA reports after the read.

By the way could you share the instrument manual I could not access to it easily ? And could you post the output of `pyvisa-info` so that I know on what system and what version you are using.

NitramC, 2020-04-11:
I directly called `visalib.read()` with a chunk size of 2 and this correctly returned a warning that the "number of bytes transferred is equal to the requested input count", until only '\n' was read, at which point it claimed a successful read. Is setting `read_termination = None` is not making it's way down the chain?

Results of executing the commands you suggest:
```
In [1]: import pyvisa

In [2]: rm = pyvisa.ResourceManager()

In [3]: inst = rm.open_resource("TCPIP::192.168.1.3::50000::SOCKET")

In [4]: inst.write_termination = '\n'

In [5]: inst.read_termination = None

In [6]: inst.write('SAI?')
Out[6]: (5, <StatusCode.success: 0>)

In [7]: print(inst.visalib.read(inst.session, 1024))
---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
<ipython-input-7-e19ed74d0297> in <module>
----> 1 print(inst.visalib.read(inst.session, 1024))

C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py in read(library, session, count)
   1582     buffer = create_string_buffer(count)
   1583     return_count = ViUInt32()
-> 1584     ret = library.viRead(session, buffer, count, byref(return_count))
   1585     return buffer.raw[:return_count.value], ret
   1586

C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py in _return_handler(self, ret_value, func, arguments)
    191
    192         if ret_value < 0:
--> 193             raise errors.VisaIOError(ret_value)
    194
    195         if ret_value in self.issue_warning_on:

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

In [8]: type(inst)
Out[8]: pyvisa.resources.tcpip.TCPIPSocket
```

pyvisa-info output:
```
λ pyvisa-info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 42 Stepping 7, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\programdata\anaconda3\python.exe
   Version:        3.6.8
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 21 2019 18:30:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
```
Yes it's a pain to get the user manual. Here is a temporary link: https://filebin.net/6p8aom6am57u9qsb

I'm not sure how to upload files here.

NitramC, 2020-04-11:
For clarity, this is what the 'SAI?' command returns:
```
In [2]: import pyvisa

In [3]: rm = pyvisa.ResourceManager()

In [4]: inst = rm.open_resource("TCPIP::192.168.1.3::50000::SOCKET")

In [5]: inst.write_termination = '\n'

In [6]: inst.read_termination = None

In [7]: inst.write('SAI?')
Out[7]: (5, <StatusCode.success: 0>)

In [8]: inst.read_bytes(8)
Out[8]: b'1 \n2 \n3\n'
```

NitramC, 2020-04-11:
Ok, found that the `read_termination` setter (line 83 pyvisa.resources.messagebased.py) is behaving as it should for the value `None`. i.e. it sets `VI_ATTR_TERMCHAR` to the default, which is a linefeed.

I set the termination character to '$' since it wasn't in the string (confirmed `VI_ATTR_TERMCHAR` changed), and now `read()` times out, rather than returning part of the response with the '\n' embedded. I then set `VI_ATTR_TERMCHAR_EN` to both `VI_FALSE` and `VI_TRUE` with no effect.

Seems like `read()` is still waiting for some response and timing out, rather than just returning what's presently in the buffer. Is there a `VI_ATTR` that I can directly set to change this behavior?

MatthieuDartiailh, 2020-04-11:
First reading the manual of your instrument VISA is not mentioned and it is not obvious the instrument use that standard. Second it seems the manufacturer provides Python bindings that you could use to communicate with your instrument.
That being said the manual says that when a response is on multiple lines the linefeed will be preceded by a space except on the last line. You could hence use something like the following to achieve your goal.
```python
instr.write("SAI?")
response = ""
while True:
    response += instr.read()
    if response[-1] != " ":
         break
```

NitramC, 2020-04-11:
I think that's a great way to deal with the issue. Thanks!

closed 2020-04-11 by NitramC