pyvisa/pyvisa #505: Rohde & Schwartz RTB2004 Oscilloscope w/ PyVISA. Acquiring time value moment voltage drops.

JosephGeorges, 2020-04-27:
I am trying to automate my R&S RTB2004 scope to set channels, triggers, cursors, etc. automatically. I have almost everything complete.

From reading the instruments manual, I'm able to do all of the things I described above. But what I want is to set a vertical cursor at the time X1 (in seconds) as soon as there is a drop in voltage.

Example: I have a constant, DC voltage being read at 13.5V, when somebody disconnects the power supply, the voltage drops to 0V of course. Is there a way to implement my PyVISA script to set the first vertical cursor at the exact time x1 in which the voltage starts to drop. 

I've tried to see if there are commands which allow me to query the time-value given a certain voltage but I'm not sure if this command exists. Thanks in advance!

MatthieuDartiailh, 2020-04-27:
Since I do not have access to your instrument I may not be able to help, but I can try. Could you share your instrument manual ? Also for that kind of question you may have a better luck posting on a R&S forum since this is not PyVISA specific.

JosephGeorges, 2020-04-27:
Sure, here is the manual (You can find the remote commands in Section 15). Thank you, I will try a R&S forum as well.

[RTB_UserManual_en.pdf](https://github.com/pyvisa/pyvisa/files/4541054/RTB_UserManual_en.pdf)

MatthieuDartiailh, 2020-04-27:
Looking at the manual it seems that you should have a look at the search functionality (in your case looking for a falling slope with a meaningful threshold, maybe 12V in your case). Next you can retrieve the time at which the event occurred using `SEARch:RESult:ALL?` and properly parsing the output. Hope this helps.

JosephGeorges, 2020-04-28:
Thank you for that information Matthieu, it looks promising. I won't be able to test it until tomorrow since I am not in the lab today.

JosephGeorges, 2020-04-29:
Hey Matthieu, 

I tried the commands you suggested on the scope. Initially I tried: 
```
RTB.write("SEARch:TRIGger:EDGE:LEVel 13.4")
RTB.query("SEARch:RESult:ALL??"))
```
I want to search for a level of 13.4V and get the time value of when it happens. But I got a timeout error:

> pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

I then tried the command:` print(RTB.query("SEARch:RCOUNT?"))` to see if I am getting an search results. That command returned 0 which I assume means the search is not really working.

MatthieuDartiailh, 2020-04-29:
Reading the manual you need some extra steps:
- SEARch:CONDition EDGE
- SEARch:SOURce CH1  (assuming your signal is on channel 1)
- SEARch:TRIGger:EDGE:SLOPe NEG
- SEARch:TRIGger:EDGE:LEVel 13.4
- SEARch:TRIGger:EDGE:LEVel:DELTa 0.2  (depends on your noise level see the documentation)
- SEARch:STATe ON

With all that your search should be properly configured and enabled !

JosephGeorges, 2020-04-29:
Worked great! Thank you for the help!

MatthieuDartiailh, 2020-04-29:
Happy to help. Since this answers your question I am closing. Feel free to reopen if the need arises.

closed 2020-04-29 by MatthieuDartiailh