pyvisa/pyvisa #506: pyvisa no longer able to load Keysight VISA in Python 3.8 on Windows

hognala, 2020-05-01:
Python 3.8 changes .dll load behavior. See https://github.com/Toblerity/Fiona/issues/851

This used to work:
```
import pyvisa
rm = pyvisa.ResourceManager('ktvisa32')
```

... but now it doesn't. The problem is that Keysight VISA stores a bunch of its .dll's in a separate directory (not c:\windows\system32, but C:\Program Files (x86)\Keysight\IO Libraries Suite\bin) that is on the PATH. With the change in behavior, the PATH is no longer searched and so the ktvisa32.dll .dll fails to load because its dependant .dll's can't be found.

As a workaround, users can use the following syntax:

**32-bit Python:**
```
import os
os.add_dll_directory('C:\\Program Files (x86)\\Keysight\\IO Libraries Suite\\bin')
import pyvisa
rm = pyvisa.ResourceManager('ktvisa32')
```

**64-bit Python:**
```
import os
os.add_dll_directory('C:\\Program Files\\Keysight\\IO Libraries Suite\\bin')
import pyvisa
rm = pyvisa.ResourceManager('ktvisa32')
```
 
Feel free to reach out to me if you need someone to test a fix, or something (alan_copeland@keysight.com)

MatthieuDartiailh, 2020-05-01:
Thanks for the report. I must say I missed this change in Python 3.8.

We can attack this in multiple ways. The first is simply to document this new behavior since the fix is quite straightforward. The second, which goes with this the first, is to add the possibility to specify additional dll path in the pyvisa configuration file. Those two are low maintenance and apply to more than Keysight.
A third thing we could is to systematically add the Keysight folder on the path if it exists and we are on Windows.

Out of curiosity, what is the usecase that requires to specifically load the Keysight VISA in this way rather than the generic VISA.

If you can provide a patch that would be great. Otherwise it may need to wait a couple of weeks since I am finishing a HUGE update (code style, dosctring, format, linting, typing), those things should not have ended up being bunched together but it is too far along to stop. If you are available to test the patch once it is ready I would appreciate it too.

hognala, 2020-05-04:
Option 2 seems to make the most sense (extra search paths in configuration file).

Do you have standard configuration files that you use for the different vendor's or just the info that is in the documentation? 

Why do we want to load Keysight VISA this way? On many of our systems we have multiple vendors' VISA installed, so we want to control which VISA it gets sent to (_to verify a specific behavior for example_). In this specific case, the user just wanted to be sure it would work the same way on all users' systems, and his usecase ensures that Keysight VISA is installed, so he explicitly uses Keysight VISA.

I'll see if someone can work on the patch!

MatthieuDartiailh, 2020-05-04:
For the configuration file, I would just add an extra field (extra_dll_paths or similar) to the file documented.

hognala, 2020-05-06:
Let me know what you think of the PR. It allows the user to specify dependency paths in the .pyvisarc file

hognala, 2020-05-06:
Documenting some things I found while searching the issue: 

- The change was documented in the change notes for Python 3.8 here - https://docs.python.org/3/whatsnew/3.8.html#bpo-36085-whatsnew
- os.add_dll_directory has some interesting commentary - https://docs.python.org/3/library/os.html

MatthieuDartiailh, 2020-05-06:
Thanks for the PR. I left a couple of comments on it.

hognala, 2020-05-07:
I think all the comments had been resolved

MatthieuDartiailh, 2020-05-09:
The PR has been merged. Closing

closed 2020-05-09 by MatthieuDartiailh

MarekPetr, 2021-01-18:
Using pyvisa-1.11.3 (installed using pip) I still had to use this code `os.add_dll_directory('C:\\Program Files\\Keysight\\IO Libraries Suite\\bin')` to avoid getting the `VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.` I have even tried to install the version [1.10.0.rc](https://github.com/pyvisa/pyvisa/releases/tag/1.10.0.rc) with no success.

MatthieuDartiailh, 2021-01-18:
As per the discussion in this issue, it was decided to simply document the behavior and add the possibility to provide the additional path in the configuration file. No changes (that would have been Keysight and Windows specific) have been made to PyVISA.

If you think the documentation and/or the error message need to be improved feel free to open a PR. Adding additional details to the error message that would be Windows specific sounds perfectly doable for example.