pyvisa/pyvisa #507: Modern Equivalent to GPIB.write() from Pyvisa 1.4

FolkSong, 2020-05-02:
Hi, I've been struggling for hours to get some code written for pyvisa 1.4 working on the latest release. The line that fails is this:
```
SG.write(':MMEM:DATA "ARBI:{}",#{}{}'.format(sig_name[0:24], len(str(data_len)), data_len) + i_data)
```
With the new version, this gives me the error

> UnicodeDecodeError: 'ascii' codec can't decode byte 0xd1 in position 42: ordinal not in range(128)

i_data is read from a binary file and starts like this: ' \x00/\xd16f0\x0e \x8d\x10\xc8\t\xa2\x0f\x04\x1d'

The closest I've come is to replace it with
```
SG.write(':MMEM:DATA "ARBI:{}",#{}{}'.format(esg_name[0:24], len(str(data_len)), data_len) )
SG.write_raw(i_data)
```
This runs and does send something to the instrument, but it hangs for a while after the transfer and there are many errors flagged by the instrument.

I did read the "migrating from <1.5" article but didn't find the solution. Do you know what would be the exact equivalent of the old .write() function?

MatthieuDartiailh, 2020-05-02:
The exact equivalent would be to do:
```python
SG.write_raw(b':MMEM:DATA "ARBI:{}",#{}{}'.format(sig_name[0:24], len(str(data_len)), data_len) + i_data + self.write_termination.encode("ascii"))
```

However if you are writing binary data I would recommend in general using the `write_binary_values` function for which you don't need to manually encode your data before hand nor format the header yourself. In your case you would something like:
```python
SG.write_binary_values(':MMEM:DATA "ARBI:{}",''.format(sig_name[0:24]), i_data, datatype='b')
```

FolkSong, 2020-05-04:
Thanks very much Matthieu. I can confirm the raw command works with my existing code.

I'll look at moving to write_binary_values for future development.

closed 2020-05-04 by FolkSong