pyvisa/pyvisa #508: [COM] Communication issue with Agilent E1411B dmm in VXI crate over lan gateway

garlick, 2020-05-02:
Warning: this might be a dumb question!

My setup is
```
                      10.0.2.181                addr=9                addr=3
linux box <----> ics8065 lan-gpib gw <----> E1406A controller <----> E141B dmm
         ethernet                     gpib                vxi-backplane
```

I was expecting this VISA address to work:
```
TCPIP::10.0.2.181::gpib0,9,3::INSTR
```
because I'm able to do an `*idn?` query using that IP (10.0.2.181) and device name (`gpib0,9,3`) on home grown vxi11 software:
```
HEWLETT-PACKARD,E1411B,0,A.08.01
```
as I understand it the device name is interpreted by the ics8065 lan-gpib gw, which then communicates with gpib address 9, secondary address 3.

Am I constructing the VISA address right?  It doesn't seem to work:
```
$ pyvisa-shell

Welcome to the VISA shell. Type help or ? to list commands.

(visa) open TCPIP::10.0.2.181::gpib0,9,3::INSTR
TCPIP::10.0.2.181::gpib0,9,3::INSTR has been opened.
You can talk to the device using "write", "read" or "query".
The default end of message is added to each message.
(open) query *idn?
VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
```
It's not clear to me what's going wrong or how to debug.  I'll keep trying, but any quick thoughts on how to proceed?

Thanks!

Instrument details
------------------
* Model: Agilent E1406A VXI slot 0 controller (gpib addr 9,0), E1411B DMM (gpib addr 9,3)
* Communication: via ICS 8065 ethernet-GPIB controller

Output of `pyvisa-info`
-----------------------
```
$ pyvisa-info
Machine Details:
   Platform ID:    Linux-4.15.0-99-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.9
   Compiler:       GCC 8.4.0
   Bits:           64bit
   Build:          Apr 18 2020 01:56:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
$
```

MatthieuDartiailh, 2020-05-02:
The issue here is that currently `pyvisa-py` that you use as a backend does not support custom lan_device_name which explains the failure you are seeing. 
I never looked into supporting it and hence I have no idea how much work it would be to get it working. I will open an issue to track it on `pyvisa-py` issue tracker.

One alternative in your case could be to install Keysight VISA library for Linux (which is free) and use it as backend. 

Let me know if you figure something out.

garlick, 2020-05-02:
Ah, thanks for pointing that out!  (Saves me time).  I'll have a deeper look into pyvisa-py.

garlick, 2020-05-03:
The issue was resolved with changes in pyvisa-py so closing this.  Thanks!

closed 2020-05-03 by garlick