pyvisa/pyvisa #510: The function write_raw returns differently in WIndows and Linux

EvgenyVainshtok, 2020-05-08:
The return of function `write_raw` is the number of written bytes. 
The written message is "*IDN?". In Windows, the return is 9, in Linux Ubuntu 18, the return is 20

To Reproduce
--------------
Steps to reproduce the behavior:
### Ubuntu
`import visa
rs = visa.ResourceManager()
rs.list_resources()
('USB0::65535::37376::802243020737510034::0::INSTR',)
device = rs.open_resource('USB0::65535::37376::802243020737510034::0::INSTR')

device.query('*IDN?')
'B&K Precision, 9201, 802243020737510034,  1.10-1.06\n'

device.write('*IDN?')
(20, <StatusCode.success: 0>)
`
### Windows
`import visa
rs = visa.ResourceManager()
rs.list_resources()
('USB0::0xFFFF::0x9200::802243020737510034::INSTR', 'TCPIP0::172.16.0.42::N6705C::INSTR', 'ASRL8::INSTR', 'ASRL10::INSTR', 'ASRL28::INSTR')
device = rs.open_resource('USB0::65535::37376::802243020737510034::0::INSTR')
device.query('*IDN?')
'B&K Precision, 9201, 802243020737510034,  1.10-1.06\n'
device.write('*IDN?')
(7, <StatusCode.success: 0>)
device.read()
'B&K Precision, 9201, 802243020737510034,  1.10-1.06\n'`

MatthieuDartiailh, 2020-05-08:
Please post the output of `pyvisa-info` on both systems. In particular I would like to know if you are using pyvisa-py on any of the system. As a side note I would recommend to switch from importing from visa to importing from pyvisa. Importing from visa can conflict with the VISA payment module and as a consequence pyvisa visa.py module will be deprecated.

EvgenyVainshtok, 2020-05-08:
Thank you for the recommendation, I will implement and check it.

`Machine Details:
   Platform ID:    Linux-4.15.0-99-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.9
   Compiler:       GCC 8.4.0
   Bits:           64bit
   Build:          Apr 18 2020 01:56:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
`

`Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\evgenyv\appdata\local\programs\python\python36-32\python.exe
   Version:        3.6.4
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Dec 19 2017 06:04:45 (#v3.6.4:d48eceb)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 19923456
         Spec. Version: 5244928
   py:
      Version: 0.4.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

`

MatthieuDartiailh, 2020-05-08:
I believe that this inconsistency has been fixed in pyvisa-py 0.4.0 Could you update your linux system and check again ?

EvgenyVainshtok, 2020-05-08:
Thank you. Updating to pyvisa-py 0.4.0 helped. But I had to re-install it. For some reason just the upgrade didn't help.

closed 2020-05-08 by EvgenyVainshtok