pyvisa/pyvisa #512: VI_ERROR_RSRC_BUSY when running PyVISA + Dash

poke2201, 2020-05-15:
```import dash
import pyvisa
from pyvisa.constants import StopBits, Parity
import dash

if __name__ == '__main__':
    rm = pyvisa.ResourceManager()
    #Any resource will cause this bug
    chip_object = rm.open_resource('ASRL10::INSTR', baud_rate=115200, parity=Parity.none,
                                   stop_bits=StopBits.one, query_delay=0.02, data_bits=8, flow_control=0)
    app = dash.Dash('Test')
    app.run_server(debug=True)
```
To Reproduce:
Run code snippet in IDLE

#411 has the same issue.

 
**Output of `pyvisa-info`**
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\JArucan\Documents\Unit Certificate Generator\PressureCalAutomation\Scripts\python.exe
   Version:        3.7.4
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Jul  8 2019 20:34:20 (#tags/v3.7.4:e09359112e)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15729920
         Spec. Version: 5243904
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15729920
         Spec. Version: 5243904

poke2201, 2020-05-15:
Running multiprocessing is a possible way to get around this issue, but it would add major unneeded complexity.

MatthieuDartiailh, 2020-05-15:
I tried to reproduce with a TCPIP resource but apart from a layout issue from dash, I did not see any error related to PyVISA. Do you see the error only with serial resources similar to #411 ? Do you just see the same error if you simply run your script in a terminal ?

poke2201, 2020-05-15:
I am also using serial USB resource.  I don't have another type of instrument available to check.

poke2201, 2020-05-15:
This big also seems to disappear when you run the server before running the open_resource command.

```import dash
import pyvisa
from pyvisa.constants import StopBits, Parity

def process2():
    rm = pyvisa.ResourceManager()
    chip_object = rm.open_resource('ASRL10::INSTR', baud_rate=115200, parity=Parity.none,
                                   stop_bits=StopBits.one, query_delay=0.02, data_bits=8, flow_control=0)
    return chip_object
if __name__ == '__main__':
    app = dash.Dash("Test")
    app.run_server(debug=True)
    chip = process2()
```

poke2201, 2020-05-15:
I forgot to add my traceback:

```Traceback (most recent call last):
  File "C:\Users\JArucan\Documents\Unit Certificate Generator\PressureCalAutomation\lib\site-packages\pyvisa\highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\JArucan\Documents\Unit Certificate Generator\PressureCalAutomation\lib\site-packages\pyvisa\resources\resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Users\JArucan\Documents\Unit Certificate Generator\PressureCalAutomation\lib\site-packages\pyvisa\highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\JArucan\Documents\Unit Certificate Generator\PressureCalAutomation\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "C:\Users\JArucan\Documents\Unit Certificate Generator\PressureCalAutomation\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.```

MatthieuDartiailh, 2020-05-15:
I really have a hard time seeing why importing dash screw up serial connection. We agree that the error occur before creating the dash app and go away if you don't import dash ?

poke2201, 2020-05-15:
That's correct, the offending line is possibly `app.run_server(debug=True)`.  It looks like it blocks access to the resource.

MatthieuDartiailh, 2020-05-15:
Do you see the error if you import dash but do not start the app ? I assume yes but I want to be sure.

poke2201, 2020-05-15:
No error when you import dash but do not run the app.

MatthieuDartiailh, 2020-05-15:
Ok then I need to understand what black magic the app does. Will probably take me some time. Feel free to ping me.

MatthieuDartiailh, 2020-05-15:
Reading the documentation of dash regarding global variable I am not sure how this could/should work. The main strategy to store global is in a hidden div it seems and that won't work here. If I am wrong please let me know.

MatthieuDartiailh, 2020-05-15:
But reading further it should not affect this particular use case.

poke2201, 2020-05-15:
I'm trying to avoid using global because with the amount of variables flying around, Id like to keep track of where they are.  

Current "working" implementation uses multiprocessing.Queue to send data between the two processes to avoid blocking.  I've also removed any trace of pyVISA from my Dash coding and just left it solely to be Dash and what it gets from Queue.

I'm just now wondering if theres anything we can do here to detect blocking due to web apps like this.

poke2201, 2020-05-15:
As a side note, [heres a resource I used to figure out a "working" implementation.](https://community.plotly.com/t/how-to-integrate-dash-with-another-process-that-might-be-blocking/6098)

MatthieuDartiailh, 2020-05-16:
Thanks for the update. We could add an entry to the FAQ. I am however surprised by the fact that I do not see a hard crash on other resources. It feels like dash is duplicating the resource and serial are more sensitive because they do not support multiple connections.

poke2201, 2020-05-16:
Well, I put it as "working" because it should theoretically work, but I haven't confirmed the execution. I'll update when I do get an execution going.

Maybe we can check other resources to be sure that its limited to serial connections?

MatthieuDartiailh, 2020-05-16:
I checked with a TCPIP resource and saw no trouble.