pyvisa/pyvisa #513: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

BlueskyEnergy, 2020-05-18:
Instrument details
------------------
* Model: IT8516C
* Communication: ASRL/dev/ttyUSB2::INSTR
* Link to the documentation (if available): http://www.itech.sh/uploadfiles/2020/04/202004290929562956.pdf

Output of `pyvisa-info`
-----------------------
pyvisa-info
/opt/BatteryTeststation/venv/lib/python3.6/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-4.15.0-99-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /opt/BatteryTeststation/venv/bin/python3
   Version:        3.6.9
   Compiler:       GCC 8.4.0
   Bits:           64bit
   Build:          Apr 18 2020 01:56:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.4.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available

MatthieuDartiailh, 2020-05-19:
Can you please post an example of what you have tried so far ? Also did you look at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#getting-the-instrument-configuration-right ?

MatthieuDartiailh, 2020-05-19:
Looking at the manual you need to set the termination character to `\n` and properly set the serial properties according to the manual 1. Baud rate: 4800,9600,19200 and 38400 (9600 is the instrument default), 2. Data bit: 8 bit, 3. Stop bit: 1, 4. Parity: None.

However the manual suggest installing a specific driver on the control computer to use USB to serial and I am not sure they support linux. Have you tried installing that driver ?

Otherwise the settings can be achieved in the following way:

```python
from pyvisa import constants
rm = pyvisa.ResourceManager("@py")
it = rm.open_resource("ASRL/dev/ttyUSB2::INSTR", baud_rate=9600, data_bits=8, parity=constants.Parity.none, stop_bits=constants.StopBits.one, write_termination="\n", read_termination="\n")
```

BlueskyEnergy, 2020-05-19:
Resource Manager of Visa Library at unset

> Can you please post an example of what you have tried so far ? Also did you look at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#getting-the-instrument-configuration-right ?

from gpib_ctypes import make_default_gpib
... make_default_gpib()
... import pyvisa
... from pyvisa import constants
... rm = pyvisa.ResourceManager("@py")
... my_instrument = rm.open_resource('ASRL/dev/device_EL2::INSTR', baud_rate=9600, data_bits=8, parity=constants.Parity.none, stop_bits=constants.StopBits.one, write_termination="\n", read_termination="\n")
/opt/BatteryTeststation/venv/lib/python3.6/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
print(my_instrument.query('*IDN?'))
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/opt/BatteryTeststation/venv/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 613, in query
    return self.read()
  File "/opt/BatteryTeststation/venv/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "/opt/BatteryTeststation/venv/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/opt/BatteryTeststation/venv/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 403, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

**this is the contents of rm:**

rm = {ResourceManager} Resource Manager of Visa Library at unset
 last_status = {str} 'Traceback (most recent call last):\n  File "/opt/BatteryTeststation/venv/lib/python3.6/site-packages/pyvisa/highlevel.py", line 160, in get_last_status_in_session\n    return self._last_status_in_session[session]\nKeyError: 1049656\n\nDuring handling of the ab
 session = {int} 1049656
 visalib = {PyVisaLibrary} Visa Library at unset
  SerialSession = {ABCMeta} <class 'pyvisa-py.serial.SerialSession'>
   session_type = {tuple: 2} (<InterfaceType.asrl: 4>, 'INSTR')
  TCPIPInstrSession = {ABCMeta} <class 'pyvisa-py.tcpip.TCPIPInstrSession'>
   session_type = {tuple: 2} (<InterfaceType.tcpip: 6>, 'INSTR')
  TCPIPSocketSession = {ABCMeta} <class 'pyvisa-py.tcpip.TCPIPSocketSession'>
   session_type = {tuple: 2} (<InterfaceType.tcpip: 6>, 'SOCKET')
  USBRawSession = {ABCMeta} <class 'pyvisa-py.usb.USBRawSession'>
   session_type = {tuple: 2} (<InterfaceType.usb: 7>, 'RAW')
  USBSession = {ABCMeta} <class 'pyvisa-py.usb.USBSession'>
   session_type = {NoneType} None
  handlers = {defaultdict: 0} defaultdict(<class 'list'>, {})
   default_factory = {type} <class 'list'>
   __len__ = {int} 0
  issue_warning_on = {set: 10} {1073676418, 1073676420, 1073676421, 1073676294, 1073676424, 1073676457, 1073676300, 1073676407, 1073676443, 1073676413}
   {int} 1073676418
   {int} 1073676420
   {int} 1073676421
   {int} 1073676294
   {int} 1073676424
   {int} 1073676457
   {int} 1073676300
   {int} 1073676407
   {int} 1073676443
   {int} 1073676413
   __len__ = {int} 10
  last_status = {int} 0
  library_path = {str} 'unset'
  resource_manager = {ResourceManager} Resource Manager of Visa Library at unset
   last_status = {str} 'Traceback (most recent call last):\n  File "/opt/BatteryTeststation/venv/lib/python3.6/site-packages/pyvisa/highlevel.py", line 160, in get_last_status_in_session\n    return self._last_status_in_session[session]\nKeyError: 1049656\n\nDuring handling of the ab
   session = {int} 1049656
   visalib = {PyVisaLibrary} Visa Library at unset
  sessions = {dict: 2} {1049656: <VisaLibrary('unset')>, 5491066: <pyvisa-py.serial.SerialSession object at 0x7f67726bdb70>}
   1049656 = {PyVisaLibrary} Visa Library at unset
    SerialSession = {ABCMeta} <class 'pyvisa-py.serial.SerialSession'>
    TCPIPInstrSession = {ABCMeta} <class 'pyvisa-py.tcpip.TCPIPInstrSession'>
    TCPIPSocketSession = {ABCMeta} <class 'pyvisa-py.tcpip.TCPIPSocketSession'>
    USBRawSession = {ABCMeta} <class 'pyvisa-py.usb.USBRawSession'>
    USBSession = {ABCMeta} <class 'pyvisa-py.usb.USBSession'>
    handlers = {defaultdict: 0} defaultdict(<class 'list'>, {})
[... 164 lines cut ...]
    ATTRS{broken_parity_status}=="0"
    ATTRS{class}=="0x0c0330"
    ATTRS{consistent_dma_mask_bits}=="64"
    ATTRS{d3cold_allowed}=="1"
    ATTRS{dbc}=="disabled"
    ATTRS{device}=="0x9c31"
    ATTRS{dma_mask_bits}=="64"
    ATTRS{driver_override}=="(null)"
    ATTRS{enable}=="1"
    ATTRS{irq}=="42"
    ATTRS{local_cpulist}=="0-3"
    ATTRS{local_cpus}=="f"
    ATTRS{msi_bus}=="1"
    ATTRS{numa_node}=="-1"
    ATTRS{revision}=="0x04"
    ATTRS{subsystem_device}=="0x2054"
    ATTRS{subsystem_vendor}=="0x8086"
    ATTRS{vendor}=="0x8086"

  looking at parent device '/devices/pci0000:00':
    KERNELS=="pci0000:00"
    SUBSYSTEMS==""
    DRIVERS==""

Thanks for your time.

BlueskyEnergy, 2020-05-25:
I am still stuck with this issue, do you have any idea for what could be the problem?

MatthieuDartiailh, 2020-05-25:
From the driver info, it does not look like you have any manufacturer specific driver and since I have no idea what that driver is supposed to do, it may or may not matter. If you can I would suggest trying the native serial port of the instrument (with a USB to serial on the computer side if necessary). Otherwise I would first try to ensure that the instrument does receive messages by using a command that is supposed to produce a visible change as explained in pyvisa documentation. 
Thanks for the ping by the way.

BlueskyEnergy, 2020-05-25:
Hi,

I re-check and the driver used is the following:
*-usb:0
                   description: Generic USB device
                   product: USB-Serial Controller
                   vendor: Prolific Technology Inc.
                   physical id: 2
                   bus info: usb@2:2
                   version: 3.00
                   capabilities: usb-2.00
                   configuration: driver=pl2303 maxpower=100mA speed=12Mbit/s

It uses the driver pl2303.

I am intrigued by your suggestion and I would like to manually send requests to the device throughout a USB->DB9 cable.

It would be te first time for me, can you give me more precise instructions on how to send commands and receive data manually?

MatthieuDartiailh, 2020-05-25:
I would not be manually exactly but simply using the true serial port of your instrument and then connecting using VISA. However in order to do so you need a USB-Serial adapter and a serial cable.

It does seem you are using the proper driver indeed so I am not sure what the issue is. Since VISA does not add much on a basic serial interface you could try to manually debug using putty. The first thing to do before reading would be to ensure the instrument does understand the messages typically by using a command that should produce a visible change.

BlueskyEnergy, 2020-05-28:
Hi,

finally I found the problem:

- the boud rate set on the device was different (115800 instead of 9600).

Changing that value solve the problem for me.

Thank you for your support.

closed 2020-05-28 by BlueskyEnergy

MatthieuDartiailh, 2020-05-28:
Happy you found a solution !