pyvisa/pyvisa #514: usb.core.USBError: [Errno 110] Operation timed out

YugoAmaryl1, 2020-05-19:
Hello,
I try to communicate between a Raspberry Pi 3B+ and a Agilent Technologies E5070B Network Analyzer with USB but I get a usb.core.USBError: [Errno 110] Operation timed out.
I have test all Permutations of the end signs \n and \r but I have still the USB Error. 
I test with visa.ResourceManager() and visa.ResourceManager('@py').
What can I do?

My configuration is:

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ python3 -m visa info
Machine Details:
   Platform ID:    Linux-4.19.97-v7+-armv7l-with-debian-10.4
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Dec 20 2019 18:57:59 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

The Programm input and output is:

>>> import visa
>>> rm = visa.ResourceManager()
2020-05-18 14:01:46,330 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2020-05-18 14:01:46,330 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2020-05-18 14:01:46,887 - pyvisa - DEBUG - Automatically found library files: [None, None, None, None, None]
2020-05-18 14:01:46,890 - pyvisa - DEBUG - No user defined library files
2020-05-18 14:01:46,891 - pyvisa - DEBUG - Did not find NI binary
2020-05-18 14:01:46,891 - pyvisa - DEBUG - pyvisa-py is available.
2020-05-18 14:01:46,891 - pyvisa - DEBUG - Created ResourceManager with session 5725689
>>> rm.timeout = 100000
>>> rm.write_termination = '\n\r'
>>> rm.read_termination = '\n\r'
>>> os = rm.open_resource('USB0::2391::1033::MY42401470::0::INSTR')
2020-05-18 14:03:09,289 - pyvisa - DEBUG - USB0::2391::1033::MY42401470::0::INSTR - opening ...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/sessions.py", line 213, in __init__
    self.after_parsing()
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/usb.py", line 201, in after_parsing
    self.parsed.serial_number)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 265, in __init__
    self._get_capabilities()
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 282, in _get_capabilities
    timeout=self.timeout)
  File "/usr/local/lib/python3.7/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.7/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python3.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

MatthieuDartiailh, 2020-05-19:
First can you try to update pyvisa-py to 0.4.0 ?

YugoAmaryl1, 2020-05-26:
Now it is PyVISA Backend py 0.4.0 but it does not help.
I get still usb.core.USBError: [Errno 110] Operation timed out

What can I do?
----------------------------------------------------------------------
Debug output below:

python -m visa info
Machine Details:
   Platform ID:    Linux-4.19.97-v7+-armv7l-with-debian-10.4
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.16
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Oct 10 2019 22:02:15 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.4.0
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named gpib
      TCPIP SOCKET: Available 

---------------------------------------------------------------------

2020-05-26 11:18:26,940 - pyvisa - DEBUG - SerialSession was correctly imported.
2020-05-26 11:18:27,077 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2020-05-26 11:18:27,109 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2020-05-26 11:18:27,113 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2020-05-26 11:18:27,114 - pyvisa - DEBUG - Created library wrapper for unset
2020-05-26 11:18:27,115 - pyvisa - DEBUG - Created ResourceManager with session 2623823
2020-05-26 11:18:27,115 - pyvisa - DEBUG - USB0::2391::1033::MY42401470::0::INSTR - opening ...
Traceback (most recent call last):
  File "osci.py", line 7, in <module>
    os = rm.open_resource('USB0::2391::1033::MY42401470::0::INSTR')
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/sessions.py", line 213, in __init__
    self.after_parsing()
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/usb.py", line 201, in after_parsing
    self.parsed.serial_number)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 265, in __init__
    self._get_capabilities()
  File "/usr/local/lib/python3.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 282, in _get_capabilities
    timeout=self.timeout)
  File "/usr/local/lib/python3.7/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.7/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python3.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
2020-05-26 11:18:28,543 - pyvisa - DEBUG - Closing ResourceManager (session: 2623823)

YugoAmaryl1, 2020-05-26:
Sorry, in my last post was a mistake. Now I hav used py: Version: 0.4.0 with Python 2.7.16
But I get also the Errno 110

What can I do?
--------------------------------------------------------------------------------------------------
sudo python osci.py
2020-05-26 12:13:37,820 - pyvisa - DEBUG - SerialSession was correctly imported.
2020-05-26 12:13:37,934 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2020-05-26 12:13:37,960 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2020-05-26 12:13:37,966 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2020-05-26 12:13:37,967 - pyvisa - DEBUG - Created library wrapper for unset
2020-05-26 12:13:37,968 - pyvisa - DEBUG - Created ResourceManager with session 9389499
2020-05-26 12:13:37,969 - pyvisa - DEBUG - USB0::2391::1033::MY42401470::0::INSTR - opening ...
Traceback (most recent call last):
  File "osci.py", line 7, in <module>
    os = rm.open_resource('USB0::2391::1033::MY42401470::0::INSTR')
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/sessions.py", line 218, in __init__
    self.after_parsing()
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 178, in after_parsing
    self.parsed.serial_number)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 294, in __init__
    self._get_capabilities()
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 311, in _get_capabilities
    timeout=self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

MatthieuDartiailh, 2020-05-26:
Since your issue is directly in opening the resource, bot in talking to it, it looks to me more like a device configuration issue. From Pyvisa-py documentation:
```
On Unix system, one may have to modify udev rules to allow non-root access to
the device you are trying to connect to. The following tutorial describes how
to do it http://ask.xmodulo.com/change-usb-device-permission-linux.html.

Another useful reference for how to configure your system is h
ttps://github.com/python-ivi/python-usbtmc.
```

YugoAmaryl1, 2020-05-27:
I cannot access http://ask.xmodulo.com/change-usb-device-permission-linux.html
I get the message: This Account has been suspended.

MatthieuDartiailh, 2020-05-27:
The second link is still valid. I will update the documentation when I get the time.

YugoAmaryl1, 2020-05-29:
[

YugoAmaryl1, 2020-06-09:
I create the usbtmc.rules and start the communication once with "python oscy.py" and once with "sudo python oscy.py", but I get still the error  "usb.core.USBError: [Errno 110] Operation timed out". Hopefully my usbtmc is correct.

What can I do?

[Oscy.py_and_usbtmc.rules.zip](https://github.com/pyvisa/pyvisa/files/4751094/Oscy.py_and_usbtmc.rules.zip)

MatthieuDartiailh, 2020-06-09:
How did you pick the manufacturer ID and product ID used in the udev rules ? I would expect them to match the Manufacturer ID and model code found in the USB address since they refer to the same attributes.

YugoAmaryl1, 2020-06-10:
Hello, I get the IDs with the "lsusb" command.

[image]