pyvisa/pyvisa #515: Pyvisa + PYUSB Set up with LibUSB backend but still not seeing resource

quasarsnax, 2020-05-20:
Device is Anritsu S331P SCPI Command ready (USB connection)
LibUSB installed via Zadig after selecting S331P in dropdown 

**IDLE Code**
>>>import visa
>>> import usb
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
** Result edited for Path Removed**  
VisaIOWarning: VI_WARN_CONFIG_NLOADED (1073676407): The specified configuration either does not exist or could not be loaded. VISA-specified defaults will be used.
  ret = library.viOpenDefaultRM(byref(session))
>>> rm.list_resources()
()

**Pyvisa-Info result**

c:\users\alexanderlong\appdata\local\programs\python\python38\lib\site-packages\pyvisa\ctwrapper\functions.py:1227: VisaIOWarning: VI_WARN_CONFIG_NLOADED (1073676407): The specified configuration either does not exist or could not be loaded. VISA-specified defaults will be used.
  ret = library.viOpenDefaultRM(byref(session))
Machine Details:
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 12, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\alexanderlong\appdata\local\programs\python\python38\python.exe
   Version:        3.8.3
   Compiler:       MSC v.1924 64 bit (AMD64)
   Bits:           64bit
   Build:          May 13 2020 22:37:02 (#tags/v3.8.3:6f8c832)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 16777216
         Spec. Version: 5244672
      #2: C:\windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 16777216
         Spec. Version: 5244672
   py:
      Version: 0.4.0
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb0
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb0
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

MatthieuDartiailh, 2020-05-24:
Since NI-VISA is installed LibUSB takes no part in the process. Can you find your device in NI-MAX ?

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh