pyvisa/pyvisa #517: NI GPIB-ENET/1000 not detected from pyVISA, library missing

MassimilianoAntonello, 2020-05-25:
Goodmorning, I'm trying to comunicate with a Kehytley 2410 trough a NI GPIB-ENET/1000 that I can correctly ping. I have also installed the NI-VISA. I have a centos7 Linux machine.

Output of `pyvisa-info`
-----------------------
/usr/local/lib/python3.6/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-3.10.0-1127.el7.x86_64-x86_64-with-centos-7.8.2003-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.8
   Compiler:       GCC 4.8.5 20150623 (Red Hat 4.8.5-39)
   Bits:           64bit
   Build:          Apr  2 2020 13:34:55 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1
Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.19.2.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 19923456
         Spec. Version: 5244928
   py:
      Version: 0.4.0
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available `
-----------------------------
As shown above I don't have the GPIB library so I suppose it is why I can't see any GPIB when i do:
` rm = pyvisa.ResourceManager()`
` rm.list_resources()`
but I don't understand how to solve it. I have already tried to re-install both NI-VISA and pyvisa.
Thank you.

MassimilianoAntonello, 2020-05-25:
Sorry I have to upload and be more clear.
I have a KEYTHLEY 4210 connected via GPIB to a NI GPIB-ENET/1000 connect trough LAN cable to the department network.
I tried to do:
'rm.open_resource('TCPIP::hostname::IPaddress)' 
but I have: VI_ERROR_RSRC_NFOUND.
Thanks

MatthieuDartiailh, 2020-05-25:
First I don't have any experience with GPIB-ENET adapters, so I may not have the perfect answer for you. 
I am assuming you followed the installation instruction found here http://www.ni.com/pdf/manuals/370426r.pdf , if not try to have a look since the device is unlikely to work before you do.
Do you have access to NI-MAX on CentOS ? If yes I would suggest starting by checking which instrument are listed there ? Usually GPIB interfaces like this one, try to be transparent and I would assume that you could connect to your device using directly its GPIB address without needing to refer to the IP address of the adapter. If you don't have access to NI-MAX try `rm.open_resource("GPIB::?::INSTR")` with ? the GPIB address of your Keithley.

MatthieuDartiailh, 2020-05-25:
You won't be using pyvisa-py since you have NI-VISA but the error about GPIB library may mean you forgot to install one part of NI-VISA (https://www.ni.com/en-us/support/downloads/drivers/download.ni-488-2.html#306164) which I am sure does not come with NI-VISA on Mac and is likely also omitted on LInux.

MassimilianoAntonello, 2020-05-25:
Thank you for your replies.
How I can obtain the GPIB of my Keithley?
Sorry but this is my first time ever with remote connections of instrumentation.
Unfortunately for linux seems NI-MAX is not available, anyway I have installed ni-488-2 but I have the same output.

MatthieuDartiailh, 2020-05-25:
Using your instrument front panel you can access the GPIB address from the COMMUNICATIONS section (see manual https://download.tek.com/manual/2400S-900-01_K-Sep2011_User.pdf). After installing NI-488-2 did you, as mentioned in  http://www.ni.com/pdf/manuals/370426r.pdf, add the GPIB hardware via the GPIB explorer utility ?
Once you have done the above `rm.open_resource("GPIB::?::INSTR")` should work.

MassimilianoAntonello, 2020-05-25:
It works now! Thank you so much

MatthieuDartiailh, 2020-05-25:
Happy you got it to work !

MassimilianoAntonello, 2020-05-26:
I'm sorry but today I tired again to connect and it works fine but at certain point I had again the same problem: I'm not able to see the GPIB connection.
Unfortunately I tried to remove and create again a new interface with the GPIB explorer of NI-488.2 but I can recognize it and add to my devices even if I can not see it when I do in PyVISA rm.list_resources().

closed 2020-05-26 by MassimilianoAntonello