pyvisa/pyvisa #518: VisaIOError: VI_ERROR_TMO when reading from TCPIP SOCKET

isanch, 2020-06-07:
Hello,

I am trying to communicate with a Smaract positioner  via an TPCIP SOCKET. I am able to connect  and write to the device but I am not able to read from it (either query or read, write works well). 

Here is my code:

```
import pyvisa
rm = pyvisa.ResourceManager() 
inst_name = 'TCPIP0::169.254.1.101::55551::SOCKET'
inst = rm.open_resource(inst_name)
inst.query("*IDN?")
```

And I get the following error:

```
  File "<ipython-input-39-5d0747d02c23>", line 1, in <module>
    inst.query("*IDN?")

  File "C:\Users\Lab\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 617, in query
    return self.read()

  File "C:\Users\Lab\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 431, in read
    message = self._read_raw().decode(enco)

  File "C:\Users\Lab\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 401, in _read_raw
    chunk, status = self.visalib.read(self.session, size)

  File "C:\Users\Lab\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1593, in read
    ret = library.viRead(session, buffer, count, byref(return_count))

  File "C:\Users\Lab\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

I am able to track the commands I send via python in the NI MAX I/O trace, and check the outputs and the buffer content. The buffer content looks as  it should (the IDN is correct) but still gives the time out error.  

[image]

Here my pyvisa-info:

```
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\Lab\Anaconda3\python.exe
   Version:        3.7.7
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          May  6 2020 11:45:54 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 19922944
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 19922944
         Spec. Version: 5244928
   py:
      Version: 0.4.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

Any ideas on what  the problem is? If I connect to the device directly via the NI MAX I have no communication issues. Thanks for any info.

MatthieuDartiailh, 2020-06-07:
Try setting the termination character used for reading (attribute `read_termination`). `\n` is a usual value in particular for SOCKET resources, but you can probably just check the last character returned in NI IO trace. You can also have a look at the documentation and in particular at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html

isanch, 2020-06-07:
Thanks for your answer. Setting inst.read_termination = '\n' or inst.read_termination = '\r\n' or solved the issue.

closed 2020-06-07 by isanch