pyvisa/pyvisa #519: [COM] Communication issue with Rigol DM3058 using TCP/IP

Oscarillo, 2020-06-09:
Instrument details
------------------
* Model:
* Communication: TCPIP, GPIB, ....
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------

Oscarillo, 2020-06-09:
This is my code in python, very simple
import sys
import visa			#pip install pyvisa-py
import time
from datetime import date
from datetime import datetime
import csv
import subprocess
import socket

rigol = visa.ResourceManager("@py")
print(rigol.list_resources())
multimeter = rigol.open_resource("TCPIP0::192.168.1.38::inst0::INSTR")
print(multimeter)
#multimeter.timeout = 10000
multimeter.read_termination = '\n'
multimeter.write_termination = '\n'
print(multimeter.query('*IDN?'))
print(multimeter.query(":MEASure:CURRent:DC?\n"))
multimeter.close()

Everything works until it gets to 
print(multimeter.query(":MEASure:CURRent:DC?\n"))
Then it trows the error

I had the multimeter connected directly to the pc and use
multimeter = rigol.open_resource("TCPIP0::10.0.0.3::inst0::INSTR")

Same problem
I have used 3 different RJ45 cables, the same problem
I've installed NI MAX and it has the same problem when I try to use a read command
I even installed the vxi11 library and the same problem, only when it tries to read from the multimeter
I even connected by USB and the same problem, only when it reads throws the error in every library or NI MAX
multimeter = rigol.open_resource('USB0::0x1AB1::0x09C4::DM3L204900833::INSTR')
My PC is a Windows 10 latest update (till today)
Paython 3.8.3

Any help would be greatly appreciated.
PS
write commands work fine.

ssn365, 2020-06-18:
I do have the same issue with rigol device. I am able to use write commands but read and query(which a combination of write and read) are not working.

I attach my code here:

Python 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 09:44:33) [MSC v.1900 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm = visa.ResourceManager('C:/Program Files (x86)/IVI Foundation/VISA/WinNT/agvisa/agbin/visa32.dll')
>>> inst = rm.open_resource('USB0::0x1AB1::0x0643::DG8A220800316::0::INSTR')
>>> print(inst.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 613, in query
    return self.read()
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

Machine Details:
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Program Files (x86)\Python37-32\python.exe
   Version:        3.7.7
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 10 2020 09:44:33 (#tags/v3.7.7:d7c567b08f)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928
   py:
      Version: 0.4.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb0
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb0
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionality.
         No module named 'gpib'

MatthieuDartiailh, 2020-06-18:
@ssn365 please see my reply in #521.

@Oscarillo sorry for the long delay I must have missed your post. First note that when specifying `write_termination` you do not need to append manually the termchar anymore. As a consequence since the `*IDN?` call works but not the next one, I would try to remove the extra `\n` you have there. (print(multimeter.query(":MEASure:CURRent:DC?\n")))

If I misunderstood and even the `*IDN?` does not work, then it is more complicated. You mention NI-MAX, can you read the `*IDN?` using it ?

Oscarillo, 2020-06-29:
Thanks for the reply, and I apologize for the late reply. 
I've found the problem and it had nothing to do with the library.
The Rigol DM3058 has 3 command settings RIGOL|AGILENT|FLUKE and with the fluke is when I have the reading problems, once I change it to RIGOL or AGILENT it worked and I found that the most compatible is RIGOL.

MatthieuDartiailh, 2020-06-29:
Thanks for the feedback @Oscarillo. That is very good to know and may help others.

closed 2020-06-29 by MatthieuDartiailh