pyvisa/pyvisa #520: [COM] Communication issue with Keysight E5061A VI_ERROR_TMO

tshawnz, 2020-06-17:
Instrument details
------------------
* Model: E5061A
* Communication: TCPIP
* Link to the documentation (if available):http://literature.cdn.keysight.com/litweb/pdf/E5061-90042.pdf

Output of `pyvisa-info`
Python:
   Implementation: CPython
   Executable:     C:\Program Files\Python37\python.exe
   Version:        3.7.7
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 10 2020 10:41:24 (#tags/v3.7.7:d7c567b08f)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377175728
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377175728
         Spec. Version: 5244928
-----------------------

Hi, I'm trying to use read() function to get raw binary information from the network analyzer. I tried read() read_raw() and readbytes() all with termination character **'\n'**

Here's what I got from the pyvisa log when I'm doing read():
**above data omitted** 
x00\n')    **end of data**
Traceback (most recent call last):
  File "<pyshell#3>", line 1, in <module>
    gettrace()
  File "C:\Users\TDT\Desktop\CommandSeq\Codetest\Binaryraw.py", line 32, in gettrace
    SCPI_E5061.query_binary_values("CALC1:DATA:FDAT?")
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\resources\messagebased.py", line 715, in query_binary_values
    data_points, chunk_size)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\resources\messagebased.py", line 520, in read_binary_values
    chunk_size=chunk_size))
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\resources\messagebased.py", line 356, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

It seems that I get the string form of the binary data and it shows that the termination character is '\n' but the timeout still happens.
Do I need to specify the chunk size or use other functions?

MatthieuDartiailh, 2020-06-17:
When using `query_binary_values` the system will determine the number of bytes to read from the header of binary block. Usually Keysight uses the IEEE standard `#XYYYY` where X is the number of bytes on which the length is encoded and Y are ascii coded numbers.
Depending on the instrument the termination character may or may not be added at the end of the block. By default PyVISA expects it to be added in which cases it try to read the extra character.
`read` can return early since your binary block can contain \n at arbitrary positions. You can find a detailed explanation in the documentation at https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html
To fix your issue, you can:
- use `read_raw` repeatedly till you get a timeout and check if the termination character is present at the end of the block / if the block is of the advertised size.
- if the termination character is indeed omitted pass `expect_termination=False` to `query_binary_values`.

tshawnz, 2020-06-24:
Thanks a lot! That solves my issue.

closed 2020-06-24 by tshawnz