pyvisa/pyvisa #521: Update to use read and query commands to communicate with RIGOL DG811 device

ssn365, 2020-06-18:
I am unable to run read and query commands on Rigol Device. I can run the same query on NI VISA test panel and communicate successfully but I fail when I do it using pyvisa.
Note: I don't face any issues when I use write command. it is only when I use read and query commands.

I attach my code below for reference: 

>>> import visa
>>> rm = visa.ResourceManager('C:/Program Files (x86)/IVI Foundation/VISA/WinNT/agvisa/agbin/visa32.dll')
>>> inst = rm.open_resource('USB0::0x1AB1::0x0643::DG8A220800316::0::INSTR')
>>> print(inst.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 613, in query
    return self.read()
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

Instrument details
------------------
* Model:  Rigol Technologies,DG811,
* Communication: USB
* Link to the documentation (if available): https://int.rigol.com/Public/Uploads/uploadfile/files/20191118/20191118102948_5dd2021c6039d.pdf

Output of `pyvisa-info`
-----------------------

MatthieuDartiailh, 2020-06-18:
It looks like a usual issue of termination character to me. Looking at the manual, things are not very clear but VisualBasic example seems to suggest that `\n` is used as termchar. Can you try the following ?
```python
import pyvisa
rm = pyvisa.ResourceManager('C:/Program Files (x86)/IVI Foundation/VISA/WinNT/agvisa/agbin/visa32.dll')
inst = rm.open_resource('USB0::0x1AB1::0x0643::DG8A220800316::0::INSTR')
inst.write_termination = '\n'
inst.read_termination = '\n'
print(inst.query("*IDN?"))
```

ssn365, 2020-06-19:
Hello,
I did try the code you have provided and i get Timeout .

>>> import pyvisa
>>> rm = pyvisa.ResourceManager('C:/Program Files (x86)/IVI Foundation/VISA/WinNT/agvisa/agbin/visa32.dll')
>>> inst = rm.open_resource('USB0::0x1AB1::0x0643::DG8A220800316::0::INSTR')
>>> inst.write_termination = '\n'
>>> inst.read_termination = '\n'
>>> print(inst.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 613, in query
    return self.read()
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

I also attach the NI Visa Test Panel output which is working fine.
[image]

MatthieuDartiailh, 2020-06-19:
So it does look like the termination characters are `\n` however I notice that you manually specify a rather old version of agilent VISA rather than letting pyvisa find your NI VISA install automatically. Rigol devices are known to have some quirks on USB and that old Agilent may not handle them but your recent NI apparently does. Can you try the following ?
```python
import pyvisa
rm = pyvisa.ResourceManager()
inst = rm.open_resource('USB0::0x1AB1::0x0643::DG8A220800316::0::INSTR')
inst.write_termination = '\n'
inst.read_termination = '\n'
print(inst.query("*IDN?"))
```

ssn365, 2020-06-24:
Hello again @MatthieuDartiailh ,
Thank you for your response. I have tried the code you have provided and it is not working. I attach the response.

>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
>>> inst = rm.open_resource('USB0::0x1AB1::0x0643::DG8A220800316::0::INSTR')
>>> inst.write_termination = '\n'
>>> inst.read_termination = '\n'
>>> print(inst.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 613, in query
    return self.read()
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_INP_PROT_VIOL (-1073807305): Device reported an input protocol error during transfer.

MatthieuDartiailh, 2020-06-24:
Honestly I have never this specific error before. Do you observe the same issue with your other USB instruments or is the issue limited to the Rigol ? Did you try turning the instrument off/on in case we are dealing with some corrupted buffer ?

ssn365, 2020-06-24:
I only have this error with Rigol Device.  I have also tested on keysight device and it works fine.

MatthieuDartiailh, 2020-06-24:
Could you send to me a screenshot of the other tabs of the VISA test panel ? Just in case there is some thing at odds with pyvisa defaults.

ssn365, 2020-06-26:
Morning the code worked fine.I didnt no what has chaged.

>>> import visa
>>> rm = visa.ResourceManager()
>>> inst = rm.open_resource('USB0::0x1AB1::0x0643::DG8A220800316::0::INSTR')
>>> print(inst.query("*IDN?"))
Rigol Technologies,DG811,DG8A220800316,00.02.04.00.00

>>> import visa
>>> rm = visa.ResourceManager()
>>> inst = rm.open_resource('USB0::0x1AB1::0x0643::DG8A220800316::0::INSTR')
>>> inst.write_termination = '\n'
>>> inst.read_termination = '\n'
>>> print(inst.query("*IDN?"))
Rigol Technologies,DG811,DG8A220800316,00.02.04.00.00
>>>

MatthieuDartiailh, 2020-07-02:
That is getting weirder and weirder. Somebody mentioned in another thread that its RIGOL instrument had support for multiple set of instructions (Agilent-like, Fluke-like, etc). Is it your case and did you try switching the set of instructions ?
Sorry I have little time at the moment to look deeper.