pyvisa/pyvisa #522: [COM] Communication issue with keithley 6487 picoammeter using pyvisa

niteia, 2020-06-26:
I have been reading the tutorial in order to configure the communication between my computer and my device, unfortunately, I have issues with termination ; when using the query : my_instrument.query('*IDN?') , I get a timeout error. According to the tutorial, I must find, in the documentation of my device (given below) the values for the read_termination and write_termination attributes of the ressource, but i didn't find them... I don't really understand what are the type of these values ( the manual is not very clear for me either) 

Instrument details
------------------
* Model: keithley 6487
* Communication:  GPIB,
* Link to the documentation (if available):
https://download.tek.com/manual/6487-901-01(B-Mar2011)(Ref).pdf

Output of `pyvisa-info`
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
-----------------------

MatthieuDartiailh, 2020-06-26:
On page 9-15, and 9-16, the manual describe Program Message terminator and the Response message terminator which are both LF (\n) (EOI is a hardware signal). As a consequence try something along the following lines:
``` python
import pyvisa
rm = pyvisa.ResourceManager()
kei = rm.open_resource("GPIB::10::INSTR")  # Use the proper address here
kei.write_termination = "\n"
kei.read_termination = "\n"
print(kei.query("*IDN?"))
```

MatthieuDartiailh, 2020-07-23:
Closing in the absence of answer. Feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh